OpenZWave/open-zwave #1309: Fibaro Roller Shutter not reacting to StartLevelChange

AlCalzone, 2017-09-04:
I'm trying to control my Fibaro Roller Shutters by using the Inc/Dec ValueIDs in order to make use of the possibility to stop them mid-motion. I tried various values of Step Size and Duration, with no success...

I see the StartLevelChange command in the OZW_Log.txt but I get no reaction from my Fibaros:
```
2017-09-04 21:10:22.059 Info, Node011, Value::Set - COMMAND_CLASS_SWITCH_MULTILEVEL - Inc - 7 - 1 - true
2017-09-04 21:10:22.060 Info, Node011, SwitchMultilevel::StartLevelChange - Starting a level change
2017-09-04 21:10:22.060 Info, Node011,   Direction:          Inc
2017-09-04 21:10:22.060 Info, Node011,   Ignore Start Level: True
2017-09-04 21:10:22.060 Info, Node011,   Start Level:        0
2017-09-04 21:10:22.060 Info, Node011,   Duration:           10
2017-09-04 21:10:22.060 Info, Node011,   Step Size:          10
2017-09-04 21:10:22.060 Detail, Node011, Queuing (Send) SwitchMultilevelCmd_StartLevelChange (Node=11): 0x01, 0x0d, 0x00, 0x13, 0x0b, 0x06, 0x26, 0x04, 0xe0, 0x00, 0x0a, 0x0a, 0x25, 0x19, 0x12
2017-09-04 21:10:22.060 Detail,
2017-09-04 21:10:22.060 Info, Node011, Sending (Send) message (Callback ID=0x19, Expected Reply=0x13) - SwitchMultilevelCmd_StartLevelChange (Node=11): 0x01, 0x0d, 0x00, 0x13, 0x0b, 0x06, 0x26, 0x04, 0xe0, 0x00, 0x0a, 0x0a, 0x25, 0x19, 0x12
2017-09-04 21:10:22.067 Detail, Node011,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2017-09-04 21:10:22.067 Detail, Node011,   ZW_SEND_DATA delivered to Z-Wave stack
2017-09-04 21:10:22.113 Detail, Node011,   Received: 0x01, 0x07, 0x00, 0x13, 0x19, 0x00, 0x00, 0x05, 0xf7
2017-09-04 21:10:22.113 Detail, Node011,   ZW_SEND_DATA Request with callback ID 0x19 received (expected 0x19)
2017-09-04 21:10:22.113 Info, Node011, Request RTT 52 Average Request RTT 51
2017-09-04 21:10:22.113 Detail,   Expected callbackId was received
2017-09-04 21:10:22.113 Detail,   Expected reply was received
2017-09-04 21:10:22.113 Detail,   Message transaction complete
2017-09-04 21:10:22.113 Detail,
2017-09-04 21:10:22.114 Detail, Node011, Removing current message
2017-09-04 21:10:22.896 Info, Node011, Value::Set - COMMAND_CLASS_SWITCH_MULTILEVEL - Dec - 8 - 1 - true
2017-09-04 21:10:22.896 Info, Node011, SwitchMultilevel::StartLevelChange - Starting a level change
2017-09-04 21:10:22.896 Info, Node011,   Direction:          Dec
2017-09-04 21:10:22.896 Info, Node011,   Ignore Start Level: True
2017-09-04 21:10:22.896 Info, Node011,   Start Level:        0
2017-09-04 21:10:22.896 Info, Node011,   Duration:           10
2017-09-04 21:10:22.896 Info, Node011,   Step Size:          10
2017-09-04 21:10:22.896 Detail, Node011, Queuing (Send) SwitchMultilevelCmd_StartLevelChange (Node=11): 0x01, 0x0d, 0x00, 0x13, 0x0b, 0x06, 0x26, 0x04, 0xe8, 0x00, 0x0a, 0x0a, 0x25, 0x1a, 0x19
2017-09-04 21:10:22.897 Detail,
2017-09-04 21:10:22.897 Info, Node011, Sending (Send) message (Callback ID=0x1a, Expected Reply=0x13) - SwitchMultilevelCmd_StartLevelChange (Node=11): 0x01, 0x0d, 0x00, 0x13, 0x0b, 0x06, 0x26, 0x04, 0xe8, 0x00, 0x0a, 0x0a, 0x25, 0x1a, 0x19
2017-09-04 21:10:22.903 Detail, Node011,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2017-09-04 21:10:22.903 Detail, Node011,   ZW_SEND_DATA delivered to Z-Wave stack
2017-09-04 21:10:22.949 Detail, Node011,   Received: 0x01, 0x07, 0x00, 0x13, 0x1a, 0x00, 0x00, 0x04, 0xf5
2017-09-04 21:10:22.950 Detail, Node011,   ZW_SEND_DATA Request with callback ID 0x1a received (expected 0x1a)
2017-09-04 21:10:22.950 Info, Node011, Request RTT 52 Average Request RTT 51
2017-09-04 21:10:22.950 Detail,   Expected callbackId was received
2017-09-04 21:10:22.950 Detail,   Expected reply was received
2017-09-04 21:10:22.950 Detail,   Message transaction complete
2017-09-04 21:10:22.950 Detail,
2017-09-04 21:10:22.950 Detail, Node011, Removing current message
```

Fishwaldo, 2017-09-06:
Do you have a link to the technical specs of the RolerShutter - I'd like to confirm that StartLevel/StopLevel commands are supported first... 

at a high level look, everything looks ok...

AlCalzone, 2017-09-06:
The product details can be found here: http://manuals.fibaro.com/roller-shutter-2/
This is the manual: http://manuals.fibaro.com/content/manuals/en/FGR-222/FGR-222-EN-A-v1.1.pdf

I'm inclined to believe that the device support those commands, based on this forum thread: http://forum.iobroker.net/viewtopic.php?p=82063#p82063 (in german though).

Basically the user CitaroMAN has issues controlling his device from ioBroker (through node-openzwave-shared), meaning positioning was unreliable and no ability to stop the motion.
But he was able to fully control it using Homee, including stopping it mid-motion.

Fishwaldo, 2017-09-21:
I just pushed what should be a fix to the master branch. Can you confirm it works now?

AlCalzone, 2017-09-21:
I will check it later tonight, thx!

AlCalzone, 2017-09-21:
Still seems a bit buggy. `Close` and `Dec` do not work:
```
2017-09-21 15:17:55.015 Info, Node011, Value::Set - COMMAND_CLASS_SWITCH_MULTILEVEL - Dec - 8 - 1 - true
2017-09-21 15:17:55.015 Info, Node011, SwitchMultilevel::StartLevelChange - Starting a level change
2017-09-21 15:17:55.015 Warning, Node011, _direction Value was greater than range. Dropping
```
`Open` and `Inc` do, but the step size is ignored (should it?).

Start and Stop does work now, 👍

Fishwaldo, 2017-09-21:
Ahhh. I know what’s wrong there. I’ll push something tomorrow to fix that. 

Thanks for testing. I guess this is the first time we have had a roller blind device cause that code is soooo old!

Fishwaldo, 2017-09-21:
On step size, can you remove zwcfg*.xml and restart. You should see a version request for switchmultilevel in the logs and a reply. I think step is only supported on versions 3 and above.

AlCalzone, 2017-09-21:
No worries :) 
The step thing doesn't work - and I think its not supposed to. Its the step size for the **secondary** switch function, which is not present in roller shutters.

Fishwaldo, 2017-09-22:
Attempt No 2. Please try again.

AlCalzone, 2017-09-22:
Works 👍

closed 2017-09-22 by AlCalzone