OpenZWave/open-zwave #1316: Continue HID support

gizmocuz, 2017-09-13:
I would like to ask this question again as there is not so many feedback from the ozw core developers on the forum here:

https://groups.google.com/forum/#!searchin/openzwave/HID%7Csort:relevance/openzwave/eFge7COuY9w/Z1o52C7AAwAJ

Is this OK ?

My idea is to make a Preprocessor (HID_SUPPORT) that is default enabled in the make script, but can be turned off.

The last week(s) i am struggling making a binary compile that is compatible between armv7 and armv6 with the new debian release.
Removing the HID dependency solves many issues.

Also for those users using a HID controller (and i think there are almost none), for a few dollars they can buy a usb stick these days

Fishwaldo, 2017-09-14:
I think thats fine. In fact, I'd default it to disabled.

Can you prepare a PR against Dev tree for this?

gizmocuz, 2017-09-15:
Sure! Please give me a few days as i'm on a tight (work) schedule till tuesday

PR #1516 (2018-07-31): Make HID support optional (and disabled by default)
This allows us to eliminate the libudev dependency.

Fixes #1316
commit 89d5b398 (2018-07-31): Build without HID/udev support by default

Allow 'USE_HID=1' to be added to the make command line to turn it on
again, and fix the pkg-config file to link against libudev when needed
for a static libopenzwave.a

Fixes #1316
closed 2018-07-31 by Fishwaldo