OpenZWave/open-zwave #1319: Can't set Qubino PWM thermostat temperature

vytautasgimbutas, 2017-09-15:
Hello,
I'm trying to set the temperature for Qubino PWM thermosat but it just rollbacks to the default (23) immediately. Even though the thermostat mode is set to "Heat" it still says "Cooling" in the screenshot. When I change the mode to "Cooling", I can set the temperature to whatever I want. After switching the mode back to Heat the temperature stayed. Which is not so bad, but still it's a bug.

[image]

I was told by the Qubino support team that it's an issue with Open ZWave:
> Thank you for reaching out to us regarding this issue. Unfortunately flush PWM and ON/OFF thermostats are not compatible with OPEN Z-Wave. Because when you configure the set point it set it for cooling but it does not set it as heating.

chedryk, 2017-09-18:
Hello,
I have the same problem, with Qubino On/Off thermostat.

jensjakob, 2017-09-25:
Using Qubino Flush ON/OFF Thermostat with Home Assistant on a Raspberry Pi with Ubuntu Server, the temperature is always the same as current_temperature. The temperature should be the setpoint temperature.

Using COMMAND_CLASS_THERMOSTAT_SETPOINT doesn't work. The log is also reporting "Cooling 1", and "Refreshed Value: old value=0, new value=0". See full log below.

```
2017-09-22 18:43:19.136 Info, Node004, Value::Set - COMMAND_CLASS_THERMOSTAT_SETPOINT - Cooling 1 - 2 - 1 - 22.6
2017-09-22 18:43:19.137 Detail, Node004, Queuing (Send) ThermostatSetpointCmd_Set (Node=4): 0x01, 0x0d, 0x00, 0x13, 0x04, 0x06, 0x43, 0x01, 0x02, 0x42, 0x08, 0xd4, 0x25, 0x2d, 0x35
2017-09-22 18:43:19.138 Detail, Node004, Queuing (Send) ThermostatSetpointCmd_Get (Node=4): 0x01, 0x0a, 0x00, 0x13, 0x04, 0x03, 0x43, 0x02, 0x02, 0x25, 0x2e, 0xa9
2017-09-22 18:43:19.138 Detail, 
2017-09-22 18:43:19.139 Info, Node004, Sending (Send) message (Callback ID=0x2d, Expected Reply=0x13) - ThermostatSetpointCmd_Set (Node=4): 0x01, 0x0d, 0x00, 0x13, 0x04, 0x06, 0x43, 0x01, 0x02, 0x42, 0x08, 0xd4, 0x25, 0x2d, 0x35
2017-09-22 18:43:19.148 Detail, Node004,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2017-09-22 18:43:19.149 Detail, Node004,   ZW_SEND_DATA delivered to Z-Wave stack
2017-09-22 18:43:19.164 Detail, Node004,   Received: 0x01, 0x07, 0x00, 0x13, 0x2d, 0x00, 0x00, 0x02, 0xc4
2017-09-22 18:43:19.165 Detail, Node004,   ZW_SEND_DATA Request with callback ID 0x2d received (expected 0x2d)
2017-09-22 18:43:19.165 Info, Node004, Request RTT 25 Average Request RTT 25
2017-09-22 18:43:19.165 Detail,   Expected callbackId was received
2017-09-22 18:43:19.165 Detail,   Expected reply was received
2017-09-22 18:43:19.166 Detail,   Message transaction complete
2017-09-22 18:43:19.166 Detail, 
2017-09-22 18:43:19.166 Detail, Node004, Removing current message
2017-09-22 18:43:19.166 Detail, 
2017-09-22 18:43:19.166 Info, Node004, Sending (Send) message (Callback ID=0x2e, Expected Reply=0x04) - ThermostatSetpointCmd_Get (Node=4): 0x01, 0x0a, 0x00, 0x13, 0x04, 0x03, 0x43, 0x02, 0x02, 0x25, 0x2e, 0xa9
2017-09-22 18:43:19.174 Detail, Node004,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2017-09-22 18:43:19.175 Detail, Node004,   ZW_SEND_DATA delivered to Z-Wave stack
2017-09-22 18:43:19.244 Detail, Node004,   Received: 0x01, 0x07, 0x00, 0x13, 0x2e, 0x00, 0x00, 0x07, 0xc2
2017-09-22 18:43:19.244 Detail, Node004,   ZW_SEND_DATA Request with callback ID 0x2e received (expected 0x2e)
2017-09-22 18:43:19.244 Info, Node004, Request RTT 77 Average Request RTT 51
2017-09-22 18:43:19.244 Detail,   Expected callbackId was received
2017-09-22 18:43:19.284 Detail, Node004,   Received: 0x01, 0x0a, 0x00, 0x04, 0x00, 0x04, 0x04, 0x43, 0x03, 0x02, 0x00, 0xb3
2017-09-22 18:43:19.284 Detail, 
2017-09-22 18:43:19.284 Info, Node004, Response RTT 117 Average Response RTT 94
2017-09-22 18:43:19.285 Detail, Node004, Refreshed Value: old value=0, new value=0, type=decimal
2017-09-22 18:43:19.285 Detail, Node004, Changes to this value are not verified
2017-09-22 18:43:19.285 Info, Node004, Received thermostat setpoint report: Setpoint Cooling 1 = 0C
2017-09-22 18:43:19.285 Detail, Node004,   Expected reply and command class was received
2017-09-22 18:43:19.285 Detail, Node004,   Message transaction complete
2017-09-22 18:43:19.285 Detail, 
2017-09-22 18:43:19.285 Detail, Node004, Removing current message
2017-09-22 18:43:19.286 Detail, Node004, Notification: ValueChanged
```

vytautasgimbutas, 2017-12-22:
Just got the on/off thermostat and having the same problem as @jensjakob. Found a workaround with setting high thresholds for hysteresis on/off parameters. This keeps the heating valve on always, but then I have to manually implement the logic for changing the "heating mode" to "off" or "heat". 

Perhaps someone found a better workaround? Like how to set the temperature at least once?

jensjakob, 2017-12-27:
I returned mine.

hanzoh, 2018-01-14:
If someone can give me a starting point where to look to change this, I could try to fix this in a PR. In FHEM it is working okay to set the temperature.

termdude, 2018-01-29:
I have the same problem, but when setting the antifreeze parameter (45). Any way to fix that as well?

sergeymaysak, 2018-08-01:
**OK** - Guys - if somebody still needs this ). I have the same situation and sounds as I have some progress in enabling Qubino Flush On/Off Thermostat in hass. 

**First** - make sure openzwave device description file contains base="0" in CommandClass id="67" element. Having this you need remove node and then add node back device to network. You should see "Heating 1" device instead "Cooling 1".

**Second** - it looks like target temperature can be set only using the following procedure:
- set thermostat mode to Off
- specify target temperature in hass UI
- save config in zwave control panel
- restart hass

Not really convenient but works. 

Cheers )

michiil, 2018-11-30:
Any news on this?
I have hot the pwm thermostat and no matter to what i set the temp, it gets reset to default.
Also the entitie shows up as 
climate.qubino_zmnhldx_flush_pwm_thermostat_cooling_1_3
So im guessing thats where the issue starts..

michiil, 2018-12-01:
Okay i changed 
`<CommandClass id="67" override_precision="2" />`
to 
`<CommandClass id="67" override_precision="2" base="0" />`

In the ZMNHLDx.xml and now it does work. (i'm gonna make a pull request for this)

Now the only problem left if that when setting the temperature, openzwave imeadeatly wants to pull the temperature to verify that it changed. But this request is always timing out:
`2018-12-01 12:56:07.475 Info, Node013, Sending (Send) message (Callback ID=0x4c, Expected Reply=0x13) - ThermostatSetpointCmd_Set (Node=13): 0x01, 0x0d, 0x00, 0x13, 0x0d, 0x06, 0x43, 0x01, 0x01, 0x42, 0x08, 0x66, 0x25, 0x4c, 0xec
2018-12-01 12:56:07.483 Detail, Node013,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2018-12-01 12:56:07.483 Detail, Node013,   ZW_SEND_DATA delivered to Z-Wave stack
2018-12-01 12:56:07.502 Detail, Node013,   Received: 0x01, 0x05, 0x00, 0x13, 0x4c, 0x00, 0xa5
2018-12-01 12:56:07.502 Detail, Node013,   ZW_SEND_DATA Request with callback ID 0x4c received (expected 0x4c)
2018-12-01 12:56:07.502 Info, Node013, Request RTT 26 Average Request RTT 25
2018-12-01 12:56:07.502 Detail,   Expected callbackId was received
2018-12-01 12:56:07.502 Detail,   Expected reply was received
2018-12-01 12:56:07.502 Detail,   Message transaction complete
2018-12-01 12:56:07.502 Detail,
2018-12-01 12:56:07.502 Detail, Node013, Removing current message
2018-12-01 12:56:07.502 Detail,
2018-12-01 12:56:07.502 Info, Node013, Sending (Send) message (Callback ID=0x4d, Expected Reply=0x04) - ThermostatSetpointCmd_Get (Node=13): 0x01, 0x0a, 0x00, 0x13, 0x0d, 0x03, 0x43, 0x02, 0x01, 0x25, 0x4d, 0xc0
2018-12-01 12:56:07.510 Detail, Node013,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2018-12-01 12:56:07.510 Detail, Node013,   ZW_SEND_DATA delivered to Z-Wave stack
2018-12-01 12:56:07.533 Detail, Node013,   Received: 0x01, 0x05, 0x00, 0x13, 0x4d, 0x00, 0xa4
2018-12-01 12:56:07.533 Detail, Node013,   ZW_SEND_DATA Request with callback ID 0x4d received (expected 0x4d)
2018-12-01 12:56:07.533 Info, Node013, Request RTT 30 Average Request RTT 27
2018-12-01 12:56:07.533 Detail,   Expected callbackId was received
2018-12-01 12:56:17.503 Error, Node013, ERROR: Dropping command, expected response not received after 1 attempt(s)
2018-12-01 12:56:17.503 Detail, Node013, Removing current message
2018-12-01 12:56:17.503 Detail, Node013, Notification: Notification - TimeOut`

But then clicking refresh right after if works as expeced. So as a workaround i set a polling for the value and now its working as expected:
`2018-12-01 12:57:59.504 Info, Node013, Sending (Poll) message (Callback ID=0x51, Expected Reply=0x04) - ThermostatSetpointCmd_Get (Node=13): 0x01, 0x0a, 0x00, 0x13, 0x0d, 0x03, 0x43, 0x02, 0x01, 0x25, 0x51, 0xdc
2018-12-01 12:57:59.512 Detail, Node013,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2018-12-01 12:57:59.512 Detail, Node013,   ZW_SEND_DATA delivered to Z-Wave stack
2018-12-01 12:57:59.530 Detail, Node013,   Received: 0x01, 0x05, 0x00, 0x13, 0x51, 0x00, 0xb8
2018-12-01 12:57:59.530 Detail, Node013,   ZW_SEND_DATA Request with callback ID 0x51 received (expected 0x51)
2018-12-01 12:57:59.530 Info, Node013, Request RTT 25 Average Request RTT 25
2018-12-01 12:57:59.530 Detail,   Expected callbackId was received
2018-12-01 12:57:59.574 Detail, Node013,   Received: 0x01, 0x0e, 0x00, 0x04, 0x00, 0x0d, 0x06, 0x43, 0x03, 0x01, 0x42, 0x08, 0x66, 0xc3, 0x00, 0x50
2018-12-01 12:57:59.575 Detail,
2018-12-01 12:57:59.575 Info, Node013, Response RTT 70 Average Response RTT 70
2018-12-01 12:57:59.575 Detail, Node013, Refreshed Value: old value=21.50, new value=21.50, type=decimal
2018-12-01 12:57:59.575 Detail, Node013, Changes to this value are not verified
2018-12-01 12:57:59.575 Info, Node013, Received thermostat setpoint report: Setpoint Heating 1 = 21.50C
2018-12-01 12:57:59.575 Detail, Node013,   Expected reply and command class was received
2018-12-01 12:57:59.575 Detail, Node013,   Message transaction complete
2018-12-01 12:57:59.575 Detail,
2018-12-01 12:57:59.575 Detail, Node013, Removing current message
2018-12-01 12:57:59.575 Detail, Node013, Notification: ValueChanged`

PR #1659 (2018-12-02): Fix Qubin PWM Thermostat ThermostatSetpoint
Fixes #1319

Without this you can't set a temperature for the thermostat.
commit 8c059419 (2018-12-02): Readd base attribute

Fixes #1319
closed 2018-12-02 by nechry