OpenZWave/open-zwave #1325: Aeotec Dual Nano Switch ZW132 Missing individual switch controls

Kanishkaz, 2017-09-25:
I'm using a Aeaseotec Z-stick Gne 5 with Homeassistant. 

I have successfully included an Aeotec Dual Nano Switch (ZW 132) into Homeassistant. However, only a single switch (Binary Switch) is available. This switch controls the on/ off functionality of both lights. It should have individual binary switches for each switch. The Zcfg file shows a single instance of a binary switch.

I have attached the config and log files.

[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/1328648/OZW_Log.txt)
[zwcfg_0xd3add528.txt](https://github.com/OpenZWave/open-zwave/files/1328660/zwcfg_0xd3add528.txt)

dcarrion87, 2017-10-26:
Exact same issue here...

nechry, 2017-10-26:
I also have the same problem as you. When I realized the configurations of the new nano switches, for both models in double relay, I also observed this problem. I do not have time to dig to understand exactly what's going on.

dcarrion87, 2017-10-28:
Looks like there's a firmware update that resolves this issue that @Kanishkaz. I'm actually going to ping them to make it publicly available. 

In saying that I noticed there's a odd issue with mapping when HAIL commands come back from the switch endpoints resulting in delayed updates:

2017-10-28 18:17:06.882 Detail, Node018,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2017-10-28 18:17:06.890 Detail, Node018,   ZW_SEND_DATA delivered to Z-Wave stack
2017-10-28 18:17:06.900 Detail, Node018,   Received: 0x01, 0x07, 0x00, 0x13, 0x5c, 0x00, 0x00, 0x02, 0xb5
2017-10-28 18:17:06.900 Detail, Node018,   ZW_SEND_DATA Request with callback ID 0x5c received (expected 0x5c)
2017-10-28 18:17:06.900 Info, Node018, Request RTT 25 Average Request RTT 30
2017-10-28 18:17:06.900 Detail,   Expected callbackId was received
2017-10-28 18:17:06.912 Detail, Node018,   Received: 0x01, 0x0d, 0x00, 0x04, 0x00, 0x12, 0x07, 0x60, 0x0d, 0x02, 0x01, 0x25, 0x03, 0xff, 0x54
2017-10-28 18:17:06.912 Detail,
2017-10-28 18:17:06.912 Info, Node018, Response RTT 37 Average Response RTT 49
2017-10-28 18:17:06.912 Info, Node018, Received a MultiChannelEncap from node 18, endpoint 2 for Command Class COMMAND_CLASS_SWITCH_BINARY
2017-10-28 18:17:06.912 Info, Node018, Received SwitchBinary report from node 18: level=On
2017-10-28 18:17:06.912 Detail, Node018, Refreshed Value: old value=false, new value=true, type=bool
2017-10-28 18:17:06.912 Detail, Node018, Changes to this value are not verified
2017-10-28 18:17:06.912 Detail, Node018,   Expected reply and command class was received
2017-10-28 18:17:06.912 Detail, Node018,   Message transaction complete
2017-10-28 18:17:06.912 Detail,
2017-10-28 18:17:06.912 Detail, Node018, Removing current message
2017-10-28 18:17:06.913 Detail, Node018, Notification: ValueChanged
2017-10-28 18:17:10.896 Detail, Node018,   Received: 0x01, 0x0c, 0x00, 0x04, 0x00, 0x12, 0x06, 0x60, 0x0d, 0x02, 0x00, 0x82, 0x01, 0x0f
2017-10-28 18:17:10.896 Detail,
**2017-10-28 18:17:10.896 Error, Node018, Cannot find endpoint map to instance for Command Class COMMAND_CLASS_HAIL endpoint 2**

nechry, 2017-10-28:
Great, I will ask again for a OTA update, I already ask in January but at this time no clues.

Pstrelnz, 2017-10-28:
Just read the engineering sheet and it looks like the switch binary on endpoints 1 and 2 has been mapped to the basic commmand class

dcarrion87, 2017-10-31:
They have released:

https://aeotec.freshdesk.com/solution/articles/6000185238-zw132-dual-nano-switch-v1-04-firmware-update-

nechry, 2017-11-20:
I updated, but I sill have a wrong behavior. anyone else perform the OTA ?

ben423423n32j14e, 2018-01-08:
yep, I applied the OTA, excluded and re-included to Domoticz and I still only see the single switch. Pretty disappointed they didn't test this, I really need to be able to control both switches individually.

ben423423n32j14e, 2018-01-08:
I've figured it out, the first few update attempts were not successful. It was on the fourth attempt that it succeeded (the utility will produce a pop-up dialogue box saying it was successful). I then reset the ZW132 and re-added it as a secure device. The two individual control switches appeared in Domoticz and I can now control each device individually as expected.

nechry, 2018-01-08:
it's fully operational from controller and with manuals operations direct on device?

ben423423n32j14e, 2018-01-08:
yes

nechry, 2018-01-09:
ok the auto discovering switch did not work in your case. 
from my side I perform some tests yesterday, and I got issue with power reporting. the instance 1 report both power (s1 + s2) and instance 2 the s2 so no way to get only power for S1. do you have same behavior? I still thinking the device is not fully compatible.

Swamp-Ig, 2018-08-11:
I'm currently using two of these devices in home assistant without any issues. Maybe this is resolved?

Fishwaldo, 2019-05-13:
Closing this

closed 2019-05-13 by Fishwaldo