OpenZWave/open-zwave #1326: Vision ZD2102-5: Not working as 'External Switch'

kmandevnull, 2017-09-25:
Hi,
I am using openzwave version - 1.4-2499-g636ec52-dirty

When I use ZD2102-5 as an [external](http://www.vesternet.com/downloads/dl/file/id/617/product/1512/z_wave_vision_door_window_sensor_gen5_manual.pdf) switch I don't receive anything. There is nothing in the OZW logs nor in the domoticz logs. For the external switch I am using a [reed switch](https://www.amazon.co.uk/gp/product/B00PZMG980/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1). 

Now, if I turn off the ZD2102-5's external switch ([see page 2](http://www.vesternet.com/downloads/dl/file/id/617/product/1512/z_wave_vision_door_window_sensor_gen5_manual.pdf)), just use it as a normal door/window with a magnet, open/close events show up correctly.

I am using aeon-gen5 ZStick. 

Not sure, how to debug this issue - when issue [885 was resolved ](https://github.com/OpenZWave/open-zwave/issues/885) was it tested for External Switch? (from the looks of it, it doesn't appear so!)

Any suggestions?

Many thanks.

kmandevnull, 2017-09-25:
Just like to add:- Against the controlled I have 'enabled' debug in domoticz. So, as I said, even with this I don't see anything in the logs when I use ZD2102-5 as an external switch.

Fishwaldo, 2017-09-25:
Check the associations.

nechry, 2017-09-26:
you must activate the external switch with param 1. 
https://products.z-wavealliance.org/products/988/configs

And special note from documentation:
> External Switch: If your window needs two sensors for fully monitoring. You could use an external switch to connect with the ZD2102 for extending detection. The ZD2102 has 2 normally closed contact terminals. These can be used for additional external switch wired contacts.
> (Remark: Either reed switch or external switch has been opened means “open”, but, both of them should be closed at the same time means “close”.)

kmandevnull, 2017-09-26:
Thanks for the responses.

Only association available is Node 1 (controller) to which it is already associated. And that's the reason it works well, with the normal reed switch (i.e. just magnet and sensor, not external switch). Issue is only when I connect the wires of the external switch, and then do open/close with this external switch - I don't receive anything.

And yes, external switch is enabled in the configuration
[image]

kmandevnull, 2017-09-26:
Any suggestions please, on how to proceed?

nechry, 2017-09-27:
check whether the magnet, in place or remote, interfere with the condition of the external contact

closed 2017-12-12 by Fishwaldo