OpenZWave/open-zwave #1332: Aeotec MultiSensor 6 (ZW100) temperature threshold change unit

cincuranet, 2017-09-28:
The [data sheet for FW 1.08](https://aeotec.freshdesk.com/helpdesk/attachments/6052990305) show that the size for value 41 is `3` and the value is now two part - the threshold itself and unit.

>2. High byte is the threshold value. Low byte is the unit (0x01=Celsius,
0x02=Fahrenheit).
> 3. The threshold value (high byte) contains one decimal point. E.g. if the
value is set to 20 (0x001401), the threshold value =2.0 ℃ (EU/AU
version) or if the value is set to 20 (0x001402), the threshold value= 2.0
℉ (US version). When the current temperature gap is more then 2.0,
which will induce a temperature report to be sent out. 

Thus probably the size, description (I can do PR for that) and probably value handling should be adjusted.

nechry, 2017-09-28:
The fw 1.08 still in beta, I prefert wait for final release to change size attribute.

cincuranet, 2017-09-28:
Not sure whether the support article is properly updated. Mine came few weeks ago with 1.08 already from factory. Go figure. :(

cincuranet, 2017-09-28:
Hmm, I'm looking into [1.07 sheet](https://aeotec.freshdesk.com/helpdesk/attachments/6053297238) and the size `3` is already there.

nechry, 2017-09-28:
ah yes, but here the problem is in openzwave we do not have the size definition of 3, for configuration parameters.
https://github.com/OpenZWave/open-zwave/blob/master/cpp/src/command_classes/Configuration.cpp#L123

nechry, 2017-09-28:
see also http://zwavepublic.com/sites/default/files/command_class_specs_2017A/SDS13781-3%20Z-Wave%20Application%20Command%20Class%20Specification.pdf page 97
Size (3 bits) This field is used to  specify the size of the actual parameter.
possible value are 1, 2 and 4 like us.

I think is a typo from aeotec, I already talk about this with them.

cincuranet, 2017-09-28:
Probably yes. Value 49 is in sheet `3` as well, but it's [`int` in xml](https://github.com/OpenZWave/open-zwave/blob/master/config/aeotec/zw100.xml#L111). Maybe I can safely change it to `int` and update the description and do the PR... Should I?

nechry, 2017-09-28:
we must perform some test. try to adjust form size 2 to 4 play with parameter values, ensure the device accept them. If he refuse, you must rollback and do try with size of 2. you have to exclude and include again every time you change the definition. I will also ask to my contact at aeotec to get news about the 1.08 beta and the eng sheet. If the 1.08 is the stable version we will update the definition in phase.

cincuranet, 2017-10-02:
Yep. Seems to be fine with `4`. Aeotec support also confirmed it.

cincuranet, 2017-10-02:
From my testing it seems that the size `4` is accepted. Aeotec support also confirms it:

> The Multisensor 6 will output a 3 byte size as a return (which most gateways will not like and show a random value for the byte size), but accepts a 4 byte input for configuring this parameter setting.

But having the return as size `3` seems to be quite confusing to me.

nechry, 2017-10-02:
quite, especially that in the specification of the command class it is not supported (the size of 3)

cincuranet, 2017-10-03:
Well, what we're gonna do now. Sending 2 bytes obviously works as long as your value fits there. But then... But then we have the hassle of reading it back. Maybe read 4 bytes back as a band-aid and developer will be left alone skipping the one byte that's basically a garbage? Lame right?

Fishwaldo, 2017-10-03:
We will have to write a quirk (and a config option) to address this. Let me think about it first tho.

cincuranet, 2017-10-04:
Reading through some docs I've spotted [this](https://aeotec.freshdesk.com/support/solutions/articles/6000036562-multisensor-6-firmware-update-6-17-2016-new-information-v1-08-11-02-2016-). At the bottom it says:
[image]

I suppose the `2` is another typo, but the note about not sending size `3` back would be good. I'll double check how it behaves on my MS6 with factory 1.08.

Fishwaldo, 2017-12-12:
is this still a issue or did they fix it?

dmulcahey, 2017-12-26:
https://aeotec.freshdesk.com/support/solutions/articles/6000036562-multisensor-6-firmware-update-12-14-2017-

They actually released a new firmware on 12-14-2017

Fishwaldo, 2017-12-27:
So the issue is fixed with the new firmware?

dmulcahey, 2017-12-28:
From the link posted above:

V1.10 + V1.11:
1. Added addition setting for Parameter 81 to completely disable LED. 
Parameter 81 [1 byte size]
0 = All LED allowed
1 = Disable Motion sensor only.
2 = Disable all LED activity

2. Size change for Parameters 41, 49, and 50 to 4 byte size.
3. Updated Z-Wave Library
4. US, AU frequency V1.10 equivalent to V1.11 EU frequency.
5. Version CC and Manufacturing CC listed under secure network.
6. Improved accuracy of lighting sensor

Specifically, Item number 2, looks like parameters 41, 49, and 50 were changed to 4 bytes.

CookieMonster87, 2018-02-06:
Will this issue be fixed? Can not adjust the paramter 41 after firmware updating Multisensor 6 to V1.11
If i specify example 10 or 20 it just change to 256.

bholveck, 2018-03-06:
Same issue here I can't set the value of parameter 41. If setting is untouched then the value 20 is correct, but as soon as I change it, the returned value is 256 and the sensor is almost impossible to configure anymore as it sends a report each second.

cincuranet, 2018-03-09:
I think Aeotec fucked this up greatly. It's a breaking change and also some value bits formats changed. Real mess.

Not sure how to solve it nicely. Of course it can be changed to size 4 and people using previous FWs would limit themselves only to using 2, but that seems, without internal knowledge, confusing. At least we can explain it in the description in XML.

If we agree on some solution I can do that then.

CookieMonster87, 2018-03-09:
I think its ok to explain it in the description/help field. Is it possible to have different xml config files based on firmware version?

bholveck, 2018-03-09:
So after some research I understood that OZW version I'm using is the one built in Domoticz (derivative of v1.4).
I then copied the ZW100 xml config file to the Domoticz' location and was able to change the value through the OZW web console.
The field value is correct regarding ZW100 v1.1 firmware, but the documentation of the field is not obviously.
The issue here, as CookieMonster87 said, would be to have support of different xml based on the device's firmware version.

Fishwaldo, 2018-03-09:
see #644

Fishwaldo, 2018-10-26:
Closing as this will be addressed with #664

cincuranet, 2018-12-26:
Never happened to me. 🤞

closed 2018-12-26 by Fishwaldo