OpenZWave/open-zwave #1337: ApplicationCommandHandler - Unhandled Command Class 0x71

jeroenst, 2017-10-07:
When receiving a 0x71 message from a PIR i see this message in the log:

ApplicationCommandHandler - Unhandled Command Class 0x71

The function registered with Manager::Get()->AddWatcher is not called..

2017-10-07 17:26:59.306 Detail, Node008,   Received: 0x01, 0x11, 0x00, 0x04, 0x00, 0x08, 0x0b, 0x71, 0x05, 0x07, 0x00, 0x00, 0xff, 0x07, 0x00, 0x01, 0x08, 0x00, 0x6b
2017-10-07 17:26:59.306 Detail,
2017-10-07 17:26:59.306 Info, Node008, ApplicationCommandHandler - Unhandled Command Class 0x71

Fishwaldo, 2017-12-04:
Make sure the Device has fully synced with OZW. This is typical of a sleeping device.

closed 2017-12-04 by Fishwaldo

daveooooooooo, 2020-01-08:
I, too, am having this same issue

Fishwaldo, 2020-01-09:
Wake up the device. If it’s sleeping OZW can’t talk to it till it wakes up.