OpenZWave/open-zwave #1340: Eurotronic Z-Wave Plus SPIRIT Thermostat - no Setpoint

michapr, 2017-10-16:
The new Z-Wave Plus radiator thermostat sis not working at OZW - the "right" SetPoint device is missing.
(using  1.4-2501-g5af6780-dirty)
Devie will be included without problems - but I get only a 
Cooling
Dry Air
Setpoint device - but this will not give the right values back - and Setpoint can not be set.

Setting a setpoint at Thermostat I get in Log:
```
2017-10-16 20:37:00.655 Detail, Node004,   Received: 0x01, 0x0c, 0x00, 0x04, 0x00, 0x04, 0x06, 0x43, 0x03, 0x01, 0x22, 0x00, 0xaa, 0x3c
2017-10-16 20:37:00.656 Detail,
...
2017-10-16 21:05:17.226 Detail, Node004,   Received: 0x01, 0x0c, 0x00, 0x04, 0x00, 0x04, 0x06, 0x43, 0x03, 0x01, 0x22, 0x00, 0xaf, 0x39
2017-10-16 21:05:17.226 Detail,
```
Changing Thermostat Mode from "Heat" to "OFF" all is working fine.
From OZW panel (Domoticz):
[image]

Inclusion log attached.
[ozw_include_log.txt](https://github.com/OpenZWave/open-zwave/files/1388745/ozw_include_log.txt)

Mabe there is missing something?
Only for comparing - Thermostat will be recognized as standard Thermostat in Fibaro HomeCenter - all values will be shown here right.

I hope soebody could check it. If any additional log is needed - please tell me.

Thanks!
Michael

roflmao, 2017-10-16:
Documentation of the device: (shows available classes) https://www.eurotronic.org/fileadmin/user_upload/eurotronic.org/Produktbilder/spirit_z_wave_plus/Spirit_Z-Wave_BAL_web_EN_view_04.pdf

Zwave alliance: https://products.z-wavealliance.org/products/2543

zwcfg:

```
	<Node id="41" name="" location="" basic="4" generic="8" specific="6" roletype="7" devicetype="4608" nodetype="0" type="General Thermostat V2" listening="false" frequentListening="true" beaming="true" routing="true" max_baud_rate="40000" version="4" secured="true" query_stage="Complete">
		<Manufacturer id="148" name="EUROtronic">
			<Product type="3" id="1" name="Unknown: type=0003, id=0001" />
		</Manufacturer>
		<CommandClasses>
			<CommandClass id="32" name="COMMAND_CLASS_BASIC" version="1" request_flags="4" mapping="64">
				<Instance index="1" />
			</CommandClass>
			<CommandClass id="38" name="COMMAND_CLASS_SWITCH_MULTILEVEL" version="1" issecured="true" innif="true">
				<Instance index="1" />
				<Value type="byte" genre="user" instance="1" index="0" label="Level" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="254" />
				<Value type="button" genre="user" instance="1" index="1" label="Bright" units="" read_only="false" write_only="true" verify_changes="false" poll_intensity="0" min="0" max="0" />
				<Value type="button" genre="user" instance="1" index="2" label="Dim" units="" read_only="false" write_only="true" verify_changes="false" poll_intensity="0" min="0" max="0" />
				<Value type="bool" genre="system" instance="1" index="3" label="Ignore Start Level" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="True" />
				<Value type="byte" genre="system" instance="1" index="4" label="Start Level" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="0" />
			</CommandClass>
			<CommandClass id="49" name="COMMAND_CLASS_SENSOR_MULTILEVEL" version="5" issecured="true" innif="true">
				<Instance index="1" />
				<Value type="decimal" genre="user" instance="1" index="1" label="Temperature" units="C" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="20.96" />
			</CommandClass>
			<CommandClass id="64" name="COMMAND_CLASS_THERMOSTAT_MODE" version="1" request_flags="4" issecured="true" innif="true">
				<Instance index="1" />
				<Value type="list" genre="user" instance="1" index="0" label="Mode" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" vindex="1" size="1">
					<Item label="Off" value="0" />
					<Item label="Heat" value="1" />
					<Item label="Heat Econ" value="11" />
				</Value>
				<SupportedModes>
					<Mode index="0" label="Off" />
					<Mode index="1" label="Heat" />
					<Mode index="11" label="Heat Econ" />
				</SupportedModes>
			</CommandClass>
			<CommandClass id="67" name="COMMAND_CLASS_THERMOSTAT_SETPOINT" version="1" request_flags="4" issecured="true" innif="true" base="1">
				<Instance index="1" />
				<Value type="decimal" genre="user" instance="1" index="2" label="Cooling 1" units="C" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="0.0" />
				<Value type="decimal" genre="user" instance="1" index="8" label="Dry Air" units="C" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="0.0" />
			</CommandClass>
			<CommandClass id="90" name="COMMAND_CLASS_DEVICE_RESET_LOCALLY" version="1" request_flags="4" issecured="true" innif="true">
				<Instance index="1" />
			</CommandClass>
			<CommandClass id="94" name="COMMAND_CLASS_ZWAVE_PLUS_INFO" version="1" request_flags="4" innif="true">
				<Instance index="1" />
				<Value type="byte" genre="system" instance="1" index="0" label="ZWave+ Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="1" />
				<Value type="short" genre="system" instance="1" index="1" label="InstallerIcon" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="-32768" max="32767" value="4608" />
				<Value type="short" genre="system" instance="1" index="2" label="UserIcon" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="-32768" max="32767" value="4608" />
			</CommandClass>
			<CommandClass id="112" name="COMMAND_CLASS_CONFIGURATION" version="1" request_flags="4" issecured="true" innif="true">
				<Instance index="1" />
			</CommandClass>
			<CommandClass id="113" name="COMMAND_CLASS_ALARM" version="8" issecured="true" innif="true">
				<Instance index="1" />
				<Value type="byte" genre="user" instance="1" index="0" label="Alarm Type" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="0" />
				<Value type="byte" genre="user" instance="1" index="1" label="Alarm Level" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="0" />
				<Value type="byte" genre="user" instance="1" index="2" label="SourceNodeId" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="0" />
[... cut ...]
				<Value type="string" genre="system" instance="1" index="1" label="Protocol Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="4.61" />
				<Value type="string" genre="system" instance="1" index="2" label="Application Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="0.15" />
			</CommandClass>
			<CommandClass id="152" name="COMMAND_CLASS_SECURITY" version="1" request_flags="4" innif="true" />
		</CommandClasses>
	</Node>
```

michapr, 2017-10-17:
@roflmao : can you confirm the issue - or is it working for you?
(not clear for me ;) ..)

What I can see - as additional information:
If device will be included as secure device I will get more information changing the temperature at the device:
```
2017-10-17 08:27:28.962 Detail, Node005,   Received: 0x01, 0x08, 0x00, 0x04, 0x00, 0x05, 0x02, 0x98, 0x40, 0x2c
2017-10-17 08:27:28.963 Info, Node005, Received SecurityCmd_NonceGet from node 5
2017-10-17 08:27:28.963 Info, NONCES: 0x08, 0x05, 0xe0, 0xa0, 0x53, 0x0e, 0x00, 0x5f
2017-10-17 08:27:28.963 Info, NONCES: 0x94, 0x20, 0x6e, 0x49, 0x10, 0x2f, 0xed, 0x44
2017-10-17 08:27:28.964 Info, NONCES: 0xec, 0x16, 0x65, 0xe7, 0x64, 0x14, 0x03, 0x60
2017-10-17 08:27:28.964 Info, NONCES: 0x03, 0x1e, 0x29, 0xd0, 0x14, 0xb8, 0x1c, 0xf5
2017-10-17 08:27:28.964 Info, NONCES: 0xa3, 0xaf, 0xca, 0x3b, 0x2f, 0x11, 0x6d, 0x95
2017-10-17 08:27:28.965 Info, NONCES: 0xfe, 0x94, 0xf5, 0xb7, 0xb1, 0xeb, 0x5a, 0x61
2017-10-17 08:27:28.965 Info, NONCES: 0xef, 0x44, 0x73, 0xb2, 0x19, 0x37, 0x93, 0x17
2017-10-17 08:27:28.965 Info, NONCES: 0x0c, 0x85, 0x20, 0x0f, 0xb4, 0xef, 0xc0, 0xa4
2017-10-17 08:27:28.966 Info, Node005, Sending (Send) message (Callback ID=0x01, Expected Reply=0x00) - Nonce_Report - 0x01, 0x11, 0x00, 0x13, 0x05, 0x0a, 0x98, 0x80, 0x94, 0x20, 0x6e, 0x49, 0x10, 0x2f, 0xed, 0x44, 0x05, 0x01, 0xeb:
2017-10-17 08:27:28.975 Detail,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2017-10-17 08:27:28.975 Detail,   ZW_SEND_DATA delivered to Z-Wave stack
2017-10-17 08:27:28.996 Detail,   Received: 0x01, 0x07, 0x00, 0x13, 0x01, 0x00, 0x00, 0x03, 0xe9
2017-10-17 08:27:28.996 Detail,   ZW_SEND_DATA Request with callback ID 0x01 received (expected 0x01)
2017-10-17 08:27:29.026 Detail, Node005,   Received: 0x01, 0x20, 0x00, 0x04, 0x00, 0x05, 0x1a, 0x98, 0x81, 0x33, 0xbb, 0x21, 0x19, 0x2d, 0xc8, 0x50, 0x19, 0xbe, 0x88, 0xbc, 0x86, 0x6f, 0x42, 0x68, 0x94, 0xfd, 0xc3, 0x14, 0x6c, 0xb5, 0xaf, 0x26, 0xef, 0x89
2017-10-17 08:27:29.027 Info, Raw: 0x98, 0x81, 0x33, 0xbb, 0x21, 0x19, 0x2d, 0xc8, 0x50, 0x19, 0xbe, 0x88, 0xbc, 0x86, 0x6f, 0x42, 0x68, 0x94, 0xfd, 0xc3, 0x14, 0x6c, 0xb5, 0xaf, 0x26, 0xef, 0x89
2017-10-17 08:27:29.027 Detail, Node005, Decrypted Packet: 0x00, 0x43, 0x03, 0x01, 0x22, 0x00, 0xcd
2017-10-17 08:27:29.027 Detail,
```

But no changes - setpoint can not be chaged from OZW.

roflmao, 2017-10-17:
I can confirm the issue. When setting setpoint it just fallbacks to the given temperature by the temperaturesensor.

tobiasgraf, 2017-10-17:
The problem here is that Thermostate_Setpoint has 2 interpretations. OZW seems to follow Interpretation B per default. 

>It has been found that early implementations of this Command Class specification apply two non-
interoperable interpretations of the bit mask advertising the support for specific Setpoint Types.
As a consequence, one may find thermostat products and controller products in the marketplace which
implement either of the two bit mask interpretations found in Table 106. The notation x.y indicates Bit
Mask byte x, bit y.

>Implementations of Thermostat Setpoint Command Class, version 3 MUST comply with Interpretation A.

This Thermostate uses version 3 and thus follows interpretation A.

[image]

I made some changes to the Thermostat_Setpoint class to support this, also support for Version 3 of Thermostat_Setpoint (Capabilities Commands).

I'll create a PR for this.

michapr, 2017-10-18:
Have tested (replaced the published changes and compiled with latest version) - it is working fine!
So we will wait for merge, that it can be live.

Thank you!!

travellingkiwi, 2017-10-25:
The XML implies we should be able to set options for reporting (Temp changes of 0.1deg) and adjust the orientation of the display
But these options don't appear in the OZW console... Why is that?

masterblasterr, 2017-11-21:
Is it live yet, i want to buy this thermostat but, i need to make sure it works with domoticz

Fishwaldo, 2017-12-12:
This should be fixed in Dev branch now with PR #1385 and #1386

closed 2017-12-12 by Fishwaldo