OpenZWave/open-zwave #1356: Support for ZSC-01DW wall mont controller/dimmer

paalkr, 2017-10-27:
The ZSC-01DW wall mount controller is currently not supported. Can it be added?

https://www.tronika.no/en/zwave-controllers/1010-wall-controller-zsc-01dw.html

Regards

paalkr, 2017-10-27:
Some more details pullet from Domoticz
Unknown: id=0330 Unknown: type=0003, id=a10d+

nechry, 2017-11-07:
Nice device

closed 2017-11-07 by nechry

paalkr, 2017-11-07:
Thx

Yes, the device is quite nice. The only thing I miss is light in the touch panel. It can be a little hard to find the correct button in a dark rom.

nechry, 2017-11-07:
I set tronika as Manufacturer but I think is wrong, tronika  is only the shop? Do you have a Manufacturer name ?

paalkr, 2017-11-07:
I did the same my self when adding the device to my local repository files. (sorry I didn't issue a pull request). I contacted Tronika, but they where not very happy to inform about the manufacturer. I guess they have OEMed the device. I'm not sure whom the real manufacturer might be, as I have not seen the device listed at any other resellers.

nechry, 2017-11-07:
Ok, we will see if we find more informations.

paalkr, 2017-11-07:
Good. I have tried to google the model number and all other references that I could think of, but with no luck. If you guys have access to closed source product lists, that might help.

btw, here's the link to the product documentation
https://www.tronika.no/en/index.php?controller=attachment&id_attachment=544

commit 588887ee (2017-11-07): Add Tronika Touch Panel Wall Controller - ZSC-01D

Close #1356