OpenZWave/open-zwave #1358: Custom LogImpl disabled, if default config path does not exist

stromnet, 2017-10-28:
When setting a custom Logimpl, it get's overwritten during startup, if default config path does not exist.

Our code does:
* OpenZWave::Log::SetLoggingClass(myLog);
* Options::Create(..);
* ...
* Manager::Create();

During Options::Create, [Log::Create is called](https://github.com/OpenZWave/open-zwave/blob/Dev/cpp/src/Options.cpp#L73). This is the first call to [Log::Create](https://github.com/OpenZWave/open-zwave/blob/Dev/cpp/src/platform/Log.cpp#L72), so s_instance is NULL and it creates a new Log instance. Log constructor does not do much though, since `m_pImpl` is already set.
So far so god.
In Manager::Create, the Manager constructor is called, which also [calls Log::Create](https://github.com/OpenZWave/open-zwave/blob/Dev/cpp/src/Manager.cpp#L187). This time, Log's `s_instance` is already set, so `Log::Create` will call `Log::Destroy()` before creating new Log.
This will call the (destructor on Log)[https://github.com/OpenZWave/open-zwave/blob/Dev/cpp/src/platform/Log.cpp#L338], which **deletes and resets my externally set** `m_pImpl`.
External logging is now not working anymore.

Fishwaldo, 2017-12-12:
Thanks for the report. I'll look into it.

Fishwaldo, 2018-03-07:
Fixed. Thanks!

closed 2018-03-07 by Fishwaldo

stromnet, 2018-03-07:
Great, thank you! :)