OpenZWave/open-zwave #1359: RemoveFailedNode seems to use node ID in response?

stromnet, 2017-10-28:
On open-zwave master, Z-Stick S2 controller.

Application code:
```
int nodeId = 16;
Manager::Get()->BeginControllerCommand(g_homeId, Driver::ControllerCommand_RemoveFailedNode, controller_update, this, true, nodeId);
```

```
2017-10-28 23:48:12.252 Detail, Node016, Queuing (Controller) Remove Failed Node
2017-10-28 23:48:12.253 Info, ControllerCommand_RemoveFailedNode
2017-10-28 23:48:12.253 Detail, contrlr, Queuing (Command) ControllerCommand_RemoveFailedNode: 0x01, 0x05, 0x00, 0x61, 0x10, 0x80, 0x0b
2017-10-28 23:48:12.253 Detail, Notification: ControllerCommand - Starting
2017-10-28 23:48:12.253209 [0x0000000808c07400]  [  DEBUG] Notification Type_Unknown received
2017-10-28 23:48:12.253 Detail, 
2017-10-28 23:48:12.253 Info, contrlr, Sending (Command) message (Callback ID=0x80, Expected Reply=0x61) - ControllerCommand_RemoveFailedNode: 0x01, 0x05, 0x00, 0x61, 0x10, 0x80, 0x0b
2017-10-28 23:48:12.255 Detail, contrlr,   Received: 0x01, 0x04, 0x01, 0x61, 0x08, 0x93
2017-10-28 23:48:12.255 Detail, 
2017-10-28 23:48:12.255 Warning, contrlr, WARNING: Received reply to FUNC_ID_ZW_REMOVE_FAILED_NODE_ID - Node not found
2017-10-28 23:48:12.255 Detail, Node147,   Expected callbackId was received
2017-10-28 23:48:12.255 Detail, Node147,   Message transaction complete
2017-10-28 23:48:12.255 Detail, 
2017-10-28 23:48:12.255 Detail, contrlr, Removing current message
2017-10-28 23:48:12.255 Detail, Notification: ControllerCommand - Failed
2017-10-28 23:48:12.255251 [0x0000000808c07400]  [  DEBUG] Notification Type_Unknown received
2017-10-28 23:48:12.255313 [0x0000000808c07400]  [  ERROR] controller state update: command has failed
2017-10-28 23:48:12.255417 [0x0000000808c07400]  [  ERROR] Controller error: Not Found
```

Where does Node147 come from?
0x93 == 147?

This may or may not be due to weird controller. After I added a few more Z-Wave devices a week ago, and upgraded the firmware on the S2 controller to latest, I've had a bunch of issues with the network (all nodes suddenly find no neighbors, or only node 001, the controller). discovered this while trying to fix issues.

stromnet, 2017-10-29:
Unrelated or not.. now I've transfered to a Gen5 controller (see below for what I did), and wanted to tell the Gen5 controller to remove the S2 controller (now offline). Called RemoveNode through the same code as above, but with 1 this time (the old controller ID):

```
2017-10-29 17:33:18.629 Detail, Node001, Queuing (Controller) Remove Failed Node                                       
2017-10-29 17:33:18.629 Info, ControllerCommand_RemoveFailedNode                                                       
2017-10-29 17:33:18.629 Detail, contrlr, Queuing (Command) ControllerCommand_RemoveFailedNode: 0x01, 0x05, 0x00, 0x61, 0x01, 0x9d, 0x07                                                                                                       
2017-10-29 17:33:18.629 Detail, Notification: ControllerCommand - Starting                                             
2017-10-29 17:33:18.629827 [0x0000000808c07400]  [  DEBUG] Notification Type_Unknown received                          
2017-10-29 17:33:18.629 Detail,                            
2017-10-29 17:33:18.629 Info, contrlr, Sending (Command) message (Callback ID=0x9d, Expected Reply=0x61) - ControllerCommand_RemoveFailedNode: 0x01, 0x05, 0x00, 0x61, 0x01, 0x9d, 0x07                                                       
2017-10-29 17:33:18.639 Detail, contrlr,   Received: 0x01, 0x04, 0x01, 0x61, 0x00, 0x9b                                
2017-10-29 17:33:18.639 Detail,                            
2017-10-29 17:33:18.639 Info, contrlr, Received reply to FUNC_ID_ZW_REMOVE_FAILED_NODE_ID - Command in progress        
2017-10-29 17:33:18.639 Detail, Node155,   Expected reply was received                                                 
2017-10-29 17:33:18.639 Detail, Notification: ControllerCommand - InProgress
2017-10-29 17:33:20.634 Error, contrlr, ERROR: Dropping command, expected response not received after 1 attempt(s)
2017-10-29 17:33:20.634 Detail, contrlr, Removing current message
2017-10-29 17:33:20.634 Detail, contrlr, Notification: Notification - TimeOut
2017-10-29 17:33:20.634 Detail, Notification: ControllerCommand - Error - Failed
2017-10-29 17:33:21.463 Detail,   Received: 0x01, 0x05, 0x00, 0x61, 0x9d, 0x01, 0x07
2017-10-29 17:33:21.463 Detail, 
Process 44448 stopped
* thread #6: tid = 103567, 0x0000000800ba76cf libopenzwave.so.1.4`OpenZWave::Driver::HandleRemoveFailedNodeRequest(this=0x0000000808cc1c00, _data="") + 207 at Driver.cpp:3409, stop reason = signal SIGSEGV: invalid address (fault address: 
0x2a)
    frame #0: 0x0000000800ba76cf libopenzwave.so.1.4`OpenZWave::Driver::HandleRemoveFailedNodeRequest(this=0x0000000808cc1c00, _data="") + 207 at Driver.cpp:3409
   3406                 }
   3407                 case FAILED_NODE_REMOVED:
   3408                 {
-> 3409                         Log::Write( LogLevel_Info, nodeId, "Received reply to FUNC_ID_ZW_REMOVE_FAILED_NODE_ID - node %d successfully moved to failed nodes list", m_currentControllerCommand->m_controllerCommandNode );
   3410                         state = ControllerState_Completed;
   3411
   3412                         {
(lldb) print m_currentControllerCommand
(OpenZWave::Driver::ControllerCommandItem *) $1 = 0x0000000000000000
(lldb) bt
* thread #6: tid = 103567, 0x0000000800ba76cf libopenzwave.so.1.4`OpenZWave::Driver::HandleRemoveFailedNodeRequest(this=0x0000000808cc1c00, _data="") + 207 at Driver.cpp:3409, stop reason = signal SIGSEGV: invalid address (fault address: 0x2a)
  * frame #0: 0x0000000800ba76cf libopenzwave.so.1.4`OpenZWave::Driver::HandleRemoveFailedNodeRequest(this=0x0000000808cc1c00, _data="") + 207 at Driver.cpp:3409
    frame #1: 0x0000000800ba1d2d libopenzwave.so.1.4`OpenZWave::Driver::ProcessMsg(this=0x0000000808cc1c00, _data="") + 5437 at Driver.cpp:2261
    frame #2: 0x0000000800b9795b libopenzwave.so.1.4`OpenZWave::Driver::ReadMsg(this=0x0000000808cc1c00) + 1883 at Driver.cpp:1729
    frame #3: 0x0000000800b95893 libopenzwave.so.1.4`OpenZWave::Driver::DriverThreadProc(this=0x0000000808cc1c00, _exitEvent=0x0000000808c9fd00) + 1491 at Driver.cpp:480
    frame #4: 0x0000000800b952b3 libopenzwave.so.1.4`OpenZWave::Driver::DriverThreadEntryPoint(_exitEvent=0x0000000808c9fd00, _context=0x0000000808cc1c00) + 51 at Driver.cpp:383
    frame #5: 0x0000000800b1c829 libopenzwave.so.1.4`OpenZWave::ThreadImpl::Run(this=0x0000000808da14c0) + 41 at ThreadImpl.cpp:176
    frame #6: 0x0000000800b1c72d libopenzwave.so.1.4`OpenZWave::ThreadImpl::ThreadProc(_pArg=0x0000000808da14c0) + 29 at ThreadImpl.cpp:161
```

After restarting the application, node 1 is gone (as desired).

----
My story on transferring to Aeon Z-Stick Gen5 controller.
# Plug Gen5 into other computer, in my case a Mac
# Running Windows 8.1 in Virtual Box, open OpenZWave PC Controller program from "zensys-tools.zip" available from misc sources online (yay...).
# Use above program to put this new controller in "Learn mode"
# Put my Z2 stick into "AddDevice" mode via `Manager::AddNode()`
# Wait a bit, and the windows app showed my whole network.
# Back to program controlling Z2-stick, call `Manager::TransferPrimaryRole()`
At this stage it seemed to do a lot of RemoveValue etc, and eventually I got another crash, but I think that was due to my application which did `notification->GetNotification()`, for a `Notification::Type_DriverReset`.
This triggered assert in `Notification::GetNotification()` because m_type was Type_DriverReset.
After restarting, it seemed to work as intended at least.

Summing up: with the old S2 controller, almost all neighbours got into some weird mode where they would get a single neighbour only: the controller. And in some cases not even that..
Not sure if this was due to the firmware update or something else, but now with that controller is replaced (Gen5 instead), all nodes can see almost all other nodes as neighbours. Hopefully network will work properly now..

Fishwaldo, 2017-12-04:
This is just a Logging Error. Nothing in the code is affected.

closed 2017-12-04 by Fishwaldo

nathanmfast, 2020-12-15:
I don't think this is just a logging error. I've experienced an identical problem but the node does not go away. The weird node in my case is Node 147 (when trying to remove node 37).