OpenZWave/open-zwave #1361: No rule to make target ... MultiInstanceAssociation.h

danreg, 2017-10-30:
Building open-zwave after git pull, will throw this error on my rpi3

> pi@raspberrypi:~/open-zwave-read-only $ make -j 3
LDFLAGS="" CPPFLAGS="" make -C /home/pi/open-zwave-read-only/cpp/build/ - -j --jobserver-fds=3,4
make[1]: Entering directory '/home/pi/open-zwave-read-only/cpp/build'
Building OpenZWave Version 1.4-2784-g1dba140-dirty
make[1]: *** No rule to make target '/home/pi/open-zwave-read-only/cpp/src/command_classes/MultiInstanceAssociation.h', needed by '/home/pi/open-zwave-read-only/.lib/CommandClasses.o'.  Stop.
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/pi/open-zwave-read-only/cpp/build'
Makefile:20: recipe for target 'all' failed
make: *** [all] Error 2

robsonke, 2017-11-06:
Same problem here, also on a rpi3.

nechry, 2017-11-07:
Please clean your local folder, this file is no longer part of the openzwave library
https://github.com/OpenZWave/open-zwave/commit/e8de6b196109e88af06455366ea8f647df0f7904

closed 2017-11-07 by nechry