OpenZWave/open-zwave #1368: Fibaro Windows Sensor 2

eumats, 2017-11-08:
A status update from Fibaro Windows Sensor 2 is only display the information in “Access_Control_1” (22=open, 23=closed) not in “Sensor_1” (true, false) as I expected from my other sensors like “Vision Security Garage Sensor”.
Can you please have a look?!

chilicheech, 2017-11-08:
Just a note (unrelated to your issue) - It looks like the Access_Control has the same issue as my Vision Door/Window sensor where it reports non-standard values. Mine also reports 22=open and 23=closed. The Alarm Command Class in the Dev branch doesn't support these values.

I filed an issue to have this addressed: #1364

nechry, 2017-11-08:
The Alarm CC in dev branch is not completed, the current progress for this CC in dev branch are perform before get the Sigma SDK. The Alarm CC must be replaced with Notification CC instead.

chilicheech, 2017-11-09:
Ah, got it. So are you guys already playing with the Sigma SDK to create the Notification CC?

@nechry, what can we do to get these sensors to work in the meantime? My sensor supports Alarm CC V1. Is there a way for me to force use Alarm CC V1 instead of Alarm/Notification CC V4?

nechry, 2017-11-09:
For the moment, with my gateway, I simply use the values 22 and 23 to determine the state of the sensor. I have to look at how to go back directly in the Sensor value

nechry, 2017-11-09:
Yes the advancements in the CC Alarm had stopped following the provision of the SDK, the current code in dev branch requires changes. I do not have any idea of delays on this work. The CC Alarm and Notification are both bulky in subtleties.

chilicheech, 2017-11-10:
@nechry do we need to do something like this in the config file locally to use the values 22 and 23? https://github.com/OpenZWave/open-zwave/blob/master/config/everspringct/hsm02.xml#L33-L57

chilicheech, 2017-11-10:
I was able to get mine to work by updating the config file with the following:
```
			<CommandClass id="113" name="COMMAND_CLASS_ALARM" version="1" request_flags="6" innif="true">
				<Instance index="1" />
				<Value type="list" genre="user" instance="1" index="9" label="Access Control" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" vindex="0" size="0">
					<Item label="Access Control - Reed Open" value="22" />
					<Item label="Access Control - Reed Closed" value="23" />
				</Value>
			</CommandClass>
```
Should I send a PR to update the config file to add the above? Here's the current config file: https://github.com/OpenZWave/open-zwave/blob/master/config/vision/zd2105us5.xml

nechry, 2017-11-11:
We must don't mix config file from Master branch to dev branch. In dev branch for config we work on a new updating system.

This patch is only a workaround, not a solution. from my point of view I prefer test a numeric value instead of a string

Fishwaldo, 2017-12-12:
those values are valid for the Alarm CC Version 1. Newer versions of this commandclass (now called Notification) have "predefined" values for various events.

If this device supports the Alarm CC higher than Version 1, then you should use the "predefined" values from the Notification Class rather than these 2 values (as they are vendor specific and not standardized across Z-Wave). 

As @nechry pointed out though, the Notification CC in Dev branch needs to be refactored again. I started working on it before the Z-Wave Specs were released. Now that they are released, I've realized how "complicated" the notification CC is (possibly one of the most complex CC in Z-Wave now!)

So, I'm going to close this Issue as it should be fixed when the I get to that.

closed 2017-12-12 by Fishwaldo