OpenZWave/open-zwave #1369: Aeotec Nano Dimmer - Parameter 85 & 86 - wrong values

rjblake, 2017-11-08:
Looking at the configuration values in the control panel, Parameter 85 is showing "-1728047104", while Parameter 86 is showing "1507328". According to the documentation, the following is valid (DWord Size is 4):

> State appointment 1: Set the ON time of output load.
> Value1 = 0, disable or =non zero, enable (day, bit0 - bit6 represent Mon to Sun).
> Value2 = ON (hour)
> Value3 = ON (minute)
> Value4 = ON (brightness level) 
> e.g. Value1=00, Value2=18 , Value3=00, Value4=99

Simply trying to enter a value of "0", is not accepted and documentation seems to indicate the following as valid values: (Min: 0 Max: 16843197), where:

0 to 99 is the brightness level
256 to 316 is the minute value of ON time
65536 to 65560 is the hour value of ON time
16777215 to 16777222 is the day value (Mon to Sun)

"0" is not accepted in the OZW control panel, but "16843040" for example is. In order for this not to apply, one needs to be able to enter a value of "0"

nechry, 2017-11-08:
I did not play with these two parameters when I validate the module. can you provide a PR?

cincuranet, 2017-11-25:
So, Aeotec is claiming the `0x120063`/`1179747` and `0x00170000`/`1507328` should be the only values to disable the parameter (because there's `0` in days). Sadly my dimmer does not accept these. 

So I'm pushing for more info. :(

cincuranet, 2017-12-01:
At the end Aeotec recommend to reset the device and/or exclude and include. Looks like nobody there knows what value should/shouldn't be used there.

nechry, 2017-12-04:
I will check again with the device soon.

rjblake, 2017-12-04:
Ok, so I have had some comms with Aeotec on this topic in review of their documentation and have the following updates (as I understand):

State appointment 1: 
Set the ON time of output load.
Value1 = 0, disable or =non zero, enable (day, bit0 - bit6 represent Mon to Sun). 
Value2 = ON (hour) 
Value3 = ON (minute) 
Value4 = ON (brightness level)	Value1=00

In HEX, the dimmer is expecting to receive these values as follows (0xAABBCCDD), where:

AA=Value1 
BB=Value2
CC=Value3
DD=Value4

HEX should be 0x(value1)(value2)(value3)(value4)

where Value1:

0 = disable days
1 = Monday
2 = Tuesday
and so on to value 7.

nechry, 2017-12-04:
for value1 (Value1 = 0, disable or =non zero, enable (day, bit0 - bit6 represent Mon to Sun).)
0 = disable days
1 = Monday
2 = Tuesday
4 = Wednesday
8 = Thursday
16 = Friday
32 = Saturday
64 = Sunday

rjblake, 2017-12-22:
I'm still confused by this. According to the Aeotec documentation and information, to disable this, you could/should enter a value such as 0x00000000 (00=Disable Days, 00=Hour, 00=Minutes, 00=Brightness), or even 0x00070010 (00=Disable Days, 07=Hour, 00=Minutes, 10=Brightness) and this functionality should be disabled. However, if I convert this HEX to Decimal, it is either 0 or 458768 (or I guess 0 or 16843025 if applying bitmasks 16777216+65543+256+10. I never was good at dealing with these conversions...Any ideas, as I think I've got my lights turning on at random and since the 'built in clock' for all the dimmers does not synchronise correctly with the controller they are all at different days/times

nechry, 2017-12-26:
try to exclude and include your device or perform a device full reset to clean parameters. For my point of view their kind of feature must be delegate to the gateway it self not to the device.

rjblake, 2017-12-27:
I'd rather set it correctly than reset as I have a number of scripts running on the devices and having to open 9x wall switches to reset is not really practical. Appreciate you view on "For my point of view their kind of feature must be delegate to the gateway it self not to the device."; but the reality is that Aeotec have included the functionality and it cannot be ignored or set incorrectly or will turn on at random. Given that their devices include a time/clock functionality that synchronises with the controller, however not with OpenZwave; this would in some cases prove useful. The 'default' values that OpenZwave use for the devices do not seem to relate to any of the values as shared (e.g. my devices default to 2566920192 and 8960 for Parameters 85 & 86 respectively - neither of which seem to relate to any values as per the engineering specs.

nechry, 2018-01-09:
Yes I agree with you, the feature is present and you should get to work. The default values ​​are indicative we try to be just right in the realization of the parameter files. However, after inclusion, you must ask the node to refresh the parameter values. In this way you ask the real values ​​in memory of the module. The values ​​in the configuration file do not affect the operation of the module and are not applied to the module.

Your problem here seems to me to be out openwave, either the module does not handle this feature well or you set it wrong. But in any case it is in the firmware of the module, not in the openzwave library that this work is done.

I have this module, if I have time, I reconnect it to see how mine reacts and what values ​​it has the inclusion, then following a complete reset. It is not impossible that the values ​​were not the original one on your module. That's why I'm asking you, at least for a module, to do an exclusion with reset. If you want to understand and isolate the problem.

rjblake, 2018-04-04:
I reset a number of nodes and the 'default' values showing for parameters 85&86 are '1179747' and '1507328' respectively. According the the Aeotec Engineering documentation, the default for parameter 85 (0x(value1)(value2)(value3)(value4) or 0x 00 18 00 99 (1573017 decimal), so I cannot even reconcile the values to what OpenZwave is showing. Also noticed that the zw111.xml file has entries parameters 83&84 which do not exist for the device

nechry, 2018-04-04:
The default values are for informations, we try get accurate with Engineering Spec, I found several differences between the documentation and the values following a reset.
Is the latest file
[Aeon Labs ZW111 Nano Dimmer Engineering Spec.pdf](https://github.com/OpenZWave/open-zwave/files/1875762/Aeon.Labs.ZW111.Nano.Dimmer.Engineering.Spec.pdf)
parameters 85 => disable 18h00 99, 00180099 => 0x00 0x12 0x00 0x63 => 1179747.
parameters 86 => disable 23h00 00 00230000 => 0x00 0x17 0x00 0x00 => 1507328.
like @cincuranet already reported.

I will change the default values according to this report. 
I previously made 2 mistakes I reversed the MSB / LSB and use hex for value instead of dec 2566920192 =>99001800

closed 2018-04-04 by nechry

rjblake, 2018-04-04:
@nechry - when you do change the default, try set it to something else (e.g. 16843007) and then change it back to 1179747 - you will find it is not accepted and cannot be changed back to 1179747

nechry, 2018-04-04:
I will check soon I reconnect the dimmer. I have lot todo before. But if you set the value 1 to 0 for disabling the feature, the fallback value is 0 for value 1? Others value can be something else if the value 1 is at 0 it's ok, for certain point of view.

commit ab5fe966 (2018-04-04): Aeotec Nano Dimmer update default values

fix #1369