OpenZWave/open-zwave #1396: FIBARO FGT-001 - The Heat Controller Radiator Thermostat

lcervo, 2017-12-13:
I bought 2 new Fibaro radiator valves (https://www.fibaro.com/en/products/the-heat-controller/), their "code" is FGT-001 (https://manuals.fibaro.com/content/manu ... T-v1.0.pdf)  but I don't see any xml for this devices. Is it possible adding the support for those new valves?

Ta.

nechry, 2017-12-15:
this device require Thermostat Mode ManufacturerSpecific in dev branch https://github.com/OpenZWave/open-zwave/pull/1386 patch to fully operate.

closed 2017-12-15 by nechry

NigelRook, 2018-01-17:
The dev branch seems to be way behind and lacks fgt001.xml and various other related changes. Is there a process to get this integrated into master?

chilicheech, 2018-01-17:
Or how do we get the the device configs merged into the dev branch from master?

rzr, 2019-01-22:
It looks like dev branch is still not merged in master:

https://github.com/OpenZWave/open-zwave/blob/master/cpp/src/command_classes/ThermostatSetpoint.cpp

https://github.com/OpenZWave/open-zwave/pull/1386/commits/47dcee16719eb0338342c372233fa86ce584b56e

commit 6e9be3c4 (2017-12-15): Add FIBARO FGT-001 The Heat Controller Radiator

base on initial @rocket13011 work.
note require CC ThermostatMode V3 to fully operate.
Close #1396