OpenZWave/open-zwave #1397: Documentation links pointing to Pepper1 DB are broken

veleek, 2017-12-15:
It looks like the Pepper1 Z-Wave product DB is no more.  Digging through some links on their site, there is the [Z-Wave Alliance Product Catalog](https://products.z-wavealliance.org/) that appears to have plenty of info.  I've never used Pepper1 so I can't say if it's the same sort of stuff.

I'd be willing to go through and update the documentation if this seems like a reasonable location to point people to.

https://products.z-wavealliance.org/

Fishwaldo, 2017-12-15:
Yeah - that would be very much appreciated!

veleek, 2017-12-17:
Okay, I've made the changes to the wiki but there's no way to make a pull request against the wiki repository...  What's the easiest way for me to do this?  I can create a repo in my own account and you can merge it yourself?

For the changes to the source (mostly just removing all the references that are broken in config xml files) I submitted PR #1400.

veleek, 2018-01-05:
Any thoughts on the best way to update the wiki?

nechry, 2018-01-05:
I just updated the wiki

veleek, 2018-01-08:
Hey @nechry.  I had some additional cleanup and examples updates that I made to the pages.  Did you just make a change to remove those links or was there something else you added?

nechry, 2018-01-08:
Hey @veleek I just remove and remplace links. The wiki cannot be merged like others projet files. You update your fork wiki?

veleek, 2018-01-08:
Yeah, the wiki's can't be merged using regular Pull Requests, but you could manually add my fork as an remote and then merge from there.

I have not yet uploaded my changes anywhere.  I'll see about getting them pushed to my fork's wiki tonight.

nechry, 2018-01-08:
let me know when it's over, I'll validate and merge

veleek, 2018-01-09:
Okay, the change (only a single commit) has been published to the wiki
here: https://github.com/veleek/open-zwave/wiki

Take a look and see if you think the changes are helpful.

nechry, 2018-01-09:
you do changes only in https://github.com/veleek/open-zwave/wiki/Adding-Devices page ?

nechry, 2018-01-09:
ok I perform the changes base on your local wiki. many thank's @veleek

closed 2018-01-09 by nechry

Fishwaldo, 2018-01-09:
I've just given @veleek collaborator access so you can update the wiki directly.

veleek, 2018-01-09:
Thanks @Fishwaldo.

PR #1400 (2018-01-22): Removing references to pepper1
Resolves #1397 

The Pepper1 DB no longer exists.  This removes a bunch of broken links from various configuration files.  And removes support for the Pepper1 metadata item (which was used by one product only).