OpenZWave/open-zwave #1398: Cannot find endpoint map to instance for Command Class COMMAND_CLASS_APPLICATION_STATUS endpoint 0

sjabby, 2017-12-15:
I re-compiled openzwave for use with homeassistant today based on this: https://github.com/OpenZWave/open-zwave/pull/1341 and using it together with the new Fibaro FGT-001

If i change the temperature on the device the change is reflected in openzwave, but if i change the set temperature trough openzwave i can see that the device receives a change( it changes the indicator light) but there is an error in the log:

```
2017-12-15 22:54:21.084 Info, Node026, Sending (Send) message (Callback ID=0x57, Expected Reply=0x13) - MultiChannel Encapsulated (instance=1): ThermostatSetpointCmd_Set (Node=26): 0x01, 0x10, 0x00, 0x13, 0x1a, 0x09, 0x60, 0x0d, 0x01, 0x01, 0x43, 0x01, 0x01, 0x01, 0x16, 0x25, 0x57, 0xa4
2017-12-15 22:54:21.090 Detail, Node026,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2017-12-15 22:54:21.091 Detail, Node026,   ZW_SEND_DATA delivered to Z-Wave stack
2017-12-15 22:54:22.311 Detail, Node026,   Received: 0x01, 0x07, 0x00, 0x13, 0x57, 0x00, 0x00, 0x7a, 0xc6
2017-12-15 22:54:22.312 Detail, Node026,   ZW_SEND_DATA Request with callback ID 0x57 received (expected 0x57)
2017-12-15 22:54:22.313 Info, Node026, Request RTT 1229 Average Request RTT 1244
2017-12-15 22:54:22.314 Detail,   Expected callbackId was received
2017-12-15 22:54:22.315 Detail,   Expected reply was received
2017-12-15 22:54:22.315 Detail,   Message transaction complete
2017-12-15 22:54:22.316 Detail, 
2017-12-15 22:54:22.316 Detail, Node026, Removing current message
2017-12-15 22:54:22.317 Detail, 
2017-12-15 22:54:22.318 Info, Node026, Sending (Send) message (Callback ID=0x58, Expected Reply=0x04) - MultiChannel Encapsulated (instance=1): ThermostatSetpointCmd_Get (Node=26): 0x01, 0x0e, 0x00, 0x13, 0x1a, 0x07, 0x60, 0x0d, 0x01, 0x01, 0x43, 0x02, 0x01, 0x25, 0x58, 0xaf
2017-12-15 22:54:22.325 Detail, Node026,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2017-12-15 22:54:22.326 Detail, Node026,   ZW_SEND_DATA delivered to Z-Wave stack
2017-12-15 22:54:22.346 Detail, Node026,   Received: 0x01, 0x07, 0x00, 0x13, 0x58, 0x00, 0x00, 0x03, 0xb0
2017-12-15 22:54:22.347 Detail, Node026,   ZW_SEND_DATA Request with callback ID 0x58 received (expected 0x58)
2017-12-15 22:54:22.347 Info, Node026, Request RTT 28 Average Request RTT 636
2017-12-15 22:54:22.348 Detail,   Expected callbackId was received
2017-12-15 22:54:22.365 Detail, Node026,   Received: 0x01, 0x0e, 0x00, 0x04, 0x00, 0x1a, 0x08, 0x60, 0x0d, 0x00, 0x01, 0x22, 0x01, 0x01, 0x02, 0xab
2017-12-15 22:54:22.365 Detail, 
2017-12-15 22:54:22.365 Info, Node026, Response RTT 45 Average Response RTT 665
2017-12-15 22:54:22.365 Error, Node026, Cannot find endpoint map to instance for Command Class COMMAND_CLASS_APPLICATION_STATUS endpoint 0
2017-12-15 22:54:22.365 Detail, Node026,   Expected reply and command class was received
2017-12-15 22:54:22.365 Detail, Node026,   Message transaction complete
2017-12-15 22:54:22.365 Detail, 
2017-12-15 22:54:22.365 Detail, Node026, Removing current message
```

And the temperature or mode is then changed back to the way it was before initiating the change from ozw. 

Any toughts?

tobiasgraf, 2017-12-18:
If you compile without this PR, does it work then?

sjabby, 2017-12-18:
I tried compiling it with 6e9be3c  and now the command times out instead:

```
2017-12-18 22:00:50.734 Info, Node027, Value::Set - COMMAND_CLASS_THERMOSTAT_SETPOINT - Heating - 1 - 1 - 22
2017-12-18 22:00:50.735 Detail, Node027, Setting Encryption Flag on Message For Command Class COMMAND_CLASS_THERMOSTAT_SETPOINT
2017-12-18 22:00:50.736 Detail, Node027, Queuing (Send) ThermostatSetpointCmd_Set (Node=27): 0x01, 0x0c, 0x00, 0x13, 0x1b, 0x05, 0x43, 0x01, 0x01, 0x01, 0x16, 0x25, 0x82, 0x0d
2017-12-18 22:00:50.737 Detail, Node027, Setting Encryption Flag on Message For Command Class COMMAND_CLASS_THERMOSTAT_SETPOINT
2017-12-18 22:00:50.737 Detail, Node027, Queuing (Send) ThermostatSetpointCmd_Get (Node=27): 0x01, 0x0a, 0x00, 0x13, 0x1b, 0x03, 0x43, 0x02, 0x01, 0x25, 0x83, 0x18
2017-12-18 22:00:50.738 Detail, 
2017-12-18 22:00:50.738 Info, Node027, Processing (Send) Nonce Request message (Callback ID=0x82, Expected Reply=0x13)
2017-12-18 22:00:50.738 Info, Node027, Sending (Send) message (Callback ID=0x82, Expected Reply=0x13) - Nonce_Get(ThermostatSetpointCmd_Set) - 0x01, 0x09, 0x00, 0x13, 0x1b, 0x02, 0x98, 0x40, 0x05, 0x02:
2017-12-18 22:00:50.744 Detail, Node027,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2017-12-18 22:00:50.746 Detail, Node027,   ZW_SEND_DATA delivered to Z-Wave stack
2017-12-18 22:00:52.019 Detail, Node027,   Received: 0x01, 0x07, 0x00, 0x13, 0x02, 0x00, 0x00, 0x80, 0x69
2017-12-18 22:00:52.021 Detail, Node027,   ZW_SEND_DATA Request with callback ID 0x02 received (expected 0x02)
2017-12-18 22:00:52.022 Info, Node027, Request RTT 1284 Average Request RTT 758
2017-12-18 22:00:52.040 Detail, Node027,   Received: 0x01, 0x10, 0x00, 0x04, 0x00, 0x1b, 0x0a, 0x98, 0x80, 0xd7, 0x14, 0x44, 0x21, 0x7a, 0x6d, 0x60, 0x12, 0x21
2017-12-18 22:00:52.042 Info, Node027, Received SecurityCmd_NonceReport from node 27
2017-12-18 22:00:52.043 Info, Node027, Sending (Send) message (Callback ID=0x84, Expected Reply=0x13) - ThermostatSetpointCmd_Set (Node=27): 0x01, 0x0c, 0x00, 0x13, 0x1b, 0x05, 0x43, 0x01, 0x01, 0x01, 0x16, 0x25, 0x84, 0x0b
2017-12-18 22:00:52.053 Detail, Node027,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2017-12-18 22:00:52.054 Detail, Node027,   ZW_SEND_DATA delivered to Z-Wave stack
2017-12-18 22:00:52.077 Detail, Node027,   Received: 0x01, 0x07, 0x00, 0x13, 0x84, 0x00, 0x00, 0x03, 0x6c
2017-12-18 22:00:52.078 Detail, Node027,   ZW_SEND_DATA Request with callback ID 0x84 received (expected 0x84)
2017-12-18 22:00:52.079 Info, Node027, Request RTT 1341 Average Request RTT 1049
2017-12-18 22:00:52.080 Detail,   Expected callbackId was received
2017-12-18 22:00:52.081 Detail,   Expected reply was received
2017-12-18 22:00:52.082 Detail,   Message transaction complete
2017-12-18 22:00:52.083 Detail, 
2017-12-18 22:00:52.083 Detail, Node027, Removing current message
2017-12-18 22:00:52.085 Detail, 
2017-12-18 22:00:52.085 Info, Node027, Processing (Send) Nonce Request message (Callback ID=0x83, Expected Reply=0x04)
2017-12-18 22:00:52.086 Info, Node027, Sending (Send) message (Callback ID=0x83, Expected Reply=0x04) - Nonce_Get(ThermostatSetpointCmd_Get) - 0x01, 0x09, 0x00, 0x13, 0x1b, 0x02, 0x98, 0x40, 0x05, 0x02:
2017-12-18 22:00:52.093 Detail, Node027,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2017-12-18 22:00:52.094 Detail, Node027,   ZW_SEND_DATA delivered to Z-Wave stack
2017-12-18 22:00:52.113 Detail, Node027,   Received: 0x01, 0x07, 0x00, 0x13, 0x02, 0x00, 0x00, 0x02, 0xeb
2017-12-18 22:00:52.114 Detail, Node027,   ZW_SEND_DATA Request with callback ID 0x02 received (expected 0x02)
2017-12-18 22:00:52.115 Info, Node027, Request RTT 27 Average Request RTT 538
2017-12-18 22:00:52.135 Detail, Node027,   Received: 0x01, 0x10, 0x00, 0x04, 0x00, 0x1b, 0x0a, 0x98, 0x80, 0x62, 0x89, 0xc4, 0xf4, 0xcd, 0x4a, 0x34, 0x87, 0x0d
2017-12-18 22:00:52.136 Info, Node027, Received SecurityCmd_NonceReport from node 27
2017-12-18 22:00:52.137 Info, Node027, Sending (Send) message (Callback ID=0x85, Expected Reply=0x04) - ThermostatSetpointCmd_Get (Node=27): 0x01, 0x0a, 0x00, 0x13, 0x1b, 0x03, 0x43, 0x02, 0x01, 0x25, 0x85, 0x1e
2017-12-18 22:00:52.147 Detail, Node027,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2017-12-18 22:00:52.148 Detail, Node027,   ZW_SEND_DATA delivered to Z-Wave stack
2017-12-18 22:00:52.170 Detail, Node027,   Received: 0x01, 0x07, 0x00, 0x13, 0x85, 0x00, 0x00, 0x03, 0x6d
2017-12-18 22:00:52.172 Detail, Node027,   ZW_SEND_DATA Request with callback ID 0x85 received (expected 0x85)
2017-12-18 22:00:52.173 Info, Node027, Request RTT 85 Average Request RTT 311
2017-12-18 22:00:52.174 Detail,   Expected callbackId was received
2017-12-18 22:00:52.191 Detail, Node027,   Received: 0x01, 0x08, 0x00, 0x04, 0x00, 0x1b, 0x02, 0x98, 0x40, 0x32
2017-12-18 22:00:52.192 Info, Node027, Received SecurityCmd_NonceGet from node 27
2017-12-18 22:00:52.193 Info, NONCES: 0x93, 0x28, 0x5a, 0x45, 0xd3, 0x19, 0x85, 0x3a
2017-12-18 22:00:52.194 Info, NONCES: 0x9c, 0x36, 0x29, 0x76, 0x19, 0xc3, 0xc2, 0x9b
2017-12-18 22:00:52.195 Info, NONCES: 0x1b, 0xf3, 0x4b, 0xd7, 0x03, 0xc8, 0x27, 0x9f
2017-12-18 22:00:52.196 Info, NONCES: 0x6b, 0x2d, 0x06, 0xb1, 0x39, 0xa2, 0x52, 0x8c
2017-12-18 22:00:52.197 Info, NONCES: 0x35, 0x68, 0xcb, 0xd3, 0xca, 0x6f, 0xd5, 0xc9
2017-12-18 22:00:52.198 Info, NONCES: 0x62, 0xa6, 0xfb, 0x9b, 0x48, 0x4e, 0x27, 0xbf
2017-12-18 22:00:52.199 Info, NONCES: 0x2b, 0x42, 0xd5, 0xf5, 0xb2, 0xab, 0xbf, 0x40
[... cut ...]
2017-12-18 22:01:29.499 Info, Raw: 0x98, 0x81, 0x4c, 0x20, 0x11, 0x00, 0x67, 0x00, 0xf5, 0x6d, 0x02, 0x0f, 0x2f, 0xba, 0xc1, 0xcf, 0x42, 0x39, 0x93, 0x7d, 0x2d, 0x43, 0xff, 0x14, 0xfb
2017-12-18 22:01:29.500 Detail, Node027, Decrypted Packet: 0x00, 0x22, 0x01, 0x01, 0x02
2017-12-18 22:01:29.502 Detail, 
2017-12-18 22:01:29.503 Info, Node027, Response RTT 163 Average Response RTT 213
2017-12-18 22:01:39.339 Error, Node027, ERROR: Dropping command, expected response not received after 1 attempt(s)
2017-12-18 22:01:39.340 Detail, Node027, Removing current message
2017-12-18 22:01:39.340 Detail, Node027, Notification: Notification - TimeOut
```

sjabby, 2017-12-18:
Maybe @nechry have some pointers?

sjabby, 2017-12-18:
If i change the setting direcly on the thermostat the settings gets read back to ozw without any issue.

nechry, 2017-12-19:
@sjabby I don't have the device. The second test the device was include with Security, can be result as the TimeOut?

sjabby, 2017-12-19:
@nechry I included with security on all my attempts, i can try without if that should make a difference.
Also if i set the temperature from ozw (Home Assistant) the valve adjusts correctly but it gets the timeout in the log and the value is not updated. However if i force a refresh node the adjusted temperature is then reported back to ozw.  strange things.

nechry, 2017-12-19:
@sjabby ok, the required polling after a setpoint was also reported by the jeedom team they actually do the initial configuration. 
Maybe it would be interesting to do a TriggerRefreshValue on the CC COMMAND_CLASS_APPLICATION_STATUS in order to automatically refresh the SetPoint

sjabby, 2017-12-19:
@nechry How would i do That ? :o

nechry, 2017-12-19:
more or less like this
<CommandClass id="34">
		<TriggerRefreshValue Genre="user" Index="0" Instance="1">
			<RefreshClassValue CommandClass="64" RequestFlags="0" Index="0" Instance="1" />
			<RefreshClassValue CommandClass="67" RequestFlags="0" Index="1" Instance="1" />
		</TriggerRefreshValue>
</CommandClass>

but again untested

sjabby, 2017-12-19:
Please excuse me, but what file should this be added to? the FGT-001.xml ?

nechry, 2017-12-19:
yes exactly l

nechry, 2017-12-19:
Also the FGT-001.xml require the https://github.com/OpenZWave/open-zwave/pull/1386 patch to work

sjabby, 2017-12-19:
@nechry does it matter where in the file i add it?

nechry, 2017-12-19:
Before closing Product element see other usage like this one:
https://github.com/OpenZWave/open-zwave/blob/master/config/horstmann/asrzw.xml

Fishwaldo, 2018-01-22:
Ping! Still a issue?

sjabby, 2018-01-22:
@fishwaldo Yes. I tried compiling the pr manually and also the git dev but same result it doesn’t work as expected.

Fishwaldo, 2018-01-22:
Need full logs from a fresh inclusion then to see what's going on.

sjabby, 2018-02-11:
`2018-02-11 00:01:31.416 Info, Node035,   Basic device class    (0x04) - Routing Slave
2018-02-11 00:01:31.417 Info, Node035,   Generic device Class  (0x08) - Thermostat
2018-02-11 00:01:31.417 Info, Node035,   Specific device class (0x06) - General Thermostat V2
2018-02-11 00:01:31.417 Info, Node035,     COMMAND_CLASS_BASIC will be mapped to COMMAND_CLASS_THERMOSTAT_MODE
2018-02-11 00:01:31.417 Info, Node035,   Mandatory Command Classes for Node 35:
2018-02-11 00:01:31.418 Info, Node035,     COMMAND_CLASS_BASIC
2018-02-11 00:01:31.418 Info, Node035,     COMMAND_CLASS_THERMOSTAT_MODE
2018-02-11 00:01:31.418 Info, Node035,     COMMAND_CLASS_THERMOSTAT_SETPOINT
2018-02-11 00:01:31.418 Info, Node035,     COMMAND_CLASS_MANUFACTURER_SPECIFIC
2018-02-11 00:01:31.418 Info, Node035,     COMMAND_CLASS_VERSION
2018-02-11 00:01:31.419 Info, Node035,   Mandatory Command Classes controlled by Node 35:
2018-02-11 00:01:31.419 Info, Node035,     None
2018-02-11 00:01:31.419 Info, Node035,   Optional command classes for node 35:
2018-02-11 00:01:31.419 Info, Node035,     COMMAND_CLASS_ZWAVE_PLUS_INFO
2018-02-11 00:01:31.420 Info, Node035,     COMMAND_CLASS_VERSION (Existing)
2018-02-11 00:01:31.420 Info, Node035,     COMMAND_CLASS_MULTI_CHANNEL_ASSOCIATION
2018-02-11 00:01:31.420 Info, Node035,     COMMAND_CLASS_SENSOR_MULTILEVEL
2018-02-11 00:01:31.420 Info, Node035,     COMMAND_CLASS_THERMOSTAT_MODE (Existing)
2018-02-11 00:01:31.420 Info, Node035,     COMMAND_CLASS_THERMOSTAT_SETPOINT (Existing)
2018-02-11 00:01:31.421 Info, Node035,   CommandClass 0x53 - NOT REQUIRED
2018-02-11 00:01:31.421 Info, Node035,     COMMAND_CLASS_SECURITY
2018-02-11 00:01:31.421 Info, Node035,   CommandClass 0x9f - NOT REQUIRED
2018-02-11 00:01:31.421 Info, Node035,   CommandClass 0x55 - NOT REQUIRED
2018-02-11 00:01:31.421 Info, Node035,     COMMAND_CLASS_CRC_16_ENCAP
2018-02-11 00:01:31.422 Info, Node035,   CommandClass 0x59 - NOT REQUIRED
2018-02-11 00:01:31.422 Info, Node035,     COMMAND_CLASS_DEVICE_RESET_LOCALLY
2018-02-11 00:01:31.422 Info, Node035,     COMMAND_CLASS_MULTI_INSTANCE/CHANNEL
2018-02-11 00:01:31.422 Info, Node035,   CommandClass 0x6c - NOT REQUIRED
2018-02-11 00:01:31.422 Info, Node035,     COMMAND_CLASS_ALARM
2018-02-11 00:01:31.423 Info, Node035,     COMMAND_CLASS_MANUFACTURER_SPECIFIC (Existing)
2018-02-11 00:01:31.423 Info, Node035,     COMMAND_CLASS_PROTECTION
2018-02-11 00:01:31.423 Info, Node035,     COMMAND_CLASS_BATTERY
2018-02-11 00:01:31.423 Info, Node035,     COMMAND_CLASS_CONFIGURATION
2018-02-11 00:01:31.424 Info, Node035,     COMMAND_CLASS_CLOCK
2018-02-11 00:01:31.424 Info, Node035,   CommandClass 0x7a - NOT REQUIRED
2018-02-11 00:01:31.424 Info, Node035,     COMMAND_CLASS_POWERLEVEL
2018-02-11 00:01:31.424 Info, Node035,     COMMAND_CLASS_APPLICATION_STATUS
2018-02-11 00:01:31.424 Info, Node035,     COMMAND_CLASS_ASSOCIATION
2018-02-11 00:01:31.425 Detail, Node035, AdvanceQueries queryPending=0 queryRetries=0 queryStage=None live=1
2018-02-11 00:01:31.425 Detail, Node035, QueryStage_ProtocolInfo
2018-02-11 00:01:31.426 Detail, Node035, Queuing (Query) Get Node Protocol Info (Node=35): 0x01, 0x04, 0x00, 0x41, 0x23, 0x99
2018-02-11 00:01:31.426 Detail, Node035, Queuing (Query) Query Stage Complete (ProtocolInfo)
2018-02-11 00:01:31.426 Info, Node035, Initializing Node. New Node: true (true)
2018-02-11 00:01:31.428 Detail, Node035, Notification: NodeAdded
2018-02-11 00:01:31.464 Detail, Node035, Notification: NodeProtocolInfo
2018-02-11 00:01:31.467 Detail, Node035, Notification: ValueAdded
2018-02-11 00:01:31.472 Detail, Node035, Notification: ValueAdded
2018-02-11 00:01:31.476 Detail, Node035, Notification: ValueAdded
2018-02-11 00:01:31.483 Detail, Node035, Notification: ValueAdded
2018-02-11 00:01:31.487 Detail, Node035, Notification: ValueAdded
2018-02-11 00:01:31.497 Detail, Node035, Notification: ValueAdded
2018-02-11 00:01:31.506 Detail, Node035, Notification: ValueAdded
2018-02-11 00:01:31.513 Detail, Node035, Notification: ValueAdded
2018-02-11 00:01:31.522 Detail, Node035, Notification: ValueAdded
2018-02-11 00:01:31.532 Detail, Node035, Notification: ValueAdded
2018-02-11 00:01:31.537 Detail, Node035, Notification: ValueAdded
2018-02-11 00:01:31.544 Detail, Node035, Notification: ValueAdded
2018-02-11 00:01:31.556 Detail, Node035, Notification: ValueAdded
2018-02-11 00:01:31.584 Detail, Node035, Notification: ValueAdded
2018-02-11 00:01:31.593 Detail, Node035, Notification: ValueAdded
2018-02-11 00:01:31.602 Detail, Node035, Notification: ValueAdded
2018-02-11 00:01:31.611 Detail, Node035, Notification: ValueAdded
2018-02-11 00:01:31.617 Detail, Node035, Notification: ValueAdded
2018-02-11 00:01:31.636 Detail, Node035, Notification: ValueAdded
2018-02-11 00:01:31.641 Detail, Node035, Notification: ValueAdded
2018-02-11 00:01:31.651 Detail, Node035, Notification: ValueAdded
2018-02-11 00:01:31.660 Detail, Node035, Notification: ValueAdded
2018-02-11 00:01:31.668 Detail, Node035, Notification: ValueAdded
2018-02-11 00:01:31.691 Detail, Node035, Notification: ValueAdded
2018-02-11 00:01:31.701 Info, Node035, Sending (Query) message (Callback ID=0x00, Expected Reply=0x41) - Get Node Protocol Info (Node=35): 0x01, 0x04, 0x00, 0x41, 0x23, 0x99
[... 907 lines cut ...]
2018-02-11 00:22:20.288 Detail, Node035,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2018-02-11 00:22:20.289 Detail, Node035,   ZW_SEND_DATA delivered to Z-Wave stack
[... cut ...]
2018-02-11 00:22:21.884 Detail, Node035,   Message transaction complete
2018-02-11 00:22:21.885 Detail, Node035, Removing current message
`

Full log from inclusion to trying to run a few changes on it.
If i change the temp setting on the valve itself the new value is shown in home assistant and in the OZW log. But if i change it from home assistant i get this error:
`Error, Node035, Cannot find endpoint map to instance for Command Class COMMAND_CLASS_APPLICATION_STATUS endpoint 0`

Its also not updating the current temperature from the external sensor.

Fishwaldo, 2018-02-19:
That doesn't look like a inclusion log. (it loads from cache). And Grepping the log file doesn't give me all the info needed. Can you try again?

But at a high level, it looks like after setting the SetPoint, the device is reporting back its "busy" via the APPLICATION_STATUS CC. I've never seen that before. 

(the endpoint map message is a error, but it wouldnt fix the issue your seeing....)

Fishwaldo, 2018-03-07:
ping

Fishwaldo, 2018-10-26:
Closing due to lack of feedback

closed 2018-10-26 by Fishwaldo