OpenZWave/open-zwave #1401: Fibaro Heat Controller fgt001 not reading tempreture

fr3sh, 2017-12-17:
Fibaro Heat Controller fgt001 not reading temperature from additional temperature sensor. 
The Heat Controller has additional temperature sensor with bluetooth connection where is reading temperature from. Only initial temperature is mesured after that temperature is not changing at all.

This is from Fibaro forum:
Additional temperature sensor is communicating with The Heat Controller every 10minutes, and The Heat Controller sends those information every 2-3hours to gateway or when the temperature differs by 0.5-1 degree from last report.

tobsch, 2018-01-05:
@fr3sh & @nechry Does this help?
https://github.com/OpenZWave/open-zwave/commit/026ddbae016eb7627deceae51063b5c9aceff9dd

How's the release policy of openzwave? I'd like to see support in homeassistant soon... :)
Many thanks!

baddar, 2018-01-05:
I have a question to you guys. Are you getting any temperature from the actual thermostat?

tobsch, 2018-01-06:
as far as i heard this is not exposed from the thermostats but only if you have the external sensor.
correct me if i am wrong.

NigelRook, 2018-01-16:
I've been tinkering through home assistant, and I've managed to persuade it to provide the once-per-ten-minute updates through polling on the current temperature entity if that helps at all.

(using head of master)

tobsch, 2018-01-30:
@NigelRook is your change included somewhere?

tobsch, 2018-02-06:
Okay, I know use the thermostat for a while and have exactly the same problem. If I turn on polling globally, it does not solve. @NigelRook could you help out with details?

Question: I read that it's sometimes an issue that devices need to be polled. Is this simply a hardware issue or is that an issue with the openzwave implementation?

tobsch, 2018-02-07:
@nechry @fr3sh or @baddar do you know what is wrong or did you just return the thermostats?
The combination of "half integrated" devices with the strange color code idea that fibaro had there really drives me (and even more my girlfriend) nuts.
Enabling polling globally seems to lead to a batter drain of thermostats (?). Charging a thermostat every few days is not an option :-) It also did not change a thing in temperature reportings...

The strange thing is that it seems to not work on its own without any software. It does not hold a temperature that you just set via the device itself...

nechry, 2018-02-07:
from what I know for the moment, Fibaro announced that the firmware was buggy and that it require an update. this update will be possible only for owners of Fibaro HomeCenter box as it does not share OTA updates, or with a RMA (you have to disconnect and send back the device... not so fun). 

you can find other details on the fibaro forum.
https://forum.fibaro.com/index.php?/topic/28622-heat-controller-stop-regulating/

I actually think it's better not to rush on the module at the moment.

Fishwaldo, 2018-03-07:
Closing - This is a product bug.....

closed 2018-03-07 by Fishwaldo