OpenZWave/open-zwave #1402: Fibaro FGT-001 Heat Controller: Corrected xml file

MrPozor, 2017-12-17:
Hi,

The current fibaro/fgt-001.xml contained spelling mistakes and two bitfields were  implemented as lists. I have changed the type to int. Tested OK on my side.

MrPozor
[fgt001.xml.txt](https://github.com/OpenZWave/open-zwave/files/1565902/fgt001.xml.txt)

nechry, 2017-12-17:
I keep list for last one but with all options

nechry, 2017-12-17:
thank's @MrPozor

closed 2017-12-17 by nechry

MrPozor, 2017-12-17:
Ah great. The list is indeed much more user-friendly, thanks.

nechry, 2017-12-17:
@MrPozor do you have the device? It's seem not all features work as expected. I don't have from my side

MrPozor, 2017-12-18:
@nechry, yes I have the device, though not with the optional temperature sensor.
The device works as far as I could see. I can set the temperature with the setpoint command class. Configuration seems to work as well as I can read back the configuration that I send.
There is no reporting whatsoever (temperature, valve position etc.) but this might be by design.

Let me know if I can help with testing a specific feature.

nechry, 2017-12-18:
Ok thanks, so if you adjust with setpoint cc you have to perform a refresh to get the back the setpoint or he report by itself?

commit 026ddbae (2017-12-17): typo fibre fgt001

close #1402