OpenZWave/open-zwave #1414: zwcfg xml displays incorrect max_baud_rate

ben423423n32j14e, 2018-01-07:
Example (secure included Aeotec SmartSwitch 6 Z-Wave Plus device) with UZB1 controller running on Domoticz:

<Node id="33" name="Fish Tank Pump" location="" basic="4" generic="16" specific="1" roletype="5" devicetype="1792" nodetype="0" type="Binary Power Switch" listening="true" frequentListening="false" beaming="true" routing="true" max_baud_rate="40000" version="4" secured="true" refreshonnodeinfoframe="false" query_stage="Complete">

All devices report baud rate of 40000

Either this is a bug in the reporting or the network is actually running at baud 40000.

Many people asking why devices are not displayed as Z-Wave Plus speeds on the Home Assistant forums as well.

Would be good to either get this resolved or definitively clarified from a member of the team.

jwillaz, 2018-01-07:
@Fishwaldo Would you be able to enlighten us regarding how baud rate is being set?  What device data is being read to set it? I see this in **node.cpp**:

> m_maxBaudRate = 9600;
	if( ( _data[0] & 0x38 ) == 0x10 )
	{
		m_maxBaudRate = 40000;
	}

ben423423n32j14e, 2018-01-07:
Unfortunately I don’t have the knowledge required to be able to answer your question (the above was discovered using a Domoticz installation) and the same issue appears to exist in Home Assistant based on a number of forum threads I read.

Even if the issue does not get resolved here, are you able to advise what the impact of baud being set in the zwcfg xml has on network performance?

Is the network actually running at 40000? If I change the value for the controller to 100000 does that now make it use 100kbps rate if available?

I have a large almost entirely Z-Wave + network but I'm seeing some delays and wondering if it's really performing at + speeds?

Fishwaldo, 2018-01-22:
This is information only and changing any setting in zwcfg_*.xml doesn't actually affect the network. Someone has to reverse engineer the serial api around that to figure out what the new encoding is for 100K.

ben423423n32j14e, 2018-01-22:
Would be nice if Sigma Designs stepped in to assist... Considering how almost every open source home automation project is depending on OpenZWave (for Z-Wave support), it's in their interest to provide the best customer experience. They should be providing this info to the team directly...

Fishwaldo, 2018-01-23:
@ben423423n32j14e So I should say they have, and continue to engage with us. They have made a decision not to release the specs of the SerialAPI though, and have a alternative way to interface with a USB Stick via their own simple stack. Unfortuantly adopting their stack is a major re-write of OZW, so we are stuck.

Who knows, in future they may change their mind about releasing the SerialAPI specs. That would close the last mile of OZW and our reverse engineering efforts :)

AlCalzone, 2018-03-09:
> have a alternative way to interface with a USB Stick via their own simple stack

Just out of curiosity, where could one find and read up on that alternative way?

rccoleman, 2019-03-11:
So if I'm understanding correctly, a network controlled by a Z-Wave Plus controller like an Aeotec Z-Stick, which should support 100K, will use the max rate available, even if OZW reports a lower rate?  In the end, it seems like the inclusion process is between the controller and the devices, with OZW just interpreting the results.  I do have a few battery-operated Z-Wave locks and I'm just trying to confirm that I'm not hindering my Z-Wave network performance in some way.

Fishwaldo, 2019-03-13:
This is cosmetic only. The device baud rate will use the best available rate and there is no way (that I’m aware of) to change the baud rates.

rccoleman, 2019-03-13:
Excellent, thanks very much!

petergebruers, 2019-04-30:
Always wondered this myself and after going through the docs I would say the situation is like this... For older devices, non-Z-Wave Plus so based on series 300 SoC, the decoding of the protocol field and translating it to 9600 or 40000 is correct. When Sigma introduced series 500 SoC and Z-Wave Plus those devices gained a channel and can do the legacy speeds plus 100 kBits per second. Then new series 700 does not add new transmission speeds, so the same rule still applies.

So I guess this code could be update with the logic "if it is Z-Wave Plus then show max speed 100000".

On the other hand, routing is dynamic and tries different speeds, so knowing the maximum speed is a bit academic.

But "Dev" branch recently got a new and much more interesting function. If your controller is based on a recent SDK, it can report "extended status". This means it can tell you the actual speed used, per packet. Like this:

`Node011, Extended TxStatus: Time: 20, Hops: 0, Rssi:  -55 --- --- --- ---, AckChannel: 0, TxChannel: 0, RouteScheme: Last Working Route, Route: 0 0 0 0, RouteSpeed: 100K, RouteTries: 1, FailedLinkFrom: 0, FailedLinkTo: 0`

It tells you this transmission took 20 ms (on the controller), uses a direct connection and 100 kBit speed...

Fishwaldo, 2019-05-22:
Done

closed 2019-05-22 by Fishwaldo