OpenZWave/open-zwave #1416: pkgconfigdir in the Makefile

oandrew, 2018-01-08:
Was there a reason to check the existence of the `$(instlibdir)/pkgconfig` before installing in that path?  
This breaks the expected behavior because in  case that path doesn't exist it will take a value from `pkg-config --variable pc_path pkg-config` which is not necessarily in the `$PREFIX`. 
https://github.com/OpenZWave/open-zwave/blob/master/cpp/build/support.mk#L103-L108

I think replacing it with 
```bash
pkgconfigdir ?= "$(instlibdir)/pkgconfig"
``` 
should be sufficient for most cases.

The reason I'm asking is that this prevents this formula from being updated on Homebrew.
https://github.com/Homebrew/homebrew-core/pull/22486#discussion_r160018571

ffbenedetti, 2019-07-08:
@Fishwaldo any progress on this investigation?
This is blocking any attempt to update the brew open-zwave formula to install the latest 1.6 version.
https://github.com/Homebrew/homebrew-core/pull/22486/
Currently, on MacOS the brew install open-zwave is useless because is stuck in installing version 1.2

Fishwaldo, 2019-07-10:
>Was there a reason to check the existence of the $(instlibdir)/pkgconfig before installing in that path?

If the path doesn't exist, its kinda hard to install to it... plus - it would indicate that pkg-config is not installed..... 

This one is complicated - It seems a lot of distributions do things differently. 

if homebrew requires a different setup, please open a PR with the brew specific changes in a config block.

Fishwaldo, 2019-07-11:
Closed as per 1769. Use the pkgconfigdir override to place the PC file where makes sense for your platform.

closed 2019-07-11 by Fishwaldo