OpenZWave/open-zwave #1417: Cannot setValue for SensorMultilevel classes

tzim-fr, 2018-01-09:
Hi. 

I'm trying to send Temperatures Values to a Radiator Thermostat Valve (Eurotronic Spirit which accept reports from an external sensor). 
I declared the value as read/write in the device config.
When I try to send a value (with ozwcp or from nodejs script), I get this error : 

```
2018-01-09 12:01:14.269 Info, Node008, Value::Set - COMMAND_CLASS_SENSOR_MULTILEVEL - Temperature - 1 - 1 - 21.44
SetValue string failed type=decimal
post: key=8-SENSOR MULTILEVEL-user-decimal-1-1 data= size=0
```

I'd like to forward data from non-zwave sensors...

Thanks for help.

nechry, 2018-01-16:
sensors values are reay-only you can't change this behavior with a config file.

tzim-fr, 2018-01-16:
The device accept temperature reports from external temperature sensors. I'd like to send reports from the controller itself (if possible). Maybe I want to do it the wrong way ?

nechry, 2018-01-16:
I think so yes.

closed 2018-01-18 by nechry

namelessone667, 2019-03-18:
would it be possible to reopen this issue (or should i create a new one?). According to user manual: Spirit Z-Wave Plus can receive
temperature readings from other Z-Wave devices
(wall thermostat for example) The external temperature can be used for temperature regulation.
This feature has to be enabled via configuration
parameter. 

Also if you check the Zwave device database here https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/710 
the SENSOR_MULTILEVEL command class has there two entries:
Current Temperature
External Temperature

So setting the external temperature from the controller should be possible

Fishwaldo, 2019-03-18:
SENSOR_MULTILEVEL is used to report sensors. It can’t accept a value from another device.

namelessone667, 2019-03-18:
Well, according to documentation it can. Also it is working in OpenHAB implementation, see here: https://github.com/openhab/org.openhab.binding.zwave/blob/master/doc/eurotronic/spirit_0_0.md

Fishwaldo, 2019-03-18:
OZW does not implement any “node” functionality, only features to control other nodes via a Controller. To implement this would require a lot of core changes to happen. 

And as states many times, we will not be implementing this as it’s not our intention to replicate the Z-Wave SDK for nodes.

namelessone667, 2019-03-18:
Thank you for your comment, however I dont think you need to implement any "node" functionality for this to work (as this device only suports association with a controller). I think all you is needed for this to work is to implement the Set Value method for the SENSOR_MULTILEVEL  command class.

Fishwaldo, 2019-03-18:
Please read the last line of the README. 

You also have to advertise CC’s in the controllers NIF, respond to associations SET messages etc. 

I will NOT be implementing Node functionality. Your free to fork and implement it if you require this functionality.