OpenZWave/open-zwave #1419: Value Jumps, Energy Meters

ado464, 2018-01-11:
Hello, i have a ton of Aeotec Energy Meters Gen5, 3p3, 3p4, latest firmware and latest openzwave.

This is the configuration
Group 1 : 10
Group 2: 1
Group 3: 4

Group 1 time: 60
Group 2 time: 600
Group 3 time: 120

And then i do a restart of openzwave and the values come up like this:
Value1: Watt
Value2: kWh
Value3: Volt
Value4: Ampere

It runs okay like this for a while, and then all of a sudden maybe a month later the values would swap like:
Value1: Watt
Value2: kWh & Voltage (switching back and forth)
Value3: kvAh? (was not even set in config)
Value4: Ampere

Has anyone else had this issue? Driving me insane. I need to "Refresh node info" on 1 meter like 10-15 times and then restart openzwave and maybe it would fall into place correctly

best regards,
Ado

RicardP, 2018-01-15:
I see this happen more or less often on different nodes. 

If selecting the erratic Node in OZW Control Panel and use function "Refresh Node Info" the fields seems restored, for a while...

Fishwaldo, 2018-02-19:
need to get some logs from around the time the values switch...

ado464, 2018-04-06:
[zwave-info.txt](https://github.com/OpenZWave/open-zwave/files/1883871/zwave-info.txt)

Hey Fishwaldo!
Hope this little log can help out, there is some report of kvAh and some kWh that are actually voltage values of 228 etc

Sorry for late reply, it's been hectic

best regards,
Ado

ado464, 2018-05-21:
Hey guys, any info or would you need another dump? 

best regards,
Ado