OpenZWave/open-zwave #1421: Support for Neo CoolCam RC01Z

chriskmn, 2018-01-20:
Hi @all,

I just got the Neo CoolCam SOS Remote Key-Fob RC01Z. It works in my Z-Wave network, but due to the fact that there is no XML config file at github, it is hard to configure. The z-wave readings tell:

model:     0x0258 0x0003 0x108a | 2018-01-20 12:05:40
modelId:  0258-0003-108a

Is anyone able to cerate an XML file for that device ?

Thanks,

  Chris

chriskmn, 2018-01-20:
And this is the little info from the manual:

Advanced Configuration

1.	Basic Set Level

Basic Set Command will be sent where contains a value when SOS is triggered if group 2 have associated a device such as siren, the receiver will take it for consideration; for instance, if associated siren is received the Basic Set Command (0xFF), the siren will take alarm on.

0~99 are mapping to Basic Set value (0 ~ 99). 100 is mapping to Basic Set Value 0xFF.
Parameter_Number	Size	Available Settings	Default			
              1	          1	    0 ~ 100         	100
			

2.	SOS Event Clear Time
This parameter defines the time to clear emergency event after emergency event is triggered when parameter #3 is set to ‘2’.
This parameter is set to ‘0’, the clearing emergency event will not occurred always.

The default value of this parameter is set to ‘60’, means that the emergency event will be cleared after 60 seconds when emergency event is triggered.

Parameter_Number	Size	Available Settings	Default
             2	                  2	  0 , 30 ~ 32767	60
			

3.	SOS Configuration

This parameter defines the SOS key function, the SOS key can be configured as emergency event key or central scene notification key.
If this parameter is set to ‘1’, the SOS key will be configured as central scene notification key.

If this parameter is set to ‘2’, the SOS key will be configured as emergency event notification key. The default settings is ‘2’, the SOS key is configured as emergency event notification key.

Parameter_Number	Size	Available Settings	Default		
             3	                  1	           1, 2                   2
			

Command Classes
This Sensor supports Command Classes as Below:

*	COMMAND_CLASS_ZWAVEPLUS_INFO (V2)
*	COMMAND_CLASS_VERSION (V2)
*	COMMAND_CLASS_MANUFACTURER_SPECIFIC (V2)
*	COMMAND_CLASS_DEVICE_RESET_LOCALLY (V1)
*	COMMAND_CLASS_POWERLEVEL (V1)
*	COMMAND_CLASS_BATTERY (V1)
*	COMMAND_CLASS_ASSOCIATION (V2)
*	COMMAND_CLASS_ASSOCIATION_GRP_INFO (V1)
*	COMMAND_CLASS_WAKE_UP (V2)
*	COMMAND_CLASS_NOTIFICATION (V8)
*	COMMAND_CLASS_SENSOR_BINARY (V2)
*	COMMAND_CLASS_CONFIGURATION (V1)

nechry, 2018-01-20:
Do you have the association group definition ? It's almost done

closed 2018-01-20 by nechry

chriskmn, 2018-01-20:
cool, sounds great !

The manual only provides this info in Asoc. Groups:

`Equipment linkage
When SOS is operated, all devices that are linked to it will receive the relevant report. SOS through the Z-Wave master device or alarm host can control other Z_wave network equipment, such as linkage jacks, alarm equipment, dimmers and so on.
SOS supports two groups of linkage:
The first group of linkage: you can control the Z-Wave master settings or alarm host status
The second group linkage: switch or activate the Z-Wave master device or the alarm host already has the active scene, so the other equipment in the empty ear scene.

Note: SOS is in direct contact with other devices, they are directly through the Z-Wave network communication, Z-Wave alarm host or Z-Wave master device is not involved.
`

nechry, 2018-01-20:
I found this manuel with association groups definition
http://adresat.kiev.ua/sites/default/files/neo_coolcam_z_wave_siren_alarm_manual_full_eng.pdf
Your have 2 or 5 groups ?

chriskmn, 2018-01-20:
only two groups

nechry, 2018-01-20:
oki so I must fix, I found another doc with 2, the first is for basic set and the 2 is for lifeline.

ca you give me the max node allowed by groups ?

chriskmn, 2018-01-20:
Well, I don't know, but all other devices from NEO have max. 5 nodes in their groups. I would estimate that it is the same here

chriskmn, 2018-01-20:
this is the manual I got from the seller:

[http://spaceegg.servegame.com:81/Neo%20Coolcam%20SOS%20RC01Z.doc](url)

And the remote looks like this:

[https://de.aliexpress.com/item/Z-wave-Elderly-Guardian-Senior-Guard-SOS-Panic-Button-Medic-Alert-Home-Safety-Alarm-System/32826202457.html](url)

chriskmn, 2018-01-20:
I think the manual you have is the non-z-wave version (RC01T). Like an RF-device or so.

nechry, 2018-01-20:
ok so I'm really not sure about group name. with our do the 2 is not the lifeline

chriskmn, 2018-01-20:
I will try to ask the seller for more specification

nechry, 2018-01-20:
let try like this 
Lifeline si master settings or alarm host status.
And according to param 1 the grp 2 is the Basic Set

chriskmn, 2018-01-20:
Yes that could make sense.

I requested more info in china. Let's see.

Thanks for your effort :-)

commit 162884a1 (2018-01-20): Add Coolcam SOS Remote Key-Fob RC01Z

Close #1421