OpenZWave/open-zwave #1425: New Device: 2GIG Technologies CT32 Thermostat

ve6rah, 2018-01-25:
I'm not exactly a coder, so I'm going to leave this to someone who knows what they're doing, but there should be enough information here to make it easy.

I have a 2GIG Technologies CT32 Theromstat, currently manufacturer_specific.xml has several 2GIG thermostats, but not the CT32 (closest is the CT30). By default this thermostat just can't be set with ozw, however I managed to get it working by modifying manufacturer_specific.xml to add the following line:
` <Product type="2002" id="0100" name="CT32 Thermostat" config="2gig/ct32.xml"/>`
under:
`<Manufacturer id="0098" name="2GIG Technologies">`
and then copying 2gig/ct30.xml to 2gig/ct32.xml

Since doing that I can now control the thermostat properly.

nechry, 2018-01-26:
The CT32 config is different from CT30
https://products.z-wavealliance.org/products/1046/configs
Can you provide a specific config file will all parameters et perform a PR ?

ve6rah, 2018-01-26:
I'm afraid I have no idea how to convert that to a config file. All I know it's that by default it didn't work at all, I fed it the ct30 config file and I'm controlling the thermostat properly.
I looked at the open zwave docs and it said to open an issue here to have it added if you weren't comfortable doing a pull request (which I'm not) but had enough information to make it work (which I do)

nechry, 2018-01-29:
I create a specific config file, is more like the CT100 version

commit d1896485 (2018-01-29): Add 2GIG Technologies CT32 Thermostat

close #1425
closed 2018-01-29 by nechry