OpenZWave/open-zwave #1426: Aeon Z-stick S2 not resetting properly (home ID is not cleared)

stromnet, 2018-01-25:
Hi,
first of all, this is perhaps more suitable for the mailing list, but after my third attempt to send to it without the message going through either from googlegroups UI or when manually sending email (no mailer-daemon reply from googlegroups, not listing in web UI, nothing showing up in digest several days later), I'll give it a shot here while at the same time raising this concern about problems with sending to list.

---

A while back I did a "transfer primary role" from my S2 stick to a new Aeon Gen5 stick. This worked fine, and the Gen5 stick is now the only one in use.

After this I've tried to reset the S2 stick to use it a development/lab stick, but it seems resetting the device partially fails. All nodes disappear, but the home ID is not reset properly. Thus, using this with any node will conflict with my existing network.

Am I (or owzcp/open-zwave) missing something? If not, has anyone else seen something similar?

Thanks!
Johan

The reset is done using owzcp:

```
2018-01-23 14:13:18.199 Always, OpenZwave Version 1.4.2768 Starting Up
webserver starting port 8090
Warning: MHD_USE_THREAD_PER_CONNECTION must be used only with MHD_USE_INTERNAL_POLLING_THREAD. Flag MHD_USE_INTERNAL_POLLING_THREAD was added. Consider setting MHD_USE_INTERNAL_POLLING_THREAD explicitly.
post: key=dev data=/dev/cu.SLAB_USBtoUART size=22
post: key=fn data=open size=4
post: key=usb data=false size=5
2018-01-23 14:13:35.884 Info, Setting Up Provided Network Key for Secure Communications
2018-01-23 14:13:35.884 Warning, Failed - Network Key Not Set
2018-01-23 14:13:35.884 Info, mgr,     Added driver for controller /dev/cu.SLAB_USBtoUART
post: key=usb data= size=0
2018-01-23 14:13:35.885 Info,   Opening controller /dev/cu.SLAB_USBtoUART
2018-01-23 14:13:35.885 Info, Trying to open serial port /dev/cu.SLAB_USBtoUART (attempt 1)
2018-01-23 14:13:35.889 Info, Serial port /dev/cu.SLAB_USBtoUART opened (attempt 1)
2018-01-23 14:13:35.889 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
2018-01-23 14:13:35.889 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_MEMORY_GET_ID: 0x01, 0x03, 0x00, 0x20, 0xdc
2018-01-23 14:13:35.889 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES: 0x01, 0x03, 0x00, 0x05, 0xf9
2018-01-23 14:13:35.889 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_GET_CAPABILITIES: 0x01, 0x03, 0x00, 0x07, 0xfb
2018-01-23 14:13:35.889 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_SUC_NODE_ID: 0x01, 0x03, 0x00, 0x56, 0xaa
2018-01-23 14:13:35.890 Detail,
2018-01-23 14:13:35.890 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x15) - FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
2018-01-23 14:13:35.893 Detail, contrlr,   Received: 0x01, 0x10, 0x01, 0x15, 0x5a, 0x2d, 0x57, 0x61, 0x76, 0x65, 0x20, 0x33, 0x2e, 0x32, 0x38, 0x00, 0x01, 0x9f
2018-01-23 14:13:35.893 Detail,
2018-01-23 14:13:35.893 Info, contrlr, Received reply to FUNC_ID_ZW_GET_VERSION:
2018-01-23 14:13:35.893 Info, contrlr,     Static Controller library, version Z-Wave 3.28
2018-01-23 14:13:35.893 Detail, Node045,   Expected reply was received
2018-01-23 14:13:35.893 Detail, Node045,   Message transaction complete
2018-01-23 14:13:35.893 Detail,
2018-01-23 14:13:35.893 Detail, contrlr, Removing current message
2018-01-23 14:13:35.893 Detail,
2018-01-23 14:13:35.893 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x20) - FUNC_ID_ZW_MEMORY_GET_ID: 0x01, 0x03, 0x00, 0x20, 0xdc
2018-01-23 14:13:35.895 Detail, contrlr,   Received: 0x01, 0x08, 0x01, 0x20, 0x01, 0x61, 0xf3, 0x4d, 0x01, 0x09
2018-01-23 14:13:35.895 Detail,
2018-01-23 14:13:35.895 Info, contrlr, Received reply to FUNC_ID_ZW_MEMORY_GET_ID. Home ID = 0x0161f34d.  Our node ID = 1
2018-01-23 14:13:35.896 Detail, Node097,   Expected reply was received
2018-01-23 14:13:35.896 Detail, Node097,   Message transaction complete
2018-01-23 14:13:35.896 Detail,
2018-01-23 14:13:35.896 Detail, contrlr, Removing current message
2018-01-23 14:13:35.896 Detail,
2018-01-23 14:13:35.896 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x05) - FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES: 0x01, 0x03, 0x00, 0x05, 0xf9
2018-01-23 14:13:35.898 Detail, contrlr,   Received: 0x01, 0x04, 0x01, 0x05, 0x1c, 0xe3
2018-01-23 14:13:35.898 Detail,
2018-01-23 14:13:35.898 Info, contrlr, Received reply to FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES:
2018-01-23 14:13:35.898 Info, contrlr,     There is a SUC ID Server (SIS) in this network.
2018-01-23 14:13:35.898 Info, contrlr,     The PC controller is an inclusion static update controller (SUC) and was the original primary before the SIS was added.
2018-01-23 14:13:35.898 Detail, Node227,   Expected reply was received
2018-01-23 14:13:35.898 Detail, Node227,   Message transaction complete
2018-01-23 14:13:35.898 Detail,
2018-01-23 14:13:35.898 Detail, contrlr, Removing current message
2018-01-23 14:13:35.898 Detail,
2018-01-23 14:13:35.898 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x07) - FUNC_ID_SERIAL_API_GET_CAPABILITIES: 0x01, 0x03, 0x00, 0x07, 0xfb
2018-01-23 14:13:35.903 Detail, contrlr,   Received: 0x01, 0x2b, 0x01, 0x07, 0x03, 0x08, 0x00, 0x86, 0x00, 0x02, 0x00, 0x01, 0xfe, 0x80, 0xfe, 0x88, 0x0f, 0x00, 0x00, 0x00, 0xfb, 0x97, 0x3f, 0x82, 0x07, 0x00, 0x00, 0x80, 0x00, 0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x8d
2018-01-23 14:13:35.903 Detail,
2018-01-23 14:13:35.903 Info, contrlr,  Received reply to FUNC_ID_SERIAL_API_GET_CAPABILITIES
[... 184 lines cut ...]
2018-01-23 14:13:59.120 Info, Node001, Node Identity Codes: 0086:0002:0001
2018-01-23 14:13:59.120 Detail, Node001, QueryStage_Associations
2018-01-23 14:13:59.120 Detail, Node001, QueryStage_Neighbors
2018-01-23 14:13:59.120 Detail, contrlr, Requesting routing info (neighbor list) for Node 1
2018-01-23 14:13:59.120 Detail, Node001, Queuing (Command) Get Routing Info (Node=1): 0x01, 0x07, 0x00, 0x80, 0x01, 0x00, 0x00, 0x03, 0x7a
2018-01-23 14:13:59.120 Detail, Node001, Queuing (Query) Query Stage Complete (Neighbors)
2018-01-23 14:13:59.120 Detail,   Expected reply was received
2018-01-23 14:13:59.120 Detail,   Message transaction complete
2018-01-23 14:13:59.120 Detail,
2018-01-23 14:13:59.120 Detail, contrlr, Removing current message
2018-01-23 14:13:59.120 Detail, Node001, Notification: DriverReady
2018-01-23 14:13:59.120 Info, Notification: Driver Ready, homeId 0161f34d, nodeId 1
2018-01-23 14:13:59.120 Detail, Node001, Notification: NodeAdded
2018-01-23 14:13:59.120 Info, Notification: Node Added Home 0161f34d Node 1 Genre basic Class NO OPERATION Instance 0 Index 0 Type bool
2018-01-23 14:13:59.120 Detail, Node001, Notification: NodeProtocolInfo
2018-01-23 14:13:59.121 Info, Notification: Node Protocol Info Home 0161f34d Node 1 Genre basic Class NO OPERATION Instance 0 Index 0 Type bool
2018-01-23 14:13:59.121 Detail, Node001, Notification: EssentialNodeQueriesComplete
2018-01-23 14:13:59.121 Info, Notification: Essential Node 1 Queries Complete
2018-01-23 14:13:59.121 Detail, Node001, Notification: ValueAdded
2018-01-23 14:13:59.121 Info, Notification: Value Added Home 0x0161f34d Node 1 Genre basic Class BASIC Instance 1 Index 0 Type byte
2018-01-23 14:13:59.121 Detail, Node001, Notification: NodeNaming
2018-01-23 14:13:59.121 Info, Notification: Node Naming Home 0161f34d Node 1 Genre basic Class NO OPERATION Instance 0 Index 0 Type bool
....

```

Fishwaldo, 2018-02-19:
No Idea whats going on here. The reset command is acknowledged by the stick... so it should reset with a new HomeID. 

Maybe use one of the Aeotec tools to see if that makes any difference?

(I reset S2 sticks all the time with OZW so "works for me")

stromnet, 2018-02-19:
@Fishwaldo I've tried firmware upgrade on it (with no errors reported at least), but besides that I'm not aware of any Aeotec tools. Tried looking a bit on their site but cannot find much, any tips?

Fishwaldo, 2018-03-07:
The firmware tools have a option to perform a hard reset I believe. (one of the menu options).

stromnet, 2018-03-07:
I don't even have a menu bar.. :) 
[image]

Fishwaldo, 2018-10-26:
Closing as there isn't much we can do about this.

closed 2018-10-26 by Fishwaldo