OpenZWave/open-zwave #1433: Radio Thermostat CT80 RTZW-01 only reporting one sensor

chilicheech, 2018-01-31:
I have a Radio Thermostat (2gig) CT80 with an RTZW-01 zwave module. I use the Dev branch of open-zwave in home-assistant. Here's the config file for that device: https://github.com/OpenZWave/open-zwave/blob/Dev/config/2gig/ct80.xml

When I start up home-assistant the device gets queried and returns a multi-channel sensor. Instances 1 and 2 are temperature sensors and instance 3 is a relative humidity sensor. Upon initial query the device reports all sensor values. After the initial query only instance 1 sends reports. I found these other issues that seem to be related to this one: #960 and #966. Do I just need to update the config for this device and this to it?
```
<!-- Multi Channel Association Groups -->
<CommandClass id="142" ForceInstances="true" />
```
I can send a PR, I just need to confirm this is what needs to be done.

Here's some logs that I think might be relevant:
```
2018-01-30 18:45:52.138 Detail, Node028, Query Stage Complete (Session)
2018-01-30 18:45:52.138 Detail, Node028, AdvanceQueries queryPending=0 queryRetries=0 queryStage=Dynamic live=1
2018-01-30 18:45:52.138 Detail, Node028, QueryStage_Dynamic
2018-01-30 18:45:52.139 Detail, Node028, Queuing (Send) SensorMultilevelCmd_Get (Node=28): 0x01, 0x09, 0x00, 0x13, 0x1c, 0x02, 0x31, 0x04, 0x25, 0x70, 0x9b
2018-01-30 18:45:52.139 Detail, Node028, Queuing (Send) MultiChannel Encapsulated (instance=2): SensorMultilevelCmd_Get (Node=28): 0x01, 0x0d, 0x00, 0x13, 0x1c, 0x06, 0x60, 0x0d, 0x01, 0x01, 0x31, 0x04, 0x25, 0x71, 0xf7
2018-01-30 18:45:52.139 Detail, Node028, Queuing (Send) MultiChannel Encapsulated (instance=3): SensorMultilevelCmd_Get (Node=28): 0x01, 0x0d, 0x00, 0x13, 0x1c, 0x06, 0x60, 0x0d, 0x01, 0x02, 0x31, 0x04, 0x25, 0x72, 0xf7
2018-01-30 18:45:52.139 Detail, Node028, Queuing (Send) ThermostatModeCmd_Get (Node=28): 0x01, 0x09, 0x00, 0x13, 0x1c, 0x02, 0x40, 0x02, 0x25, 0x73, 0xef
2018-01-30 18:45:52.139 Detail, Node028, Queuing (Send) ThermostatOperatingStateCmd_Get (Node=28): 0x01, 0x09, 0x00, 0x13, 0x1c, 0x02, 0x42, 0x02, 0x25, 0x74, 0xea
2018-01-30 18:45:52.140 Detail, Node028, Queuing (Send) ThermostatFanModeCmd_Get (Node=28): 0x01, 0x09, 0x00, 0x13, 0x1c, 0x02, 0x44, 0x02, 0x25, 0x75, 0xed
2018-01-30 18:45:52.140 Detail, Node028, Queuing (Send) ThermostatFanStateCmd_Get (Node=28): 0x01, 0x09, 0x00, 0x13, 0x1c, 0x02, 0x45, 0x02, 0x25, 0x76, 0xef
2018-01-30 18:45:52.140 Detail, Node028, Queuing (Send) ClockCmd_Get (Node=28): 0x01, 0x09, 0x00, 0x13, 0x1c, 0x02, 0x81, 0x05, 0x25, 0x77, 0x2d
2018-01-30 18:45:52.140 Detail, Node028, Queuing (Send) IndicatorCmd_Get (Node=28): 0x01, 0x09, 0x00, 0x13, 0x1c, 0x02, 0x87, 0x02, 0x25, 0x78, 0x23
2018-01-30 18:45:52.140 Detail, Node028, Queuing (Query) Query Stage Complete (Dynamic)
2018-01-30 18:45:52.141 Detail,
2018-01-30 18:45:52.141 Info, Node028, Sending (Send) message (Callback ID=0x70, Expected Reply=0x04) - SensorMultilevelCmd_Get (Node=28): 0x01, 0x09, 0x00, 0x13, 0x1c, 0x02, 0x31, 0x04, 0x25, 0x70, 0x9b
2018-01-30 18:45:52.149 Detail, Node028,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2018-01-30 18:45:52.149 Detail, Node028,   ZW_SEND_DATA delivered to Z-Wave stack
2018-01-30 18:45:52.329 Detail, Node028,   Received: 0x01, 0x07, 0x00, 0x13, 0x70, 0x00, 0x00, 0x12, 0x89
2018-01-30 18:45:52.330 Detail, Node028,   ZW_SEND_DATA Request with callback ID 0x70 received (expected 0x70)
2018-01-30 18:45:52.330 Info, Node028, Request RTT 188 Average Request RTT 192
2018-01-30 18:45:52.330 Detail,   Expected callbackId was received
2018-01-30 18:45:52.505 Detail, Node028,   Received: 0x01, 0x0c, 0x00, 0x04, 0x00, 0x1c, 0x06, 0x31, 0x05, 0x01, 0x2a, 0x02, 0x7b, 0x8b
2018-01-30 18:45:52.506 Detail,
2018-01-30 18:45:52.506 Info, Node028, Response RTT 364 Average Response RTT 367
2018-01-30 18:45:52.506 Info, Node028, Received SensorMultiLevel report from node 28, instance 1, Temperature: value=63.5F
2018-01-30 18:45:52.506 Detail, Node028, Refreshed Value: old value=63.5, new value=63.5, type=decimal
2018-01-30 18:45:52.506 Detail, Node028, Changes to this value are not verified
2018-01-30 18:45:52.506 Detail, Node028,   Expected reply and command class was received
2018-01-30 18:45:52.507 Detail, Node028,   Message transaction complete
2018-01-30 18:45:52.507 Detail,
2018-01-30 18:45:52.507 Detail, Node028, Removing current message
2018-01-30 18:45:52.507 Detail, Node028, Notification: ValueChanged
2018-01-30 18:45:52.549 Detail,
2018-01-30 18:45:52.549 Info, Node028, Sending (Send) message (Callback ID=0x71, Expected Reply=0x04) - MultiChannel Encapsulated (instance=2): SensorMultilevelCmd_Get (Node=28): 0x01, 0x0d, 0x00, 0x13, 0x1c, 0x06, 0x60, 0x0d, 0x01, 0x01, 0x31, 0x04, 0x25, 0x71, 0xf7
2018-01-30 18:45:52.558 Detail, Node028,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2018-01-30 18:45:52.558 Detail, Node028,   ZW_SEND_DATA delivered to Z-Wave stack
2018-01-30 18:45:52.736 Detail, Node028,   Received: 0x01, 0x07, 0x00, 0x13, 0x71, 0x00, 0x00, 0x12, 0x88
2018-01-30 18:45:52.736 Detail, Node028,   ZW_SEND_DATA Request with callback ID 0x71 received (expected 0x71)
2018-01-30 18:45:52.736 Info, Node028, Request RTT 186 Average Request RTT 189
2018-01-30 18:45:52.737 Detail,   Expected callbackId was received
2018-01-30 18:45:52.920 Detail, Node028,   Received: 0x01, 0x10, 0x00, 0x04, 0x00, 0x1c, 0x0a, 0x60, 0x0d, 0x01, 0x01, 0x31, 0x05, 0x01, 0x2a, 0x02, 0x7b, 0xf6
2018-01-30 18:45:52.921 Detail,
2018-01-30 18:45:52.921 Info, Node028, Response RTT 371 Average Response RTT 369
2018-01-30 18:45:52.921 Info, Node028, Received a MultiChannelEncap from node 28, endpoint 1 for Command Class COMMAND_CLASS_SENSOR_MULTILEVEL
2018-01-30 18:45:52.921 Info, Node028, Received SensorMultiLevel report from node 28, instance 2, Temperature: value=63.5F
2018-01-30 18:45:52.921 Detail, Node028, Initial read of value
2018-01-30 18:45:52.921 Detail, Node028,   Expected reply and command class was received
2018-01-30 18:45:52.921 Detail, Node028,   Message transaction complete
2018-01-30 18:45:52.922 Detail,
2018-01-30 18:45:52.922 Detail, Node028, Removing current message
2018-01-30 18:45:52.922 Detail, Node028, Notification: ValueChanged
2018-01-30 18:45:52.945 Detail,
2018-01-30 18:45:52.946 Info, Node028, Sending (Send) message (Callback ID=0x72, Expected Reply=0x04) - MultiChannel Encapsulated (instance=3): SensorMultilevelCmd_Get (Node=28): 0x01, 0x0d, 0x00, 0x13, 0x1c, 0x06, 0x60, 0x0d, 0x01, 0x02, 0x31, 0x04, 0x25, 0x72, 0xf7
2018-01-30 18:45:52.955 Detail, Node028,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2018-01-30 18:45:52.955 Detail, Node028,   ZW_SEND_DATA delivered to Z-Wave stack
2018-01-30 18:45:53.146 Detail, Node028,   Received: 0x01, 0x07, 0x00, 0x13, 0x72, 0x00, 0x00, 0x13, 0x8a
2018-01-30 18:45:53.146 Detail, Node028,   ZW_SEND_DATA Request with callback ID 0x72 received (expected 0x72)
2018-01-30 18:45:53.147 Info, Node028, Request RTT 201 Average Request RTT 195
2018-01-30 18:45:53.147 Detail,   Expected callbackId was received
2018-01-30 18:45:53.328 Detail, Node028,   Received: 0x01, 0x0f, 0x00, 0x04, 0x00, 0x1c, 0x09, 0x60, 0x0d, 0x02, 0x01, 0x31, 0x05, 0x05, 0x01, 0x2d, 0x92
2018-01-30 18:45:53.329 Detail,
2018-01-30 18:45:53.329 Info, Node028, Response RTT 383 Average Response RTT 376
2018-01-30 18:45:53.329 Info, Node028, Received a MultiChannelEncap from node 28, endpoint 2 for Command Class COMMAND_CLASS_SENSOR_MULTILEVEL
2018-01-30 18:45:53.329 Info, Node028, Received SensorMultiLevel report from node 28, instance 3, Relative Humidity: value=45%
2018-01-30 18:45:53.329 Detail, Node028, Initial read of value
2018-01-30 18:45:53.330 Detail, Node028,   Expected reply and command class was received
2018-01-30 18:45:53.330 Detail, Node028,   Message transaction complete
2018-01-30 18:45:53.330 Detail,
2018-01-30 18:45:53.330 Detail, Node028, Removing current message
2018-01-30 18:45:53.330 Detail, Node028, Notification: ValueChanged
2018-01-30 18:45:53.353 Detail,
```

Cheers,

chilicheech, 2018-02-01:
I installed pkg-config via `apt-get install pkg-config` and it worked. This was previously working without installing pkg-config for version 0.4.0.35. Maybe we should document somewhere that we also need to install pkg-config beforehand? Maybe it's already documented and I missed it.

Cheers,

chilicheech, 2018-02-01:
I stopped home-assistant, removed the entry for Node028 in the ozwcache file, then restarted home-assistant. Here's some more logs:
```
2018-02-01 02:52:11.933 Detail, Node028, AdvanceQueries queryPending=0 queryRetries=0 queryStage=Instances live=1
2018-02-01 02:52:11.933 Detail, Node028, QueryStage_Instances
2018-02-01 02:52:11.933 Info, Node028, MultiChannelCmd_EndPointGet for node 28
2018-02-01 02:52:11.933 Detail, Node028, Queuing (Query) MultiChannelCmd_EndPointGet (Node=28): 0x01, 0x09, 0x00, 0x13, 0x1c, 0x02, 0x60, 0x07, 0x25, 0x66, 0xdf
2018-02-01 02:52:11.933 Detail, Node028, Queuing (Query) Query Stage Complete (Instances)
2018-02-01 02:52:11.934 Info, Node028, Sending (Query) message (Callback ID=0x66, Expected Reply=0x04) - MultiChannelCmd_EndPointGet (Node=28): 0x01, 0x09, 0x00, 0x13, 0x1c, 0x02, 0x60, 0x07, 0x25, 0x66, 0xdf
2018-02-01 02:52:11.942 Detail, Node028,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2018-02-01 02:52:11.942 Detail, Node028,   ZW_SEND_DATA delivered to Z-Wave stack
2018-02-01 02:52:12.147 Detail, Node028,   Received: 0x01, 0x07, 0x00, 0x13, 0x66, 0x00, 0x00, 0x14, 0x99
2018-02-01 02:52:12.147 Detail, Node028,   ZW_SEND_DATA Request with callback ID 0x66 received (expected 0x66)
2018-02-01 02:52:12.147 Info, Node028, Request RTT 214 Average Request RTT 212
2018-02-01 02:52:12.321 Detail, Node028,   Received: 0x01, 0x0a, 0x00, 0x04, 0x00, 0x1c, 0x04, 0x60, 0x08, 0x40, 0x02, 0xc3
2018-02-01 02:52:12.321 Info, Node028, Response RTT 388 Average Response RTT 386
2018-02-01 02:52:12.322 Info, Node028, Received MultiChannelEndPointReport from node 28. All 2 endpoints are the same.
2018-02-01 02:52:12.322 Info, Node028, MultiChannelCmd_CapabilityGet for endpoint 1
2018-02-01 02:52:12.322 Detail, Node028, Queuing (Send) MultiChannelCmd_CapabilityGet (Node=28): 0x01, 0x0a, 0x00, 0x13, 0x1c, 0x03, 0x60, 0x09, 0x01, 0x25, 0x67, 0xd3
2018-02-01 02:52:12.322 Detail, Node028,   Expected reply and command class was received
2018-02-01 02:52:12.322 Detail, Node028,   Message transaction complete
2018-02-01 02:52:12.322 Detail, Node028, Removing current message
2018-02-01 02:52:12.323 Info, Node028, Sending (Send) message (Callback ID=0x67, Expected Reply=0x04) - MultiChannelCmd_CapabilityGet (Node=28): 0x01, 0x0a, 0x00, 0x13, 0x1c, 0x03, 0x60, 0x09, 0x01, 0x25, 0x67, 0xd3
2018-02-01 02:52:12.331 Detail, Node028,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2018-02-01 02:52:12.331 Detail, Node028,   ZW_SEND_DATA delivered to Z-Wave stack
2018-02-01 02:52:12.539 Detail, Node028,   Received: 0x01, 0x07, 0x00, 0x13, 0x67, 0x00, 0x00, 0x16, 0x9a
2018-02-01 02:52:12.539 Detail, Node028,   ZW_SEND_DATA Request with callback ID 0x67 received (expected 0x67)
2018-02-01 02:52:12.539 Info, Node028, Request RTT 216 Average Request RTT 214
2018-02-01 02:52:12.717 Detail, Node028,   Received: 0x01, 0x0c, 0x00, 0x04, 0x00, 0x1c, 0x06, 0x60, 0x0a, 0x01, 0x08, 0x06, 0x31, 0xb9
2018-02-01 02:52:12.717 Info, Node028, Response RTT 394 Average Response RTT 390
2018-02-01 02:52:12.717 Info, Node028, Received MultiChannelCapabilityReport from node 28 for endpoint 1
2018-02-01 02:52:12.717 Info, Node028,     Endpoint is not dynamic, and is a General Thermostat V2
2018-02-01 02:52:12.717 Info, Node028,     Command classes supported by the endpoint are:
2018-02-01 02:52:12.718 Warning, Node028,         31d (Invalid)
2018-02-01 02:52:12.718 Detail, Node028,   Expected reply and command class was received
2018-02-01 02:52:12.718 Detail, Node028,   Message transaction complete
2018-02-01 02:52:12.718 Detail, Node028, Removing current message
2018-02-01 02:52:12.718 Detail, Node028, Query Stage Complete (Instances)
2018-02-01 02:52:12.719 Detail, Node028, AdvanceQueries queryPending=0 queryRetries=0 queryStage=Static live=1
```
The thing that interests me is the part that says: `Command classes supported by the endpoint are: 31d (Invalid)`.

chilicheech, 2018-02-01:
I also see this in the ozwcache:
```
<CommandClass id="96" name="COMMAND_CLASS_MULTI_INSTANCE/CHANNEL" version="96" request_flags="1" innif="true">
    <Instance index="1" />
</CommandClass>
```
`version="96"`? That seems wrong..

Fishwaldo, 2018-02-19:
Could you provide full (unfiltered) logs of a inclusion of this device? 

Something seems funky here...

chilicheech, 2018-02-26:
Apologies for taking so long. I'll put the logs in as soon as I can.

PatrickCronin, 2018-07-29:
@chilicheech - I'd be interested to hear how this is going/went?

Fishwaldo, 2018-10-26:
Closing due to no feedback.

closed 2018-10-26 by Fishwaldo