OpenZWave/open-zwave #1447: Z-Wave.Me RaZberry Controller 2016 ZWave+ is not recognized as a Zwave Plus device

nicolaevladescu, 2018-02-17:
Z-Wave.Me RaZberry Controller 2016 ZWave+ is not recognized as a Zwave Plus device

```
averageRequestRTT: 0
averageResponseRTT: 0
capabilities: beaming,staticUpdateController,listening,primaryController
friendly_name: Z-Wave.Me RaZberry Controller 2016 ZWave+
is_awake: true
is_failed: false
is_info_received: true
is_ready: true
is_zwave_plus: false
lastRequestRTT: 0
lastResponseRTT: 0
manufacturer_name: Z-Wave.Me
max_baud_rate: 40000
neighbors: 2
new_entity_id: zwave.zwaveme_razberry_controller_2016_zwave
node_id: 1
node_name: Z-Wave.Me RaZberry Controller 2016 ZWave+
old_entity_id: zwave.zwaveme_razberry_controller_2016_zwave_1
product_name: RaZberry Controller 2016 ZWave+
query_stage: Complete
receivedCnt: 0
receivedDups: 0
receivedTS: 2018-02-18 00:34:31:873
receivedUnsolicited: 0
retries: 0
sentCnt: 1
sentFailed: 0
sentTS: 2018-02-18 00:34:32:155
```

Fishwaldo, 2018-02-19:
that's what the device reports - IF you can provide full inclusion logs we can verify. (it also depends upon the software stack you run on the Razberry, so please ensure its a zwave+ stack)

nicolaevladescu, 2018-02-19:
I am using home assistant which uses python-openzwave which in turn call the c libraries from this project.

The above output is from the`pyozw_check` script provided by python-openzwave

The software stack is raspbian, home-assistant(python module) and python-openzwave, i'm not sure if this answers your question.

This is just a controller, i have not yet included any nodes so i don't have any inclusion logs.

bibi21000, 2018-02-21:
Please use -l option with pyozw_check annd attach OZW_Log.log (this is the file Fishwaldo need)

nicolaevladescu, 2018-02-22:
```
2018-02-22 10:27:54.422 Always, OpenZwave Version 1.4.2926 Starting Up
2018-02-22 10:27:55.424 Info, Setting Up Provided Network Key for Secure Communications
2018-02-22 10:27:55.425 Warning, Failed - Network Key Not Set
2018-02-22 10:27:55.425 Info, mgr,     Added driver for controller /dev/ttyAMA0
2018-02-22 10:27:55.425 Info,   Opening controller /dev/ttyAMA0
2018-02-22 10:27:55.425 Info, Trying to open serial port /dev/ttyAMA0 (attempt 1)
2018-02-22 10:27:55.426 Info, Serial port /dev/ttyAMA0 opened (attempt 1)
2018-02-22 10:27:55.426 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
2018-02-22 10:27:55.427 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_MEMORY_GET_ID: 0x01, 0x03, 0x00, 0x20, 0xdc
2018-02-22 10:27:55.427 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES: 0x01, 0x03, 0x00, 0x05, 0xf9
2018-02-22 10:27:55.427 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_GET_CAPABILITIES: 0x01, 0x03, 0x00, 0x07, 0xfb
2018-02-22 10:27:55.427 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_SUC_NODE_ID: 0x01, 0x03, 0x00, 0x56, 0xaa
2018-02-22 10:27:55.427 Detail, 
2018-02-22 10:27:55.427 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x15) - FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
2018-02-22 10:27:55.430 Detail, contrlr,   Received: 0x01, 0x10, 0x01, 0x15, 0x5a, 0x2d, 0x57, 0x61, 0x76, 0x65, 0x20, 0x34, 0x2e, 0x30, 0x35, 0x00, 0x01, 0x97
2018-02-22 10:27:55.430 Detail, 
2018-02-22 10:27:55.430 Info, contrlr, Received reply to FUNC_ID_ZW_GET_VERSION:
2018-02-22 10:27:55.431 Info, contrlr,     Static Controller library, version Z-Wave 4.05
2018-02-22 10:27:55.431 Detail, Node045,   Expected reply was received
2018-02-22 10:27:55.431 Detail, Node045,   Message transaction complete
2018-02-22 10:27:55.431 Detail, 
2018-02-22 10:27:55.431 Detail, contrlr, Removing current message
2018-02-22 10:27:55.431 Detail, 
2018-02-22 10:27:55.431 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x20) - FUNC_ID_ZW_MEMORY_GET_ID: 0x01, 0x03, 0x00, 0x20, 0xdc
2018-02-22 10:27:55.434 Detail, contrlr,   Received: 0x01, 0x08, 0x01, 0x20, 0xcb, 0x71, 0x71, 0xc8, 0x01, 0xd4
2018-02-22 10:27:55.434 Detail, 
2018-02-22 10:27:55.434 Info, contrlr, Received reply to FUNC_ID_ZW_MEMORY_GET_ID. Home ID = 0xcb7171c8.  Our node ID = 1
2018-02-22 10:27:55.434 Detail, Node113,   Expected reply was received
2018-02-22 10:27:55.434 Detail, Node113,   Message transaction complete
2018-02-22 10:27:55.434 Detail, 
2018-02-22 10:27:55.434 Detail, contrlr, Removing current message
2018-02-22 10:27:55.434 Detail, 
2018-02-22 10:27:55.435 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x05) - FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES: 0x01, 0x03, 0x00, 0x05, 0xf9
2018-02-22 10:27:55.437 Detail, contrlr,   Received: 0x01, 0x04, 0x01, 0x05, 0x1c, 0xe3
2018-02-22 10:27:55.437 Detail, 
2018-02-22 10:27:55.437 Info, contrlr, Received reply to FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES:
2018-02-22 10:27:55.437 Info, contrlr,     There is a SUC ID Server (SIS) in this network.
2018-02-22 10:27:55.437 Info, contrlr,     The PC controller is an inclusion static update controller (SUC) and was the original primary before the SIS was added.
2018-02-22 10:27:55.437 Detail, Node227,   Expected reply was received
2018-02-22 10:27:55.437 Detail, Node227,   Message transaction complete
2018-02-22 10:27:55.437 Detail, 
2018-02-22 10:27:55.437 Detail, contrlr, Removing current message
2018-02-22 10:27:55.438 Detail, 
2018-02-22 10:27:55.438 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x07) - FUNC_ID_SERIAL_API_GET_CAPABILITIES: 0x01, 0x03, 0x00, 0x07, 0xfb
2018-02-22 10:27:55.443 Detail, contrlr,   Received: 0x01, 0x2b, 0x01, 0x07, 0x05, 0x04, 0x01, 0x47, 0x04, 0x00, 0x00, 0x02, 0xfe, 0x83, 0xff, 0x88, 0xcf, 0x1f, 0x00, 0x00, 0xfb, 0x9f, 0x7d, 0xa0, 0x67, 0x00, 0x80, 0x80, 0x00, 0x80, 0x86, 0x00, 0x00, 0x00, 0xe8, 0x73, 0x00, 0x00, 0x0e, 0x00, 0x00, 0x40, 0x1a, 0x00, 0x5e
2018-02-22 10:27:55.443 Detail, 
2018-02-22 10:27:55.443 Info, contrlr,  Received reply to FUNC_ID_SERIAL_API_GET_CAPABILITIES
2018-02-22 10:27:55.444 Info, contrlr,     Serial API Version:   5.4
2018-02-22 10:27:55.444 Info, contrlr,     Manufacturer ID:      0x0147
2018-02-22 10:27:55.444 Info, contrlr,     Product Type:         0x0400
2018-02-22 10:27:55.444 Info, contrlr,     Product ID:           0x0002
2018-02-22 10:27:55.444 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_RANDOM: 0x01, 0x04, 0x00, 0x1c, 0x20, 0xc7
2018-02-22 10:27:55.444 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_GET_INIT_DATA: 0x01, 0x03, 0x00, 0x02, 0xfe
2018-02-22 10:27:55.444 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_SET_TIMEOUTS: 0x01, 0x05, 0x00, 0x06, 0x64, 0x0f, 0x97
2018-02-22 10:27:55.444 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_APPL_NODE_INFORMATION: 0x01, 0x07, 0x00, 0x03, 0x01, 0x02, 0x01, 0x00, 0xf9
2018-02-22 10:27:55.444 Detail, Node004,   Expected reply was received
2018-02-22 10:27:55.444 Detail, Node004,   Message transaction complete
2018-02-22 10:27:55.445 Detail, 
2018-02-22 10:27:55.445 Detail, contrlr, Removing current message
2018-02-22 10:27:55.445 Detail, 
2018-02-22 10:27:55.445 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x56) - FUNC_ID_ZW_GET_SUC_NODE_ID: 0x01, 0x03, 0x00, 0x56, 0xaa
[... cut ...]
2018-02-22 10:28:05.453 Always, Out of frame data flow errors:  . . . . . . . . . . . . . 0
2018-02-22 10:28:05.453 Always, Messages retransmitted: . . . . . . . . . . . . . . . . . 0
2018-02-22 10:28:05.453 Always, Messages dropped and not delivered: . . . . . . . . . . . 0
2018-02-22 10:28:05.454 Always, ***************************************************************************
2018-02-22 10:28:07.457 Detail, Node002, Removing current message
2018-02-22 10:28:07.458 Info, mgr,     Driver for controller /dev/ttyAMA0 removed

```

bibi21000, 2018-02-22:
can you give the output of the pyozw_check command ?

nicolaevladescu, 2018-02-22:
`pyozw_check`

```
Import libs
Try to import libopenzwave
Try to import libopenzwave.PyLogLevels
Try to get options
Try to get manager
Try to get python_openzwave version
0.4.3
Try to get python_openzwave full version
python_openzwave version 0.4.3 (embed-cython / Feb  1 2018 - 23:57:55)
Try to get openzwave version
1.4.2926
Try to get default config path
/srv/homeassistant/lib/python3.5/site-packages/python_openzwave/ozw_config
Try to destroy manager
Try to destroy options
Try to import openzwave (API)
```

bibi21000, 2018-02-27:
Sorry that was the result of this command I need :
pyozw_check -l -t 3600 -d /dev/ttyAMA0

nicolaevladescu, 2018-02-27:
```
(homeassistant) root@hassbian:/srv/homeassistant# pyozw_check -l -t 3600 -d /dev/ttyAMA0
usage: pyozw_check [-h] [-o {txt,raw}] [-d DEVICE] [-m] [-i] [-t TIMEOUT]
pyozw_check: error: unrecognized arguments: -l
```

Fishwaldo, 2018-03-07:
`2018-02-22 10:27:55.575 Info, Node001, Received reply to FUNC_ID_ZW_GET_NODE_PROTOCOL_INFO
2018-02-22 10:27:55.576 Info, Node001,   Protocol Info for Node 1:
2018-02-22 10:27:55.576 Info, Node001,     Listening     = true
2018-02-22 10:27:55.576 Info, Node001,     Beaming       = true
2018-02-22 10:27:55.576 Info, Node001,     Routing       = false
2018-02-22 10:27:55.576 Info, Node001,     Max Baud Rate = 40000
2018-02-22 10:27:55.576 Info, Node001,     Version       = 4
2018-02-22 10:27:55.576 Info, Node001,     Security      = false
2018-02-22 10:27:55.588 Info, Node001,   Basic device class    (0x02) - Static Controller
2018-02-22 10:27:55.588 Info, Node001,   Generic device Class  (0x02) - Static Controller
2018-02-22 10:27:55.588 Info, Node001,   Specific device class (0x01) - Static PC Controller
2018-02-22 10:27:55.588 Info, Node001,     COMMAND_CLASS_BASIC is not mapped
2018-02-22 10:27:55.588 Info, Node001,   Mandatory Command Classes for Node 1:
2018-02-22 10:27:55.588 Info, Node001,     None
2018-02-22 10:27:55.588 Info, Node001,   Mandatory Command Classes controlled by Node 1:
2018-02-22 10:27:55.589 Info, Node001,     COMMAND_CLASS_BASIC`

Whatever software running on the Rpi is not advertising the COMMAND_CLASS_ZWAVE_PLUS_INFO. Thats why its not picked up as ZWave+ Device.

So OZW is working as expected.. Not a problem on our end.

closed 2018-03-07 by Fishwaldo