OpenZWave/open-zwave #1455: Retry to slow and low retry time not working

jeroenst, 2018-03-03:
When a device is not responding, the library waits about 10 seconds before retrying, that is in my opinion to long and causes delay when multiple devices are set in a scene.

It is possible to lower the wait time before retries but when lowering that value causes nothing to work anymore.

Fishwaldo, 2018-03-07:
You can alter the timeout in Defs.h. 

Its 10 seconds (it was 40 at one stage) as when Networks get large, the latency increases. (remember, Z-Wave is a low bandwidth protocol)

But ideally - Instead of adjusting the timeout (which is a symptom!) why not fix the problematic nodes so retries are not even needed?

anyway, This is a issue tracker and not a support forum. 

Please use our Forums for Questions regarding OpenZWave: 
https://groups.google.com/forum/#!forum/openzwave

Thanks

closed 2018-03-07 by Fishwaldo

jeroenst, 2018-03-07:
It's a wireless protocol.  It will have lost messages due to radio interference, waiting 10 seconds after a failed message when more messages are pending (like lamp off, tv off, spots off) all the pending messages have to wait 10 seconds. I don't think this is a desired behavior imo..

Fishwaldo, 2018-03-07:
its real world experience - its a mesh network - Messages get retransmitted by neighbours, some older Z-Wave Chips are slow at decrypting/encrypting messages when the Security CC is in use, other devices only ACK after they have "enacted" whatever change has been requested. 

If I lower the timeout, then we will get other people with larger networks complaining about "false positive" timeouts. Been there - tested it, this is the result.

You are free to change on your local instance. 

As I said before - You are looking at fixing a "symptom" of a bigger problem. If you are having timeouts on your network, fix the reason why there are timeouts. Then you wont have to worry about the 10 second lag.

jeroenst, 2018-03-07:
Okay, thanx for your explenation! I appreciate it!