OpenZWave/open-zwave #1457: Blue color value not reported by ZIPATO BULB 2

jeroenst, 2018-03-06:
When I set the blue color value of the Zipato Bulb 2, the blue value returned from open-zwave is always 00/

`Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: MQTT RECEIVED: home/zwave/send/11/1/color/0=#0000FF0000
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: SENDING TO ZWAVE: NODEID:11 INSTANCEID:1 COMMANDCLASS=51 INDEX=0 VALUE=#0000FF0000
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m2018-03-06 12:52:32.460 Info, Node011, Value::Set - COMMAND_CLASS_COLOR - Color - 0 - 1 - #0000FF0000
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[39m2018-03-06 12:52:32.460 Info, Node011, Color::SetValue - Setting Color value
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.460 Detail, Node011, Queuing (Send) ColorCmd_Set (Node=11): 0x01, 0x14, 0x00, 0x13, 0x0b, 0x0d, 0x33, 0x05, 0x05, 0x02, 0x00, 0x03, 0x00, 0x04, 0xff, 0x00, 0x00, 0x01, 0x00, 0x25, 0xe7, 0xf4
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.460 Detail, Node011, Queuing (Send) ColorCmd_Get (Node=11): 0x01, 0x0a, 0x00, 0x13, 0x0b, 0x03, 0x33, 0x03, 0x00, 0x25, 0xe8, 0x13
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.460 Detail, Node011, Queuing (Send) ColorCmd_Get (Node=11): 0x01, 0x0a, 0x00, 0x13, 0x0b, 0x03, 0x33, 0x03, 0x01, 0x25, 0xe9, 0x13
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.460 Detail, Node011, Queuing (Send) ColorCmd_Get (Node=11): 0x01, 0x0a, 0x00, 0x13, 0x0b, 0x03, 0x33, 0x03, 0x02, 0x25, 0xea, 0x13
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.460 Detail, Node011, Queuing (Send) ColorCmd_Get (Node=11): 0x01, 0x0a, 0x00, 0x13, 0x0b, 0x03, 0x33, 0x03, 0x03, 0x25, 0xeb, 0x13
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.461 Detail,
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[39m2018-03-06 12:52:32.461 Info, Node011, Sending (Send) message (Callback ID=0xe7, Expected Reply=0x00) - ColorCmd_Set (Node=11): 0x01, 0x14, 0x00, 0x13, 0x0b, 0x0d, 0x33, 0x05, 0x05, 0x02, 0x00, 0x03, 0x00, 0x04, 0xff, 0x00, 0x00, 0x01, 0x00, 0x25, 0xe7, 0xf4
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.469 Detail, Node011,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.469 Detail, Node011,   ZW_SEND_DATA delivered to Z-Wave stack
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.486 Detail, Node011,   Received: 0x01, 0x07, 0x00, 0x13, 0xe7, 0x00, 0x00, 0x02, 0x0e
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.486 Detail, Node011,   ZW_SEND_DATA Request with callback ID 0xe7 received (expected 0xe7)
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[39m2018-03-06 12:52:32.486 Info, Node011, Request RTT 25 Average Request RTT 44
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.486 Detail,   Expected callbackId was received
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.486 Detail,   Message transaction complete
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.486 Detail,
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.487 Detail, Node011, Removing current message
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.487 Detail,
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[39m2018-03-06 12:52:32.487 Info, Node011, Sending (Send) message (Callback ID=0xe8, Expected Reply=0x04) - ColorCmd_Get (Node=11): 0x01, 0x0a, 0x00, 0x13, 0x0b, 0x03, 0x33, 0x03, 0x00, 0x25, 0xe8, 0x13
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.494 Detail, Node011,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.494 Detail, Node011,   ZW_SEND_DATA delivered to Z-Wave stack
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.578 Detail, Node011,   Received: 0x01, 0x07, 0x00, 0x13, 0xe8, 0x00, 0x00, 0x08, 0x0b
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.578 Detail, Node011,   ZW_SEND_DATA Request with callback ID 0xe8 received (expected 0xe8)
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[39m2018-03-06 12:52:32.578 Info, Node011, Request RTT 90 Average Request RTT 67
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.578 Detail,   Expected callbackId was received
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.589 Detail, Node011,   Received: 0x01, 0x0a, 0x00, 0x04, 0x00, 0x0b, 0x04, 0x33, 0x04, 0x00, 0x00, 0xc9
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.589 Detail,
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[39m2018-03-06 12:52:32.589 Info, Node011, Response RTT 102 Average Response RTT 89
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.589 Detail, Node011,   Expected reply and command class was received
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.589 Detail, Node011,   Message transaction complete
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.589 Detail,
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.589 Detail, Node011, Removing current message
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.589 Detail,
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[39m2018-03-06 12:52:32.589 Info, Node011, Sending (Send) message (Callback ID=0xe9, Expected Reply=0x04) - ColorCmd_Get (Node=11): 0x01, 0x0a, 0x00, 0x13, 0x0b, 0x03, 0x33, 0x03, 0x01, 0x25, 0xe9, 0x13
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.598 Detail, Node011,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.598 Detail, Node011,   ZW_SEND_DATA delivered to Z-Wave stack
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.614 Detail, Node011,   Received: 0x01, 0x07, 0x00, 0x13, 0xe9, 0x00, 0x00, 0x02, 0x00
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.614 Detail, Node011,   ZW_SEND_DATA Request with callback ID 0xe9 received (expected 0xe9)
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[39m2018-03-06 12:52:32.614 Info, Node011, Request RTT 24 Average Request RTT 45
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.614 Detail,   Expected callbackId was received
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.625 Detail, Node011,   Received: 0x01, 0x0a, 0x00, 0x04, 0x00, 0x0b, 0x04, 0x33, 0x04, 0x01, 0x00, 0xc8
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.625 Detail,
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[39m2018-03-06 12:52:32.625 Info, Node011, Response RTT 35 Average Response RTT 62
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.625 Detail, Node011,   Expected reply and command class was received
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.625 Detail, Node011,   Message transaction complete
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.625 Detail,
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.625 Detail, Node011, Removing current message
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.625 Detail,
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[39m2018-03-06 12:52:32.626 Info, Node011, Sending (Send) message (Callback ID=0xea, Expected Reply=0x04) - ColorCmd_Get (Node=11): 0x01, 0x0a, 0x00, 0x13, 0x0b, 0x03, 0x33, 0x03, 0x02, 0x25, 0xea, 0x13
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.634 Detail, Node011,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.634 Detail, Node011,   ZW_SEND_DATA delivered to Z-Wave stack
[... cut ...]
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.697 Detail, Node011, Notification: ValueChanged
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39m#033[34m2018-03-06 12:52:32.697 Detail, Node011, Notification: ValueChanged
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: #033[39mMQTT PUBLISHING: home/zwave/11/1/color/color=#0000000000
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: 1520337152697 zwave: value changed
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: 11, 51, [object Object]
Mar  6 12:52:32 server01 zwaveMQTT.js[26811]: MQTT PUBLISHING: home/zwave/11/1/color/color_index=Off
`

nechry, 2018-03-06:
try to changes the Color Channels with 31.

jeroenst, 2018-03-06:
Actual the bulb turns blue but the value received from openzwave says it is 0..

I don't understand what you mean witj change the color channels with 31..

jeroenst, 2018-03-06:
I also see that open zwave is only requesting color register 0 - 3. 

According to the zwave specification it'snot requesting the blue color:

0 Warm White 0x00 – 0xFF: 0 – 100%
1 Cold White 0x00: - 0xFF: 0 – 100%
2 Red 0x00 – 0xFF: 0 – 100%
3 Green 0x00 – 0xFF: 0 – 100%
4 Blue 0x00 – 0xFF: 0 – 100%
5 Amber (for 6ch Color mixing) 0x00 – 0xFF: 0 – 100%
6 Cyan (for 6ch Color mixing) 0x00 – 0xFF: 0 – 100%
7 Purple (for 6ch Color mixing) 0x00 – 0xFF: 0 – 100%
8 Indexed Color 0x00 – 0xFF: Color Index 0-255

nechry, 2018-03-06:
Color Channels is a System value. I don't know how to change value with our gateway. in jeedom is strike-forward. I also have this bulb with same behavior, if I set the Color Channels to 31, I receive right color report

jeroenst, 2018-03-06:
You're right, setting colorchannels to 31 works, thanks...

But still I think this is a bug, because 5 channels should be okay and it would include the blue channel..

Also I tried to set colorchannels to 6 but then also the blue value is not requested by openzwave...

jeroenst, 2018-03-06:
Okay, I made a workaround in the zwave library, will create a pull request in a few minutes..

jeroenst, 2018-03-06:
I've created a pull request fixing this issue..

Fishwaldo, 2018-03-07:
Could you provide a log showing the get capabilities query (and response?) 

Or could you confirm if its a off by 1 issue (eg, index based on 1 rather than 0 in the spec). Its unusual to see a "warm white" and "cold white" in the same bulb (at least I've not seen it yet)

jeroenst, 2018-03-07:
Yes this bulb has warm white and cold white leds in the same bulb. Please wait for logging, i will provide it later.

jeroenst, 2018-03-07:
Logging as requested (as you can see no Blue color is reported and warm and cold white is):

2018-03-07 08:25:52.342 Info, Node011, Sending (Query) message (Callback ID=0x42, Expected Reply=0x04) - ColorCmd_CapabilityGet (Node=11): 0x01, 0x09, 0x00, 0x13, 0x0b, 0x02, 0x33, 0x01, 0x25, 0x42, 0xb9
2018-03-07 08:25:52.349 Detail, Node011,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2018-03-07 08:25:52.349 Detail, Node011,   ZW_SEND_DATA delivered to Z-Wave stack
2018-03-07 08:25:52.365 Detail, Node011,   Received: 0x01, 0x07, 0x00, 0x13, 0x42, 0x00, 0x00, 0x02, 0xab
2018-03-07 08:25:52.365 Detail, Node011,   ZW_SEND_DATA Request with callback ID 0x42 received (expected 0x42)
2018-03-07 08:25:52.365 Info, Node011, Request RTT 23 Average Request RTT 23
2018-03-07 08:25:52.376 Detail, Node011,   Received: 0x01, 0x0a, 0x00, 0x04, 0x00, 0x0b, 0x04, 0x33, 0x02, 0x0f, 0x00, 0xc0
2018-03-07 08:25:52.377 Info, Node011, Response RTT 34 Average Response RTT 34
2018-03-07 08:25:52.377 Info, Node011, Received an Color Capability Report: Capability=fd
2018-03-07 08:25:52.377 Info, Node011, Red (0x02)
2018-03-07 08:25:52.377 Info, Node011, Green (0x03)
2018-03-07 08:25:52.377 Info, Node011, Warm White (0x00)
2018-03-07 08:25:52.377 Info, Node011, Cold White (0x01)

jeroenst, 2018-03-07:
Here is logging with my patch: 

2018-03-07 08:52:33.265 Info, Node011, Sending (Query) message (Callback ID=0x40, Expected Reply=0x04) - ColorCmd_CapabilityGet (Node=11): 0x01, 0x09, 0x00, 0x13, 0x0b, 0x02, 0x33, 0x01, 0x25, 0x40, 0xbb
2018-03-07 08:52:33.273 Detail, Node011,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2018-03-07 08:52:33.273 Detail, Node011,   ZW_SEND_DATA delivered to Z-Wave stack
2018-03-07 08:52:33.289 Detail, Node011,   Received: 0x01, 0x07, 0x00, 0x13, 0x40, 0x00, 0x00, 0x02, 0xa9
2018-03-07 08:52:33.289 Detail, Node011,   ZW_SEND_DATA Request with callback ID 0x40 received (expected 0x40)
2018-03-07 08:52:33.289 Info, Node011, Request RTT 24 Average Request RTT 24
2018-03-07 08:52:33.300 Detail, Node011,   Received: 0x01, 0x0a, 0x00, 0x04, 0x00, 0x0b, 0x04, 0x33, 0x02, 0x0f, 0x00, 0xc0
2018-03-07 08:52:33.300 Info, Node011, Response RTT 35 Average Response RTT 35
2018-03-07 08:52:33.301 Info, Node011, Received an Color Capability Report: Capability=1fd
2018-03-07 08:52:33.301 Info, Node011, Red (0x02)
2018-03-07 08:52:33.301 Info, Node011, Green (0x03)
2018-03-07 08:52:33.301 Info, Node011, Blue (0x04)
2018-03-07 08:52:33.301 Info, Node011, Warm White (0x00)
2018-03-07 08:52:33.301 Info, Node011, Cold White (0x01)

jeroenst, 2018-03-07:
Okay, pull request is ready...

Fishwaldo, 2018-03-07:
Ok - Buggy device. Please report to Zipato so they can fix in a future version.

In the mean time, no need to create a PR, the capabilities can already be specified in the config file with a 
`capabilities="31"` in the zapato config file and removing the ozw_config.xml file

Fishwaldo, 2018-03-07:
Typo it should be `colorchannels="31"`

jeroenst, 2018-03-07:
Do you have an example so I can update the open-zwave/config/RGBBulb2.xml and create a pull request?

Or even better, can you fix this in the next commit to open-zwave repo?

jeroenst, 2018-03-07:
Sorry, I see your label ConfigUpdate, I suppose it's taking care of it..

Thanx in advance!

Fishwaldo, 2018-03-07:
Nope - I don't have a device to test. See https://github.com/OpenZWave/open-zwave/blob/725277e2ce1ba1bf7899e11a814f6447f03a2ea2/config/fibaro/fgrgbwm441.xml#L13
but instead of coloridxbug, use colorchannels

jeroenst, 2018-03-07:
I made a new pull request and informed zipato about this bug.

commit 404bb658 (2018-03-08): Merge pull request #1461 from jeroenst/master

This commit fixes #1457 Blue color value not reported by ZIPATO BULB 2
commit 7bfc4312 (2018-03-08): This commit fixes #1457 Blue color value not reported by ZIPATO BULB 2
closed 2018-03-08 by Fishwaldo