OpenZWave/open-zwave #1462: BeginControllerCommand deprecated with no suitable replacement

jviki, 2018-03-14:
I did not get any response in the forum (https://groups.google.com/forum/#!topic/openzwave/GHySe_Zkc_Q) so I'll put it here because I consider this issue to be a bug.

I am implementing an application with OZW for accessing the Z-Wave network and dealing with issuing commands. The API is not very clear to me and the documentation does not clarify this.

When I issue a command via OZW (mostly AddNode, RemoveNode) in a multi-threaded application, I need some feedback about the progress of the command. Well, the doc says, use a Watcher and look for Notification of type Type_ControllerCommand . OK, I started with that but discovered that there is no way how distinguish which command is being reported by this API to me. And OZW itself sometimes issues commands on its own internally. So, if I call Manager::Get()->AddNode(...) and receive a notification, I can only guess which command is being reported to me.

So, after this, I found BeginControllerCommand call. Cool, that is what I was looking for! But, it is deprecated... So, I'd like to ask, are you going to change anything about this? From my current level of understanding the OZW, I think that without the BeginControllerCommand, the OZW API for issuing commands is **broken by design** and **cannot be used right**.

There are at least 2 possible solutions:

1. Introduce the same callback mechanism into the new calls AddDevice, RemoveDevice, etc.
2. Introduce another field into the Notification class that can tell me what type of command is the notification reporting about.

Zkral, 2018-03-14:
In the Dev branch you can use Notification::GetCommand to get the Controller Command type.

jviki, 2018-03-15:
Thanks for clarification.

closed 2018-03-21 by Fishwaldo