OpenZWave/open-zwave #1468: Aeotec Nano Dimmer, Group association and wrong behaviour

rjblake, 2018-03-21:
Have a number of Nano dimmer exhibiting the following problem using OpenZwave:

**Setup:**
One dimmer (Dimmer A) has another dimmer (Dimmer B) associated with the second switch (S2). 
Dimmer A: 2x momentary switches connected - S1 (controls own light), S2 (controls Dimmer B)
Dimmer B: 1x momentary switch connected - S1 (controls own light)
Operation:
When using the physical momentary switches to turn ON or OFF; there is no problem - single press turns ON light connected to Dimmer A and turns it off. The same applies when pressing S2 of Dimmer A associated with Dimmer B - it turns ON and OFF as expected.

If I use the Control Panel or other app using OpenZwave (e.g. Domoticz) and turn Dimmer B light on through the software, it turns ON and OFF without issue - all good. However, if I turn on with the Control Panel or App, I have to press the S2 on Dimmer A, or S1 on Dimmer B 2x for it to turn off. There are no errors being recorded; but it seems that somewhere the info is not in sync

Fishwaldo, 2018-03-21:
sounds like a issue with the Nodes talking to each other. Not a OZW issue. Check your associations etc to make sure the Dimmers are associated with the Switches etc.

closed 2018-03-21 by Fishwaldo

rjblake, 2018-03-21:
The nodes do talk to each other as stated. The only thing that causes the issue is when OpenZwave has any interaction with the network. I.e. from a cold start using only manual switches there is no issue