OpenZWave/open-zwave #1474: Zipato bulb colorclass gone

jeroenst, 2018-04-02:
Since bug https://github.com/OpenZWave/open-zwave/issues/1457 was solved, the COMMAND_COLOR_CLASS is not added anymore to the zwave config cache file causing it that colors not working anymore.

I'm aware that this issue is caused by my own bugfix in the past for the missing bluechannel in the color value, but I have not enougd knowledge to solve this bug.

When checking out the commit before the blue value fix everything works again after deleting the ozw cache file (except the blue color channel). When checking out after the commit and removing the ozw cache file it is not possible to read or control the color anymore.

Here is the part of the Zipato node in the cachefile where you clearly can see the COMMAND_CLASS_COLOR is not saved:

`         <Node id="11" name="" location="" basic="4" generic="17" specific="1" type="Multilevel Power Switch" listening="true" frequentListening="false" beaming="true" routing="true" max_baud_rate="40000" version="4" query_stage="Probe">
                <Manufacturer id="0" name="">
                        <Product type="0" id="0" name="" />
                </Manufacturer>
                <CommandClasses>
                        <CommandClass id="32" name="COMMAND_CLASS_BASIC" version="1" mapping="38">
                                <Instance index="1" />
                        </CommandClass>
                        <CommandClass id="38" name="COMMAND_CLASS_SWITCH_MULTILEVEL" version="1">
                                <Instance index="1" />
                                <Value type="byte" genre="user" instance="1" index="0" label="Level" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="0" />
                                <Value type="button" genre="user" instance="1" index="1" label="Bright" units="" read_only="false" write_only="true" verify_changes="false" poll_intensity="0" min="0" max="0" />
                                <Value type="button" genre="user" instance="1" index="2" label="Dim" units="" read_only="false" write_only="true" verify_changes="false" poll_intensity="0" min="0" max="0" />
                                <Value type="bool" genre="system" instance="1" index="3" label="Ignore Start Level" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="True" />
                                <Value type="byte" genre="system" instance="1" index="4" label="Start Level" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="0" />
                        </CommandClass>
                        <CommandClass id="39" name="COMMAND_CLASS_SWITCH_ALL" version="1">
                                <Instance index="1" />
                                <Value type="list" genre="system" instance="1" index="0" label="Switch All" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" vindex="0" size="1">
                                        <Item label="Disabled" value="0" />
                                        <Item label="Off Enabled" value="1" />
                                        <Item label="On Enabled" value="2" />
                                        <Item label="On and Off Enabled" value="255" />
                                </Value>
                        </CommandClass>
                </CommandClasses>
        </Node>`

jeroenst, 2018-04-04:
Please check if my commit is ok. It works for me at the moment. 
When agreed I will post a pull request.

jeroenst, 2018-04-25:
I can confirm this patch works great, creating pull request.

BramJ85, 2018-08-13:
Patch works for me as well.

Fishwaldo, 2018-10-05:
Need Logs.

closed 2018-11-14 by no-response[bot]