OpenZWave/open-zwave #1476: ZW111.xml - non existent parameters

rjblake, 2018-04-04:
Update from commit 0ee8b07775d585ce540185b87f061e728bff0fcd includes non existent parameters form an engineering spec that does not seem to exist. 

Latest Aeotec Engineering Spec on their site for Nano Dimmer is dated 06/01/2017 (1-Jun-2017). Parameters 83&84 do not exist (nor make sense in the case of the dimmer) as shown in lines 33-60 of the zw111.xml file. Seems that code from a different device (Smart Dimmer perhaps) has ended up in this file

nechry, 2018-04-04:
Please check the latest version Aeon Labs ZW111 Nano Dimmer Engineering Spec (3/6/2018)
[Aeon Labs ZW111 Nano Dimmer Engineering Spec.pdf](https://github.com/OpenZWave/open-zwave/files/1875726/Aeon.Labs.ZW111.Nano.Dimmer.Engineering.Spec.pdf)
Parameters 83 & 84 exist.
Also check this commit: https://github.com/OpenZWave/open-zwave/commit/0ee8b07775d585ce540185b87f061e728bff0fcd#diff-4c140e402769591b3d9b2aea6aa74684

closed 2018-04-04 by nechry

nechry, 2018-04-04:
Yes its with the latest application version, and is not wet published. I receive the ota file directly from Aeotec, with the updated file.

rjblake, 2018-04-04:
That would explain why I have additional problems when using this version perhaps. Can you share the OTA file somehow? I have the EU versions

rjblake, 2018-04-11:
I have updated a total of 12 Nano Dimmers with the v2.01 test firmware, following Aeotec's 2-step approach, but not excluding/including the devices. All appear to be working correctly, but their full functionality is not reflected on investigating the devices using the Zensys tool. Adding the devices using the Zensys 'Add Node' provides a more complete device. I will exclude/include some test devices using both Zensys and OpenZwave to document the differences. With regards functionality, it looks like the Multilevel set is not functioning as expected. Best I can describe is as follows:
When setting Parameter 82 to 'Switch Multilevel Set', and Parameter 124 to 'Other Nodes', the ramp up/ramp down dimming functionality does not work (for nodes associated in Group4). However, if one sets Parameter 124 to 'Output loads of itself and other nodes' it does, but as a result affects the state of the light connected to the dimmer being switched. If you wish to use Switch2 to control another Nano Dimmer independently, this doesn't work. Aeotec are investigating to see if this is an issue with the way the firmware works, as initial thoughts are that independent operation should be possible.

nechry, 2018-04-11:
Yes is possible firmware issues. I did not play deeper as you with this advanced parameters. I just figured the device accepted them. Aeotec provides good support and listening users issues and providing OTA update to address issues.