OpenZWave/open-zwave #1478: add device Zooz V2 Zen21

ghvader, 2018-04-12:
please add device light switch (paddle) V2 of Zooz Zen21
(rename attachement to xml)
[zen21v2.txt](https://github.com/OpenZWave/open-zwave/files/1900869/zen21v2.txt)

```
<Manufacturer id="27a" name="Zooz">
			<Product type="b111" id="1e1c" name="Unknown: type=b111, id=1e1c" />

```

nechry, 2018-04-30:
this is not a valid configuration file. please add parameters and association groups definitions like 
https://github.com/OpenZWave/open-zwave/wiki/Adding-Devices

http://www.getzooz.com/downloads/zooz-z-wave-plus-on-off-wall-switch-zen21-ver2-user-manual.pdf

ghvader, 2018-05-03:
[zen21v2.txt](https://github.com/OpenZWave/open-zwave/files/1972984/zen21v2.txt)
how about this one? rename to .xml

commit 86f234eb (2018-08-20): add Zooz ZEN21 Switch V 2.0

Close #1478
closed 2018-08-20 by nechry