OpenZWave/open-zwave #1482: Refreshing zwcfg file

cincuranet, 2018-04-16:
I'm now implementing configuration for Qubino 3-phase meter and whenever I'm testing new XML file I do exclude and include. And that's pretty inconvenient, given the device is 3-phase (aka a lot of wires around). Is it **safe** to just delete the `zwcfg` file (stopping OZW (in my case used via Domoticz)) before?

nechry, 2018-04-16:
Yes is safe, but I prefer/recommend exclude and include device, this is faster then wait your network is ready.
In jeedom I implemented a rediscovering of device, this helper stop the network, remove in zwcfg file the xml node and restart the network. with faster restart, and you did loose any cached values for other device in your network.

nechry, 2018-04-16:
I will close this issue. by the way thanks for your PRs

closed 2018-04-16 by nechry