OpenZWave/open-zwave #1483: The variable 'usercode' is used both as a parameter and as destination in s[n]printf().

nihilus, 2018-04-19:
As outputted from 'cppcheck':
[DoorLockLogging.cpp:312]: (error) The variable 'usercode' is used both as a parameter and as destination in s[n]printf(). The origin and destination buffers overlap. Quote from glibc (C-library) documentation (http://www.gnu.org/software/libc/manual/html_mono/libc.html#Formatted-Output-Functions): "If copying takes place between objects that overlap as a result of a call to sprintf() or snprintf(), the results are undefined."

nihilus, 2018-04-19:
Here is a fix:
`char usercode[254], tmpusercode[254];
snprintf(usercode, sizeof(usercode), "UserCode:");
if (usercodelength > 0)
				for (int i = 0; i < usercodelength; i++ )
				{
					strncpy(tmpusercode, usercode, sizeof(tmpusecode));
					snprintf(usercode, sizeof(usercode), "%s %d", tmpusercode, (int)_data[12+i]);
				}`

sjakub, 2018-05-11:
I think I have a better fix for this issue :)
[fix_patch.txt](https://github.com/OpenZWave/open-zwave/files/1997036/fix_patch.txt)

nihilus, 2018-05-11:
@sjakub yeah that is much cleaner, I just run cppcheck over the codebase and made a quick fix.

waxhell, 2018-05-16:
sjakub's patch worked like a charm. 

This fixed compilation errors on Arch ARM Linux, on armv5te.
gcc 8.1.0-1
make 4.2.1-2

Chris-V, 2018-07-04:
That also solved my issues on Arch Linux. Has anyone opened a PR yet with this patch?

closed 2018-07-31 by Fishwaldo

sjakub, 2018-07-31:
@Fishwaldo It looks like the ugly version went in. It uses 0.5KB of memory and copies everything several times back and forth, just to construct a silly string.