OpenZWave/open-zwave #1485: Compilation error

pipiche38, 2018-04-23:
While compiling open-wave-read-only on a RPi3B+ I'm getting a compilation error

Building DoorLockLogging.o
/home/support/domoticz/open-zwave-read-only/cpp/src/command_classes/DoorLockLogging.cpp: In member function ‘virtual bool OpenZWave::DoorLockLogging::HandleMsg(const uint8*, uint32, uint32)’:
/home/support/domoticz/open-zwave-read-only/cpp/src/command_classes/DoorLockLogging.cpp:312:15: error: passing argument 1 to restrict-qualified parameter aliases with argument 4 [-Werror=restrict]
      snprintf(usercode, sizeof(usercode), "%s %d", usercode, (int)_data[12+i]);
               ^~~~~~~~                             ~~~~~~~~
cc1plus: all warnings being treated as errors
make[1]: *** [/home/support/domoticz/open-zwave-read-only/cpp/build/support.mk:136: /home/support/domoticz/open-zwave-read-only/.lib/DoorLockLogging.o] Error 1
make[1]: Leaving directory '/home/support/domoticz/open-zwave-read-only/cpp/build'
make: *** [Makefile:20: all] Error 2

pipiche38, 2018-04-23:
Fedora 28 - Kernel 4.16.3-300.fc28.armv7hl

$ rpm -q cpp
cpp-8.0.1-0.20.fc28.armv7hl
[$ rpm -q gcc
gcc-8.0.1-0.20.fc28.armv7hl
$ rpm -q make
make-4.2.1-6.fc28.armv7hl

Fishwaldo, 2018-04-26:
Duplicate as #1483

closed 2018-04-26 by Fishwaldo