OpenZWave/open-zwave #1489: Popp Keypad not working

stiansoevik, 2018-04-24:
I am trying to add the Popp Keypad using ozwcp. The final goal is to use it with Home Assistant, but I have isolated my setup to a separate Z-Wave controller (UZB1) and fresh install of ozwcp. I have successfully included the keypad with Fibaro HC2.

Test environment:
- open-zwave (commit aba9046)
- open-zwave-control-panel (bbdb461)
- libmicrohttpd (0.9.59)
- Macbook Pro
- UZB1

Reproduction steps:
- Build ozwcp
- Enable security key in options.xml
- Reset UZB1 via ozwcp
- Reset keypad via 5 *
- Add secure via ozwcp, device is right next (< 10 cm) to UZB
- Include keypad via 4 *

Expected results:
- Association groups available in ozwcp
- User codes available in ozwcp

Results:
- No association groups or user codes available

Observations:
- No association groups, codes etc were available after secure inclusion. However, I first forgot to enable the key in options.xml, and then group 1 (of 3) was available, and 20 user codes possible to enter.

Log (I reset the controller and deleted zwcfg* first, yet it complains): http://www.openzwave.com/log-analyzer/public/e3de406f-7165-4dd1-8a23-1b1ea375438c/

Fishwaldo, 2018-04-26:
Node 2? Based on the logs, it doesn't find the config files... hence no association groups. Fix that, and try again. 

Finally, 
This is a issue tracker and not a support forum. 

Please use our Forums for Questions regarding OpenZWave: 
https://groups.google.com/forum/#!forum/openzwave

Thanks

closed 2018-04-26 by Fishwaldo

Fishwaldo, 2018-05-03:
This isn't a help forum. Its for tracking bugs in OZW. If you need help, the mailing lists are the place to go.