OpenZWave/open-zwave #1491: Debug logging does not work

nowox, 2018-04-26:
I have this option file where I configured the `SaveLogLevel` to `9`. So I am expecting to see Debug data in my log file. 

```
<Options xmlns='http://code.google.com/p/open-zwave/'>
  <Option name="logging" value="true" />
  <Option name="Associate" value="true" />
  <Option name="NotifyTransactions" value="false" />
  <Option name="DriverMaxAttempts" value="0" />
  <Option name="SaveConfiguration" value="true" />
  <Option name="LogFileName" value="output.log" />
  <Option name="AppendLogFile" value="true" />
  <Option name="SaveLogLevel" value="9" />
  <Option name="SaveConfiguration" value="true" />
</Options>
```

In the `output.log` I only see `Info`, `Detail` and `Warning`. 

```
2018-04-26 07:35:43.777 Always, OpenZwave Version 1.4.2926 Starting Up
2018-04-26 07:35:43.780 Info, Setting Up Provided Network Key for Secure Communications
2018-04-26 07:35:43.785 Warning, Failed - Network Key Not Set
2018-04-26 07:35:43.786 Info, mgr,     Added driver for controller /dev/ttyACM1
2018-04-26 07:35:43.786 Info,   Opening controller /dev/ttyACM1
2018-04-26 07:35:43.786 Info, Trying to open serial port /dev/ttyACM1 (attempt 1)
2018-04-26 07:35:43.787 Info, Serial port /dev/ttyACM1 opened (attempt 1)
2018-04-26 07:35:43.787 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
2018-04-26 07:35:43.787 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_MEMORY_GET_ID: 0x01, 0x03, 0x00, 0x20, 0xdc
2018-04-26 07:35:43.787 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES: 0x01, 0x03, 0x00, 0x05, 0xf9
2018-04-26 07:35:43.787 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_GET_CAPABILITIES: 0x01, 0x03, 0x00, 0x07, 0xfb
2018-04-26 07:35:43.787 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_SUC_NODE_ID: 0x01, 0x03, 0x00, 0x56, 0xaa
2018-04-26 07:35:43.787 Detail,
2018-04-26 07:35:43.787 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x15) - FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
2018-04-26 07:35:43.792 Detail, contrlr,   Received: 0x01, 0x10, 0x01, 0x15, 0x5a, 0x2d, 0x57, 0x61, 0x76, 0x65, 0x20, 0x34, 0x2e, 0x33, 0x34, 0x00, 0x01, 0x95
2018-04-26 07:35:43.796 Detail,
2018-04-26 07:35:43.796 Info, contrlr, Received reply to FUNC_ID_ZW_GET_VERSION:
2018-04-26 07:35:43.796 Info, contrlr,     Static Controller library, version Z-Wave 4.34
2018-04-26 07:35:43.796 Detail, Node045,   Expected reply was received
2018-04-26 07:35:43.796 Detail, Node045,   Message transaction complete
2018-04-26 07:35:43.796 Detail,
2018-04-26 07:35:43.796 Detail, contrlr, Removing current message
2018-04-26 07:35:43.796 Detail,
2018-04-26 07:35:43.796 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x20) - FUNC_ID_ZW_MEMORY_GET_ID: 0x01, 0x03, 0x00, 0x20, 0xdc
2018-04-26 07:35:43.799 Detail, contrlr,   Received: 0x01, 0x08, 0x01, 0x20, 0xdc, 0x0a, 0xcc, 0xf7, 0x01, 0x3a
2018-04-26 07:35:43.799 Detail,
2018-04-26 07:35:43.799 Info, contrlr, Received reply to FUNC_ID_ZW_MEMORY_GET_ID. Home ID = 0xdc0accf7.  Our node ID = 1
2018-04-26 07:35:43.799 Detail, Node010,   Expected reply was received
2018-04-26 07:35:43.799 Detail, Node010,   Message transaction complete
2018-04-26 07:35:43.799 Detail,
...
```

I this a bug?

I am using the latest version of Python-OpenZWave

nechry, 2018-04-26:
check in your python app if you have other log level override like this:
options.set_save_log_level(someLogLevel)

nowox, 2018-04-26:
Here is my App running on IPython: 

```python
#!/usr/bin/env python
from openzwave.network import ZWaveNetwork
from openzwave.option import ZWaveOption

options = ZWaveOption("/dev/ttyACM1", config_path="../openzwave/config", user_path=".", cmd_line="")
options.lock()

network = ZWaveNetwork(options, log="Debug")
```

nowox, 2018-05-01:
I tried that too.

nechry, 2018-05-01:
ok so try the log level 8 for debug instead of 9 (Protocol Information)
<Option name="SaveLogLevel" value="8" />

nowox, 2018-05-01:
I also tried that :(