OpenZWave/open-zwave #1506: Linker error MinGW.

Broekman, 2018-05-08:
When compiling openzwave with either mingw32 or mingw64, a linker error appears when compiling projects.

```
hid.o: In function `hid_enumerate':
C:\projects\open-zwave\cpp\build\windows\mingw32/../../../hidapi/windows/hid.cpp:278: undefined reference to `__imp_SetupDiGetClassDevsA'
C:\projects\open-zwave\cpp\build\windows\mingw32/../../../hidapi/windows/hid.cpp:287: undefined reference to `__imp_SetupDiEnumDeviceInterfaces'
C:\projects\open-zwave\cpp\build\windows\mingw32/../../../hidapi/windows/hid.cpp:480: undefined reference to `__imp_SetupDiDestroyDeviceInfoList'
C:\projects\open-zwave\cpp\build\windows\mingw32/../../../hidapi/windows/hid.cpp:302: undefined reference to `__imp_SetupDiGetDeviceInterfaceDetailA'
C:\projects\open-zwave\cpp\build\windows\mingw32/../../../hidapi/windows/hid.cpp:316: undefined reference to `__imp_SetupDiGetDeviceInterfaceDetailA'
C:\projects\open-zwave\cpp\build\windows\mingw32/../../../hidapi/windows/hid.cpp:336: undefined reference to `__imp_SetupDiEnumDeviceInfo'
C:\projects\open-zwave\cpp\build\windows\mingw32/../../../hidapi/windows/hid.cpp:340: undefined reference to `__imp_SetupDiGetDeviceRegistryPropertyA'
C:\projects\open-zwave\cpp\build\windows\mingw32/../../../hidapi/windows/hid.cpp:347: undefined reference to `__imp_SetupDiGetDeviceRegistryPropertyA'
Manager.o:Manager.cpp:(.rdata$.refptr.ozw_version_string[.refptr.ozw_version_string]+0x0): undefined reference to `ozw_version_string'
Manager.o:Manager.cpp:(.rdata$.refptr.ozw_vers_revision[.refptr.ozw_vers_revision]+0x0): undefined reference to `ozw_vers_revision'
Manager.o:Manager.cpp:(.rdata$.refptr.ozw_vers_minor[.refptr.ozw_vers_minor]+0x0): undefined reference to `ozw_vers_minor'
Manager.o:Manager.cpp:(.rdata$.refptr.ozw_vers_major[.refptr.ozw_vers_major]+0x0): undefined reference to `ozw_vers_major'
```

Also, I have to edit the Makefile regarding the vers.c from:
```echo 'char ozw_vers[] = "OpenZWave version 1.0.'`svnversion ../..`'```
to e.g.:
```echo char ozw_vers[] = "OpenZWave version 1.0."; > vers.c```
to prevent the following error from occurring:
```
vers.c:1:1: warning: character constant too long for its type
 'char ozw_vers[] = "OpenZWave version 1.0.'`svnversion ../..`'";'
 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
vers.c:1:1: error: expected identifier or '(' before '\x312e302e'
vers.c:1:44: error: stray '`' in program
 'char ozw_vers[] = "OpenZWave version 1.0.'`svnversion ../..`'";'
                                            ^
vers.c:1:61: error: stray '`' in program
 'char ozw_vers[] = "OpenZWave version 1.0.'`svnversion ../..`'";'
                                                             ^
vers.c:1:62: warning: multi-character character constant [-Wmultichar]
 'char ozw_vers[] = "OpenZWave version 1.0.'`svnversion ../..`'";'
                                                              ^~~
```

Any tips on how to get a working openzwave under mingw?

Fishwaldo, 2018-05-09:
Use the latest version? Looks like your using a very very old version

Broekman, 2018-05-09:
I freshly cloned the repository yesterday. I do see that the Makefile hasn't been updated for over a year.

Broekman, 2018-05-09:
I managed to get it to work. Created the following Makefile:

```
#
# Makefile for OpenZWave: MinGW-W64 build
# Maarten Damen
# Mal Lansell <openzwave@lansell.org>
# Stefan Broekman

.SUFFIXES:	.d .cpp .o .a
.PHONY:	default clean

#Temp version definitions
VERSION_MAJ ?= 1
VERSION_MIN ?= 4
VERSION := $(VERSION_MAJ).$(VERSION_MIN)
GITVERSION	:= $(VERSION_MAJ).$(VERSION_MIN).-1
VERSION_REV	:= 0

CC     := $(CROSS_COMPILE)gcc
CXX    := $(CROSS_COMPILE)g++
LD     := $(CROSS_COMPILE)g++
AR     := $(CROSS_COMPILE)ar rc
RANLIB := $(CROSS_COMPILE)ranlib

DEBUG_CFLAGS    := -Wall -Wno-unknown-pragmas -Wno-inline -Wno-format -Wno-error=sequence-point -Wno-sequence-point -ggdb -DDEBUG -fPIC
RELEASE_CFLAGS  := -Wall -Wno-unknown-pragmas -Wno-format -Wno-error=sequence-point -Wno-sequence-point -O3 -DNDEBUG -fPIC
DEBUG_LDFLAGS	:= -g

top_srcdir := ../../../..

STATIC_LIB_NAME=libopenzwave.a

# Change for DEBUG or RELEASE
CFLAGS	:= -c $(RELEASE_CFLAGS)

LIBDIR	:= .

INCLUDES := -I $(top_srcdir)/cpp/src -I $(top_srcdir)/cpp/tinyxml/ -I $(top_srcdir)/cpp/hidapi/hidapi/

SOURCES_HIDAPI =$(top_srcdir)/cpp/hidapi/windows

SOURCES		:= $(top_srcdir)/cpp/src $(top_srcdir)/cpp/src/command_classes $(top_srcdir)/cpp/tinyxml \
	$(top_srcdir)/cpp/src/value_classes $(top_srcdir)/cpp/src/platform $(top_srcdir)/cpp/src/platform/windows $(SOURCES_HIDAPI) $(top_srcdir)/cpp/src/aes/

VPATH = $(top_srcdir)/cpp/src:$(top_srcdir)/cpp/src/command_classes:$(top_srcdir)/cpp/tinyxml:\
	$(top_srcdir)/cpp/src/value_classes:$(top_srcdir)/cpp/src/platform:$(top_srcdir)/cpp/src/platform/windows:$(SOURCES_HIDAPI):$(top_srcdir)/cpp/src/aes/

%.d : %.cpp
	@$(CXX) -MM $(CFLAGS) $(INCLUDES) $< > $*.d
	
tinyxml := $(notdir $(wildcard $(top_srcdir)/cpp/tinyxml/*.cpp))
hidapi := $(notdir $(wildcard $(top_srcdir)/cpp/hidapi/linux/*.c)) # we do not want the libusb version
cclasses := $(notdir $(wildcard $(top_srcdir)/cpp/src/command_classes/*.cpp))
vclasses := $(notdir $(wildcard $(top_srcdir)/cpp/src/value_classes/*.cpp))
pform := $(notdir $(wildcard $(top_srcdir)/cpp/src/platform/*.cpp)) \
	$(notdir $(wildcard $(top_srcdir)/cpp/src/platform/unix/*.cpp))
indep := $(notdir $(filter-out $(top_srcdir)/cpp/src/vers.cpp, $(wildcard $(top_srcdir)/cpp/src/*.cpp)))
aes := $(notdir $(wildcard $(top_srcdir)/cpp/src/aes/*.c))

%.o : %.cpp
	$(CXX) $(CFLAGS) $(INCLUDES) -o $@ $<

%.o : %.c
	$(CC) $(CFLAGS) $(INCLUDES) -o $@ $<

default:	$(LIBDIR)/$(STATIC_LIB_NAME)
clean:		
	rm -rf *.d* *.o $(STATIC_LIB_NAME) $(top_srcdir)/cpp/src/vers.cpp

-include $(patsubst %.cpp,%.d,$(tinyxml))
-include $(patsubst %.c,%.d,$(hidapi))
-include $(patsubst %.cpp,%.d,$(cclasses))
-include $(patsubst %.cpp,%.d,$(vclasses))
-include $(patsubst %.cpp,%.d,$(pform))
-include $(patsubst %.cpp,%.d,$(indep))
-include $(patsubst %.c,%.d,$(aes))

$(top_srcdir)/cpp/src/vers.cpp:
	@echo "Creating vers.cpp"
	@echo #include "Defs.h" > $(top_srcdir)/cpp/src/vers.cpp
	@echo uint16_t ozw_vers_major = $(VERSION_MAJ); >> $(top_srcdir)/cpp/src/vers.cpp
	@echo uint16_t ozw_vers_minor = $(VERSION_MIN); >> $(top_srcdir)/cpp/src/vers.cpp
	@echo uint16_t ozw_vers_revision = $(VERSION_REV); >> $(top_srcdir)/cpp/src/vers.cpp
	@echo char ozw_version_string[] = "$(GITVERSION)"; >> $(top_srcdir)/cpp/src/vers.cpp

$(LIBDIR)/$(STATIC_LIB_NAME): $(patsubst %.cpp,%.o,$(tinyxml)) \
			$(patsubst %.c,%.o,$(hidapi)) \
			$(patsubst %.cpp,%.o,$(cclasses)) \
			$(patsubst %.cpp,%.o,$(vclasses)) \
			$(patsubst %.c,%.o,$(aes)) \
			$(patsubst %.cpp,%.o,$(pform)) \
			$(patsubst %.cpp,%.o,$(indep)) vers.o
	echo "Linking Static Library"
	$(AR) $@ $+
	$(RANLIB) $@
```

Also had to edit /cpp/src/Defs.h line 38 from to add support for MinGW64:
`#if (defined _WINDOWS || defined WIN32 || defined _MSC_VER) && !defined MINGW`
To:
`#if (defined _WINDOWS || defined WIN32 || defined _MSC_VER) && !defined MINGW && !defined __MINGW64__`

And finally, regarding the undefined references to the hidapi stuff whilst linking, I had to link "-lsetupapi" to my projects.

Broekman, 2018-05-10:
See [Pull request - 1507](https://github.com/OpenZWave/open-zwave/pull/1507).

closed 2018-07-06 by Fishwaldo