OpenZWave/open-zwave #1510: AlarmIndex_Type and AlarmIndex_Level events not being raised for Schlage BE469

ryanwinter, 2018-05-17:
After moving from the master to dev branch, I no longer have the AlarmIndex_Type and AlarmIndex_Level events being raised. Home Assistant uses both of these to communicate the current state of the lock.

Looking at the code in Alarm.cpp, I noticed that the events are still being raised in HandleMsg, but now they are restricted to version 1 via an if block. In the master branch, this code executes regardless of the version.

In the Dev branch, should the AlarmIndex_Type and AlarmIndex_Level events be raised regardless of the version to retain the same function as master?

Master:
		ValueByte* value;
		if( (value = static_cast<ValueByte*>( GetValue( _instance, AlarmIndex_Type ) )) )
		{
			value->OnValueRefreshed( _data[1] );
			value->Release();
		}
		if( (value = static_cast<ValueByte*>( GetValue( _instance, AlarmIndex_Level ) )) )
		{
			value->OnValueRefreshed( _data[2] );
			value->Release();
		}

Dev:
		if( GetVersion() == 1 )
		{
			Log::Write( LogLevel_Info, GetNodeId(), "Received Alarm report: type=%d, level=%d", _data[1], _data[2] );

			ValueByte* value;
			if( (value = static_cast<ValueByte*>( GetValue( _instance, AlarmIndex_Type ) )) )
			{
				value->OnValueRefreshed( _data[1] );
				value->Release();
			}
			// For device on version 1 the level could have different value. This level value correspond to a list of alarm type.
			if ( Value* value = GetValue( _instance, AlarmIndex_Level ) )
			{
				switch ( value->GetID().GetType() )
				{
				case ValueID::ValueType_Byte:
				{
					ValueByte* valueByte = static_cast<ValueByte*>( value );
					valueByte->OnValueRefreshed( _data[2] );
					break;
				}
				case ValueID::ValueType_List:
				{
					ValueList* valueList = static_cast<ValueList*>( value );
					valueList->OnValueRefreshed( _data[2] );
					break;
				}
				default:
				{
					Log::Write( LogLevel_Info, GetNodeId(), "Invalid type (%d) for Alarm Level %d", value->GetID().GetType(), _data[2] );
				}
				}
				value->Release();
			}
		}

dshokouhi, 2018-06-21:
@ryanwinter have you tried to find out why this lock is not reporting codes 24 and 25 for electronic lock/unlock reporting?  It should be a Alarm Type being sent back but I do not see it for this lock.  Manual and Keypad codes are working fine.

Fishwaldo, 2018-07-06:
What Version of the Notification (formally Alarm CC) does this device report. (should be in the zwcfg_*.xml file. 

Alarm V1 is really really old. Newer Notification classes have predefined types for alarms messages that this device should report intead. (the old way in Alarm V1 is depreciated by the Specs)

Fishwaldo, 2018-07-27:
ping?

Fishwaldo, 2018-08-03:
Closing as no response.

closed 2018-08-03 by Fishwaldo

ryanwinter, 2018-08-17:
Sorry for the slow response, I just had a baby!

Makes sense that v1 is deprecated. I will work with home-assistant to upgrade to v2.