OpenZWave/open-zwave #1521: Checksum Aeonlabs Hem and OpenZwave

ado464, 2018-05-30:
Hi
I have been in discussion with Chris at Aeonlabs.
He reports that there is a  checksum implemented in the energy meters.
So the question is, is there any verification from the openzwave side? Since the meters can send glitched out values where the decimal has moved eather to the right or left creating an absurd value.

best regards,
Ado

Fishwaldo, 2018-06-24:
Yes - We do checksome's on the packets. Corrupt Packets will not be processed.

closed 2018-06-24 by Fishwaldo

LoonSongSoftware, 2018-07-02:
I’ve looked at thousands of packets over the years. I note a lot more garbled data values (like your megawatt reading) than bad checksums. In fact, I don’t recall ever seeing a bad checksum. 

If you’ve been logging the results, I encourage you to send the packets (especially the Report packet with .the bad data and the good checksum) to Aeon. I can’t explain it in any way other than a problem in Aeon’s device firmware.