OpenZWave/open-zwave #1524: Cooper rfwc5.xml missing command classes to link Scene ID to Group ID

aviphysics, 2018-06-04:
The Cooper rfwc5.xml missing command classes to link Scene ID to Group ID. Setting these parameters is required to have the device work as a scene controller.

Command_class: COMMAND_CLASS_SCENE_CONTROLLER_CONF_V1
Command_name: SCENE_CONTROLLER_CONF_SET
the variables are:
-Group ID (1:255)
-Scene ID (1:255)
-Dimming Duration (1:255)
Screenshot of command in zensys-tools: https://imgur.com/FfrSZV0.png

There is also a Command_Class for setting the indicators, which would be helpful to have.
Command_class: COMMAND_CLASS_INDICATOR
Command_Name: INDICATOR_SET
variable: Value (0 or 1)
Screenshot of command in zensys-tools: https://imgur.com/JDpvuNG.png

nechry, 2018-06-04:
the COMMAND_CLASS_INDICATOR already exist and is implemented it's should be work do you get the value?

the 2 other CC are not yet implemented any help will be appreciate

aviphysics, 2018-06-08:
You are right about the indicators, though now I need to figure out how to implement it in Home Assistant?