OpenZWave/open-zwave #1536: Possible Bug: Aeotec DSB05 Gen2 MultiSensor, values not updated and errors in logging.

RogierSchuring, 2018-06-19:
Errors recieved when querying Aeotec DSB05 MultiSensor.
See diagnosed logs: http://www.openzwave.com/log-analyzer/public/bc9c3433-38b5-46e2-af32-8f6ca7efbab9/
(Using domoticz, I am unable to generate log files without the zwcfg_*.xml file. After removing, it is recreated every minute)

Hardware:
Problem node: Aeotec DSB05 Gen2 MultiSensor
Controller: Aeotec ZW090 Z-Stick Gen5
Device: rpi3
Software: Domoticz (latest stable: 3.8153)
Open-zwave version: 1.4-2501-g5af6780-dirty

If more info is needed, let me know.
If anyone knows how to get logging without the zwcfg_*.xml file using domoticz, also please let me know so I can upload a new, complete log.

Fishwaldo, 2018-07-06:
Hi,
This is a issue tracker and not a support forum. 

Please use our Forums for Questions regarding OpenZWave: 
https://groups.google.com/forum/#!forum/openzwave

Thanks

closed 2018-07-06 by Fishwaldo

RogierSchuring, 2018-07-12:
Okay, so errors (issues) in openzwave are support forum issues?