OpenZWave/open-zwave #1541: Fibaro Roller Shutter 2 spurious GET when a SET is sent

jonferreira, 2018-06-21:
Been living with this issue since I made the move to Domoticz, but only now it has become really annoying given that I can't use any of these devices as a "scene trigger".

Have a couple of FIBARO FGRM222 Roller Shutter Controller 2 which work "fine" except for the fact when once controls them via Z-Wave it sends 2 "spurious" messages

```
2018-06-21 09:11:39.057 Info, Node021, Value::Set - COMMAND_CLASS_SWITCH_MULTILEVEL - Level - 0 - 1 - 70
2018-06-21 09:11:39.058 Info, Node021, SwitchMultilevel::Set - Setting to level 70
2018-06-21 09:11:39.059 Info, Node021,   Duration: Default
2018-06-21 09:11:39.060 Detail, Node021, Queuing (Send) SwitchMultilevelCmd_Set (Node=21): 0x01, 0x0b, 0x00, 0x13, 0x15, 0x04, 0x26, 0x01, 0x46, 0xff, 0x25, 0xf4, 0xb9
2018-06-21 09:11:39.061 Detail, Node021, Queuing (Send) SwitchMultilevelCmd_Get (Node=21): 0x01, 0x09, 0x00, 0x13, 0x15, 0x02, 0x26, 0x02, 0x25, 0xf5, 0x06
2018-06-21 09:11:39.062 Detail, 
2018-06-21 09:11:39.062 Info, Node021, Sending (Send) message (Callback ID=0xf4, Expected Reply=0x13) - SwitchMultilevelCmd_Set (Node=21): 0x01, 0x0b, 0x00, 0x13, 0x15, 0x04, 0x26, 0x01, 0x46, 0xff, 0x25, 0xf4, 0xb9
2018-06-21 09:11:39.071 Detail, Node021,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2018-06-21 09:11:39.072 Detail, Node021,   ZW_SEND_DATA delivered to Z-Wave stack
2018-06-21 09:11:39.089 Detail, Node021,   Received: 0x01, 0x07, 0x00, 0x13, 0xf4, 0x00, 0x00, 0x02, 0x1d
2018-06-21 09:11:39.089 Detail, Node021,   ZW_SEND_DATA Request with callback ID 0xf4 received (expected 0xf4)
2018-06-21 09:11:39.090 Info, Node021, Request RTT 27 Average Request RTT 26
2018-06-21 09:11:39.091 Detail,   Expected callbackId was received
2018-06-21 09:11:39.092 Detail,   Expected reply was received
2018-06-21 09:11:39.093 Detail,   Message transaction complete
2018-06-21 09:11:39.094 Detail, 
2018-06-21 09:11:39.094 Detail, Node021, Removing current message
2018-06-21 09:11:39.095 Detail, 
2018-06-21 09:11:39.095 Info, Node021, Sending (Send) message (Callback ID=0xf5, Expected Reply=0x04) - SwitchMultilevelCmd_Get (Node=21): 0x01, 0x09, 0x00, 0x13, 0x15, 0x02, 0x26, 0x02, 0x25, 0xf5, 0x06
2018-06-21 09:11:39.105 Detail, Node021,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2018-06-21 09:11:39.105 Detail, Node021,   ZW_SEND_DATA delivered to Z-Wave stack
2018-06-21 09:11:39.122 Detail, Node021,   Received: 0x01, 0x07, 0x00, 0x13, 0xf5, 0x00, 0x00, 0x02, 0x1c
2018-06-21 09:11:39.122 Detail, Node021,   ZW_SEND_DATA Request with callback ID 0xf5 received (expected 0xf5)
2018-06-21 09:11:39.122 Info, Node021, Request RTT 25 Average Request RTT 25
2018-06-21 09:11:39.123 Detail,   Expected callbackId was received
2018-06-21 09:11:39.132 Detail, Node021,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x15, 0x03, 0x26, 0x03, 0x32, 0xf3
2018-06-21 09:11:39.132 Detail, 
2018-06-21 09:11:39.133 Info, Node021, Response RTT 36 Average Response RTT 36
2018-06-21 09:11:39.133 Info, Node021, Received SwitchMultiLevel report: level=50
2018-06-21 09:11:39.133 Detail, Node021, Refreshed Value: old value=50, new value=50, type=byte
2018-06-21 09:11:39.134 Detail, Node021, Changes to this value are not verified
2018-06-21 09:11:39.134 Detail, Node021,   Expected reply and command class was received
2018-06-21 09:11:39.135 Detail, Node021,   Message transaction complete
2018-06-21 09:11:39.135 Detail, 
2018-06-21 09:11:39.135 Detail, Node021, Removing current message
2018-06-21 09:11:39.136 Detail, Node021, Notification: ValueChanged
2018-06-21 09:11:40.249 Detail, Node021,   Received: 0x01, 0x0c, 0x00, 0x04, 0x00, 0x15, 0x06, 0x31, 0x05, 0x04, 0x22, 0x05, 0x55, 0xa6
2018-06-21 09:11:40.249 Detail, 
2018-06-21 09:11:40.249 Info, Node021, Received SensorMultiLevel report from node 21, instance 1, Power: value=136.5W
2018-06-21 09:11:40.250 Detail, Node021, Refreshed Value: old value=0.0, new value=136.5, type=decimal
2018-06-21 09:11:40.250 Detail, Node021, Changes to this value are not verified
2018-06-21 09:11:40.251 Detail, Node021, Notification: ValueChanged
2018-06-21 09:11:45.305 Detail, Node021,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x15, 0x03, 0x26, 0x03, 0x46, 0x87
2018-06-21 09:11:45.306 Detail, 
2018-06-21 09:11:45.306 Info, Node021, Received SwitchMultiLevel report: level=70
2018-06-21 09:11:45.306 Detail, Node021, Refreshed Value: old value=50, new value=70, type=byte
2018-06-21 09:11:45.306 Detail, Node021, Changes to this value are not verified
2018-06-21 09:11:45.306 Detail, Node021, Notification: ValueChanged
2018-06-21 09:11:45.325 Detail, Node021,   Received: 0x01, 0x0c, 0x00, 0x04, 0x00, 0x15, 0x06, 0x31, 0x05, 0x04, 0x22, 0x00, 0x00, 0xf6
2018-06-21 09:11:45.325 Detail, 
2018-06-21 09:11:45.325 Info, Node021, Received SensorMultiLevel report from node 21, instance 1, Power: value=0.0W
2018-06-21 09:11:45.326 Detail, Node021, Refreshed Value: old value=136.5, new value=0.0, type=decimal
2018-06-21 09:11:45.326 Detail, Node021, Changes to this value are not verified
2018-06-21 09:11:45.326 Detail, Node021, Notification: ValueChanged

```

It seems that OZW sends a Set and a Get immediatly

```
2018-06-21 09:11:39.060 Detail, Node021, Queuing (Send) SwitchMultilevelCmd_Set (Node=21): 0x01, 0x0b, 0x00, 0x13, 0x15, 0x04, 0x26, 0x01, 0x46, 0xff, 0x25, 0xf4, 0xb9
2018-06-21 09:11:39.061 Detail, Node021, Queuing (Send) SwitchMultilevelCmd_Get (Node=21): 0x01, 0x09, 0x00, 0x13, 0x15, 0x02, 0x26, 0x02, 0x25, 0xf5, 0x06
```

I would imagine that the Set is Acked (and the blind starts to move) but the Get is then processed by the module immediately which returns the current (old) value). Then once the blind has finished moving it reports it's current level.

PS: related with https://github.com/domoticz/domoticz/issues/2500#issuecomment-399028716

Fishwaldo, 2018-06-24:
The Device is not in accordance with the Spec. To work around that, use Manager::SetChangeVerified on the ValueID in question, and it should keep querying till the Value Stabalizes.

jonferreira, 2018-06-25:
thinks for replying @Fishwaldo 

in all honesty I'm not so sure what to do with this information though :(

gizmocuz, 2018-06-25:
@Fishwaldo, is this something we can configure in the node configuration file ?

jonferreira, 2018-07-04:
bump in case @Fishwaldo missed this

Fishwaldo, 2018-07-06:
You could create the Value's in the device config file (like danfoss/rsroom.xml) but with the ValueID's that are affected, set verify_changes="true"

But I'm not 100% sure if that will work.

jonferreira, 2018-07-12:
@Fishwaldo @gizmocuz  sorry for the delay, I've not seen any notification that you had replied!

I've looked at rsroom.xml and suppose you're referring to

> <CommandClass base="0" override_precision="2" request_flags="4" version="1" name="COMMAND_CLASS_THERMOSTAT_SETPOINT" id="67">
>     <Instance index="1"/>
>     <Value index="1" value="0.0" max="0" min="0" verify_changes="false" write_only="false" read_only="false" units="C" label="Heating" instance="1" genre="user" type="decimal"/>
>     <Value index="2" value="0.0" max="0" min="0" verify_changes="false" write_only="false" read_only="false" units="C" label="Cooling" instance="1" genre="user" type="decimal"/>
>     <Value index="3" value="0.0" max="0" min="0" verify_changes="false" write_only="false" read_only="false" units="C" label="Unused" instance="1" genre="user" type="decimal"/>
>   </CommandClass>

I'm unsure how, or better, what values I should use on the FGRM222.xml file though

Fishwaldo, 2018-07-16:
You need to modify it according to your device. rsroom.xml is just a example of how to use verify_changes.

jonferreira, 2018-07-16:
Yes point is I'm not sure "how" to use it.

The FGRM-222 xls only have Command Class 112 and 113 (configuration and associations) so presumably I'll need to add another CommandClass section for Multilevel (?) where I'll use the verify_changes=true?

jonferreira, 2018-07-16:
@Fishwaldo just realised how dumb I was being!

So I've edited my zwcfg_X.xml file and set

<CommandClass id="38" name="COMMAND_CLASS_SWITCH_MULTILEVEL" version="3" innif="true">
                <Instance index="1" />
                <Value type="byte" genre="user" instance="1" index="0" label="Level" units="" read_only="false" write_only="false" verify_changes="true" poll_intensity="0" min="0" max="255" value="67" />
                <Value type="button" genre="user" instance="1" index="1" label="Open" units="" read_only="false" write_only="true" verify_changes="false" poll_intensity="0" min="0" max="0" />
                <Value type="button" genre="user" instance="1" index="2" label="Close" units="" read_only="false" write_only="true" verify_changes="false" poll_intensity="0" min="0" max="0" />
                <Value type="bool" genre="system" instance="1" index="3" label="Ignore Start Level" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="True" />
                <Value type="byte" genre="system" instance="1" index="4" label="Start Level" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="0" />
                <Value type="byte" genre="system" instance="1" index="5" label="Dimming Duration" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="255" />
                <Value type="byte" genre="user" instance="1" index="6" label="Step Size" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="0" />
                <Value type="button" genre="user" instance="1" index="7" label="Inc" units="" read_only="false" write_only="true" verify_changes="false" poll_intensity="0" min="0" max="0" />
                <Value type="button" genre="user" instance="1" index="8" label="Dec" units="" read_only="false" write_only="true" verify_changes="false" poll_intensity="0" min="0" max="0" />
            </CommandClass>

Behaviour has changed slight but still sending a GET immediately after the SET

2018-07-16 13:13:28.455 Info, Node021, Value::Set - COMMAND_CLASS_SWITCH_MULTILEVEL - Level - 0 - 1 - 49
2018-07-16 13:13:28.456 Info, Node021, SwitchMultilevel::Set - Setting to level 49
2018-07-16 13:13:28.456 Info, Node021,   Duration: Default
2018-07-16 13:13:28.456 Detail, Node021, Queuing (Send) SwitchMultilevelCmd_Set (Node=21): 0x01, 0x0b, 0x00, 0x13, 0x15, 0x04, 0x26, 0x01, 0x31, 0xff, 0x25, 0x54, 0x6e
2018-07-16 13:13:28.456 Detail, Node021, Queuing (Send) SwitchMultilevelCmd_Get (Node=21): 0x01, 0x09, 0x00, 0x13, 0x15, 0x02, 0x26, 0x02, 0x25, 0x55, 0xa6
2018-07-16 13:13:28.457 Detail, 
2018-07-16 13:13:28.457 Info, Node021, Sending (Send) message (Callback ID=0x54, Expected Reply=0x13) - SwitchMultilevelCmd_Set (Node=21): 0x01, 0x0b, 0x00, 0x13, 0x15, 0x04, 0x26, 0x01, 0x31, 0xff, 0x25, 0x54, 0x6e
2018-07-16 13:13:28.467 Detail, Node021,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2018-07-16 13:13:28.467 Detail, Node021,   ZW_SEND_DATA delivered to Z-Wave stack
2018-07-16 13:13:28.484 Detail, Node021,   Received: 0x01, 0x07, 0x00, 0x13, 0x54, 0x00, 0x00, 0x03, 0xbc
2018-07-16 13:13:28.485 Detail, Node021,   ZW_SEND_DATA Request with callback ID 0x54 received (expected 0x54)
2018-07-16 13:13:28.485 Info, Node021, Request RTT 27 Average Request RTT 296
2018-07-16 13:13:28.485 Detail,   Expected callbackId was received
2018-07-16 13:13:28.485 Detail,   Expected reply was received
2018-07-16 13:13:28.485 Detail,   Message transaction complete
2018-07-16 13:13:28.486 Detail, 
2018-07-16 13:13:28.486 Detail, Node021, Removing current message
2018-07-16 13:13:28.486 Detail, 
2018-07-16 13:13:28.486 Info, Node021, Sending (Send) message (Callback ID=0x55, Expected Reply=0x04) - SwitchMultilevelCmd_Get (Node=21): 0x01, 0x09, 0x00, 0x13, 0x15, 0x02, 0x26, 0x02, 0x25, 0x55, 0xa6
2018-07-16 13:13:28.495 Detail, Node021,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2018-07-16 13:13:28.496 Detail, Node021,   ZW_SEND_DATA delivered to Z-Wave stack
2018-07-16 13:13:28.512 Detail, Node021,   Received: 0x01, 0x07, 0x00, 0x13, 0x55, 0x00, 0x00, 0x02, 0xbc
2018-07-16 13:13:28.512 Detail, Node021,   ZW_SEND_DATA Request with callback ID 0x55 received (expected 0x55)
2018-07-16 13:13:28.513 Info, Node021, Request RTT 26 Average Request RTT 161
2018-07-16 13:13:28.513 Detail,   Expected callbackId was received
2018-07-16 13:13:28.522 Detail, Node021,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x15, 0x03, 0x26, 0x03, 0x00, 0xc1
2018-07-16 13:13:28.523 Detail, 
2018-07-16 13:13:28.523 Info, Node021, Response RTT 36 Average Response RTT 41
2018-07-16 13:13:28.523 Info, Node021, Received SwitchMultiLevel report: level=0
2018-07-16 13:13:28.523 Detail, Node021, Refreshed Value: old value=0, new value=0, type=byte
2018-07-16 13:13:28.524 Detail, Node021, Changes to this value are not verified
2018-07-16 13:13:28.524 Detail, Node021,   Expected reply and command class was received
2018-07-16 13:13:28.524 Detail, Node021,   Message transaction complete
2018-07-16 13:13:28.524 Detail, 
2018-07-16 13:13:28.525 Detail, Node021, Removing current message
2018-07-16 13:13:28.525 Detail, Node021, Notification: ValueChanged
2018-07-16 13:13:29.684 Detail, Node021,   Received: 0x01, 0x0c, 0x00, 0x04, 0x00, 0x15, 0x06, 0x31, 0x05, 0x04, 0x22, 0x05, 0x6a, 0x99
2018-07-16 13:13:29.685 Detail, 
[... cut ...]
2018-07-16 13:13:42.686 Detail, Node021,   Received: 0x01, 0x0c, 0x00, 0x04, 0x00, 0x15, 0x06, 0x31, 0x05, 0x04, 0x22, 0x00, 0x00, 0xf6
2018-07-16 13:13:42.688 Detail, 
2018-07-16 13:13:42.689 Info, Node021, Received SensorMultiLevel report from node 21, instance 1, Power: value=0.0W
2018-07-16 13:13:42.690 Detail, Node021, Refreshed Value: old value=138.6, new value=0.0, type=decimal
2018-07-16 13:13:42.691 Detail, Node021, Changes to this value are not verified
2018-07-16 13:13:42.691 Detail, Node021, Notification: ValueChanged

Fishwaldo, 2018-08-03:
> Behaviour has changed slight but still sending a GET immediately after the SET

It will always do that... but the config change I suggested would keep doing a GET till the value is consistent. 

But it didn't work - Seems we can't do it from a config file:

> 2018-07-16 13:13:28.524 Detail, Node021, Changes to this value are not verified

You can either edit the zwcfg_*.xml file to set that option on the ValueID for the device, or via your application, use the Manager::SetChangeVerified method.

jonferreira, 2018-08-03:
@Fishwaldo sorry for being noob but not sure **exactly** what you mean by "to set that option on the ValueID for the device"

my zwcfg_*.xml file looks like this for node21

```
<Node id="21" name="Office Roller S" location="" basic="4" generic="17" specific="6" type="Motor Control Class B" listening="true" frequentListening="false" beaming="true" routing="true" max_baud_rate="40000" version="4" query_stage="Complete">
        <Manufacturer id="10f" name="FIBARO System">
            <Product type="301" id="1001" name="FGRM222 Roller Shutter Controller 2" />
        </Manufacturer>
        <CommandClasses>
            <CommandClass id="32" name="COMMAND_CLASS_BASIC" version="1" request_flags="4" mapping="38">
                <Instance index="1" />
            </CommandClass>
            <CommandClass id="37" name="COMMAND_CLASS_SWITCH_BINARY" version="1" request_flags="4" innif="true">
                <Instance index="1" />
                <Value type="bool" genre="user" instance="1" index="0" label="Switch" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False" />
            </CommandClass>
            <CommandClass id="38" name="COMMAND_CLASS_SWITCH_MULTILEVEL" version="3" innif="true">
                <Instance index="1" />
                <Value type="byte" genre="user" instance="1" index="0" label="Level" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="50" />
                <Value type="button" genre="user" instance="1" index="1" label="Open" units="" read_only="false" write_only="true" verify_changes="false" poll_intensity="0" min="0" max="0" />
                <Value type="button" genre="user" instance="1" index="2" label="Close" units="" read_only="false" write_only="true" verify_changes="false" poll_intensity="0" min="0" max="0" />
                <Value type="bool" genre="system" instance="1" index="3" label="Ignore Start Level" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="True" />
                <Value type="byte" genre="system" instance="1" index="4" label="Start Level" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="0" />
                <Value type="byte" genre="system" instance="1" index="5" label="Dimming Duration" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="255" />
                <Value type="byte" genre="user" instance="1" index="6" label="Step Size" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="0" />
                <Value type="button" genre="user" instance="1" index="7" label="Inc" units="" read_only="false" write_only="true" verify_changes="false" poll_intensity="0" min="0" max="0" />
                <Value type="button" genre="user" instance="1" index="8" label="Dec" units="" read_only="false" write_only="true" verify_changes="false" poll_intensity="0" min="0" max="0" />
            </CommandClass>
            <CommandClass id="43" name="COMMAND_CLASS_SCENE_ACTIVATION" version="1" request_flags="4" after_mark="true" innif="true">
                <Instance index="1" />
            </CommandClass>
            <CommandClass id="49" name="COMMAND_CLASS_SENSOR_MULTILEVEL" version="2" innif="true">
                <Instance index="1" />
                <Value type="decimal" genre="user" instance="1" index="4" label="Power" units="W" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="0.0" />
            </CommandClass>
            <CommandClass id="50" name="COMMAND_CLASS_METER" version="2" request_flags="2" innif="true">
                <Instance index="1" />
                <Value type="decimal" genre="user" instance="1" index="0" label="Energy" units="kWh" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="2.73" />
                <Value type="decimal" genre="user" instance="1" index="8" label="Power" units="W" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="0.0" />
                <Value type="bool" genre="user" instance="1" index="32" label="Exporting" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False" />
                <Value type="button" genre="system" instance="1" index="33" label="Reset" units="" read_only="false" write_only="true" verify_changes="false" poll_intensity="0" min="0" max="0" />
            </CommandClass>
            <CommandClass id="112" name="COMMAND_CLASS_CONFIGURATION" version="1" request_flags="4" innif="true">
                <Instance index="1" />
                <Value type="list" genre="config" instance="1" index="1" label="Local Protection" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" vindex="0" size="1">
                    <Help>Once the Local Protection is activated, the module stops responding to S1 and S2 push buttons. SCENE ID and association commands will not be sent as well. The only exception is the B-button. Menu and Z-Wave network inclusion, after the B-button or S1 push button triple click, are still active.</Help>
[... cut ...]
                <Value type="string" genre="system" instance="1" index="1" label="Protocol Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="3.52" />
                <Value type="string" genre="system" instance="1" index="2" label="Application Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="22.22" />
            </CommandClass>
        </CommandClasses>
    </Node>
```