OpenZWave/open-zwave #1563: Support for firmware update

RadoslavMarinov, 2018-07-31:
Is there support for over the air (OTA) firmware update of devices. COMMAND_CLASS_FIRMWARE_UPDATE
I was searching [http://www.openzwave.com/dev/index.html](http://www.openzwave.com/dev/index.html) but I can not find API to support it.

Fishwaldo, 2018-08-01:
No. someone needs to write it.

vincemic, 2018-10-31:
Aeotech just announced that they worked with HomeSeer to get OTA updates working for at least their devices.  Bitdog LLC would be willing to donate a small bounty to this feature development. Would anyone else be willing to contribute to a bounty?

aromanro, 2018-11-14:
Are there really firmware files available for devices? The Aeotec ones I have can have the firmware updated using their own executables, the firmwares are not available separately.

Fishwaldo, 2018-11-14:
You can download "hec" files from the support site at Aeotec... but I had a cursory glance at the headers and couldn't deduce the format.. I suspect they might be encrypted.

rrozema, 2019-01-20:
zwave.me has updates available for at least their razberry board. These are available for download from the z-way.me software. I installed the latest firmware into my board and got much snappier response to commands issued to all devices in my network. Selected other devices are also showing an option to have a firmware update uploaded. My guess is that this is based on detected command classes. A.o. my fibaro double switch 2 and dimmer 2 are shown to have this update feature. Don't know where to get the files to upload though, nor if they even can be downloaded anywhere.

RicardP, 2019-05-13:
I got a new FW for my Philio PAN04-01B a year ago or so, but to be able to use it I had to purchase the HomeSeer Z-Flash utility. It was the only (or most straight forward) universal software I could find, at that time.
It worked fine but not the smoothest solution to first need to stop the running application (Domoticz) and then start Z-Flash, select COM-port etc... 

OZW built in feature would of course be very appreciated, I believe also it would in some way promote manufactures to release their FW more often if it is not giving them headache due to lack in/complicated flashing options...

Thanks!

niemyjski, 2020-03-29:
Supporting OTA updates would be amazing.

sveip, 2020-04-09:
I'm also willing to contribute to a bounty for this feature. If there is a doc on how to help on the coding part, I can also do my best there,

cliffbraun, 2020-04-30:
+1 to this, Zooz files aren't available publicly, but they'll readily email updates to purchasers.

LordMike, 2020-06-22:
@Fishwaldo I was reading up on the protocol spec for `COMMAND_CLASS_FIRMWARE_UPDATE_MD`, and I noticed the following.

> The Firmware Update Meta Data Command Class may be used to transfer a firmware image to or from a Z-Wave node.

In principal, if the node supports it, it should be possible to fetch the firmwares from the devices themselves. In theory, a community effort could gather all firmwares from devices .. :)

Also. In [another issue](https://github.com/OpenZWave/open-zwave/issues/1057#issuecomment-267900174), you mentioned the firmwares are embedded and not extractable. At least for Aeotec/AEON labs, their software is a .NET binary which can easily be decompiled using Jetbrains Dotpeek. In it, I found a resources section (.NET resources) which contains base64 encodings of what I presume is the firmware.

I've written (also) to Aeotec asking for the firmwares themselves, but after reading other comments I think that's a non-starter.. sadly.

sveip, 2020-06-23:
Aeotec are happy to send you the firmware files in my experience. They sent
me the Doorbell 6 fw files when I asked.

LordMike, 2020-06-23:
I got a reply from Aeotec, saying that the HEC files are actually encrypted .. damn. Apparently HomeSeer can decrypt them... Why the f....

Anyhoo. I can get the HEC files, but they're not useful for me until we can decrypt them. 

Aeotec did say they were working on using a feature of the Z-wave chip, to allow that to decrypt the binaries instead. Meaning we can happily ship encrypted firmwares, as if they were just binaries..

What a stinking pile of poo... :|

sveip, 2020-06-23:
Maybe I got this wrong, but they sent me an archive (OTA of ZW162_Doorbell
6_EU_V1_04.zip) that I flashed the product with from the Silabs SW. AFIK
it's the standard FW upgrade mechanism supported by Z-wave?

LordMike, 2020-06-23:
Nice. Could you share it with me, @sveip? .. I have that product too.. so I'd be covered there :)

sveip, 2020-06-23:
I will look for the file when I get home, it was a wetransfer link, and
when I checked now it was expired. It is hopefully still on my laptop.

LordMike, 2020-08-05:
Just for good measure, I received 1.06 firmwares for ZW162 and ZW164

[OTA of ZW162_Doorbell 6_EU_V1_06.zip](https://github.com/OpenZWave/open-zwave/files/5029809/OTA.of.ZW162_Doorbell.6_EU_V1_06.zip)

[OTA of ZW164_Siren 6_EU_V1_06.zip](https://github.com/OpenZWave/open-zwave/files/5029813/OTA.of.ZW164_Siren.6_EU_V1_06.zip)

Mikes.

marleyjaffe, 2020-09-04:
The current upgrade process for Inovelli devices FYI: https://support.inovelli.com/portal/en/kb/articles/how-to-s-general-how-to-update-firmware-z-stick-z-wave-pc-controller#What_Youll_Need

tokyotexture, 2020-09-06:
Interesting development over at Hubitat on the same topic. https://community.hubitat.com/t/sneak-peek-z-wave-firmware-updating-inside-he/35836/26

RicardP, 2020-09-06:
@tokyotexture yes that was very interesting reading!

OverZealous, 2020-10-14:
It probably doesn't help much, but Zooz has basically the same instructions for OTA updates:  
https://www.support.getzooz.com/kb/article/253-how-to-perform-an-ota-firmware-update-on-zooz-devices/

The problem for me is I don't have easy access to a Windows machine, so while I have the new firmware for my switches, I cannot update them without borrowing a computer. So I would love to see this supported directly with OZW (which I'm using on HomeAssistant).

mpb-studios, 2020-12-21:
Is there any update on this?

silverairedale, 2021-04-19:
HEC encryption is AES256 CBC using key `b"\xd7\xa6\x8d\xef\x0fJ\x12A\x94\x0fl\xb8\x01q!\xd1_\x0e&\x82\xe2X\xc9\xf7U>pn\x83I#\xb7"` and IV `b"\x0ee\x19)u0X7\x08a*(#f8D"` (Python byte string format). These values are generated when executing the Homeseer flash utility.

https://gist.github.com/silverairedale/e560eebcace43e95da7c74aa82dd824c

VaillantHassIo, 2022-01-13:
any news here? Where can i join to raise the bounty?

LordMike, 2022-01-13:
I'm unsubscribing from this - I've moved wholly to zwavejs, which _does_ support OTA updates, even from a number of formats including Aeotec's exe files.