OpenZWave/open-zwave #1574: Device config file entry not used.

kdschlosser, 2018-08-06:
I have a GE Dimmer model 12724 and the level min and max are not correct. the dimmer ranges from 0 to 99. the current min is 0 and the max is 255 if a setting above 99 is made the dimmer does nothing.

I am modified the 12724-dimmer.xml file to include the following xml data

    <CommandClass id="38">
        <Value type="byte" index="0" genre="user" label="Level" min="0" max="99" value="0" size="1">
            <Help>Sets the light level.</Help>
        </Value>
    </CommandClass>

I deleted all cached information and user configuration files. openzwave does not use the configuration data for setting the light level. it still has a max of 255.

I am not sure if I am doing something incorrectly or if there is an issue. I checked the data type as well as the index and label they are all correct I am assuming that the size would be a one because it is a single byte that is used for the level. I also confirmed the command class as well.

Any help would be appreciated

Thanks again
Kevin

nechry, 2018-08-07:
The min max  attributes are not use as threshold values during set operation. Also a set level with 255 will restore the previous levels on a dimmer, the max with 255 is good

closed 2018-08-07 by nechry

kdschlosser, 2018-08-07:
what i am guessing from you statement is that if i set the level to 255 from an off state it will put the light back to the level it was at before the light was turned off??? I will have to test to see if my dimmers do this. Do all dimmers operate in a range of 0-99 and not do anything from 100 to 254?

Then my next question is what is the purpose to the min and max if not for setting boundaries?