OpenZWave/open-zwave #1589: Versioning confusion

Nico0084, 2018-08-28:
Hi,
 
It seems to me that there is possible confusion between version 1.4 (master) and 1.5 (dev)
The log file for the 2 branches is the same and deals with version 1.5.
Outside the compile of the master branch creates a version 1.4.
Can you clarify this point.

Thks,

Fishwaldo, 2018-08-31:
I  can’t see how this happens. If you checked out from Git, did you make clean between builds from different branches? Vers.c gets created by the makefile.

Nico0084, 2018-09-02:
Sorry, I did not express myself clearly !

Problem is between [Change log file](https://github.com/OpenZWave/open-zwave/blob/master/ChangeLog) in master branch and build from master.
ChangeLog file indicate version 1.5 and my compiled vers.cpp : 

> #include "Defs.h"
> uint16_t ozw_vers_major = 1;
> uint16_t ozw_vers_minor = 4;
> uint16_t ozw_vers_revision = 0;
> char ozw_version_string[] = "1.4.-1";

Make clean was do before compile.