OpenZWave/open-zwave #1593: Can't build master and Dev version on FreeBSD v11.2

megapearl, 2018-09-04:
Hi,

I can't seem to build on FreeBSD, tried the master and the Dev version.
How to debug?

```
root@fileserver:/home/donald # git clone -b Dev https://github.com/OpenZWave/open-zwave.git
Cloning into 'open-zwave'...
remote: Counting objects: 28544, done.
remote: Compressing objects: 100% (5/5), done.
remote: Total 28544 (delta 0), reused 4 (delta 0), pack-reused 28539
Receiving objects: 100% (28544/28544), 106.97 MiB | 6.18 MiB/s, done.
Resolving deltas: 100% (20001/20001), done.
root@fileserver:/home/donald # cd open-zwave/
root@fileserver:/home/donald/open-zwave # gmake
LDFLAGS="" CPPFLAGS="" gmake -C /home/donald/open-zwave/cpp/build/ -
gmake[1]: Entering directory '/home/donald/open-zwave/cpp/build'
Building OpenZWave Version V1.5-3342-g8d2c743d-dirty
Building tinyxmlparser.o
Building tinyxml.o
Building tinyxmlerror.o
Building tinystr.o
Building hid.o
Building aes_modes.o
Building aestab.o
Building aescrypt.o
Building aeskey.o
Building SwitchToggleMultilevel.o
Building Protection.o
Building DoorLockLogging.o
Building CommandClass.o
Building EnergyProduction.o
Building NoOperation.o
Building ThermostatSetpoint.o
Building CommandClasses.o
Building Alarm.o
Building Clock.o
Building ZWavePlusInfo.o
Building Security.o
Building Version.o
Building CentralScene.o
Building SwitchMultilevel.o
Building AssociationCommandConfiguration.o
Building DoorLock.o
Building Powerlevel.o
Building SceneActivation.o
Building MultiCmd.o
Building ThermostatMode.o
Building Lock.o
Building SwitchAll.o
Building ManufacturerProprietary.o
Building Meter.o
Building ThermostatOperatingState.o
Building Color.o
Building NodeNaming.o
Building ControllerReplication.o
Building ApplicationStatus.o
Building Hail.o
Building ClimateControlSchedule.o
Building Association.o
Building ThermostatFanState.o
Building ManufacturerSpecific.o
Building TimeParameters.o
Building UserCode.o
Building DeviceResetLocally.o
Building SensorMultilevel.o
/home/donald/open-zwave/cpp/src/command_classes/SensorMultilevel.cpp:501:55: error: adding 'uint8' (aka 'unsigned char') to a string does not append to the string [-Werror,-Wstring-plus-int]
                                case SensorType_BloodPressure:                          units = "mmHg " + scale ? "(Diastollic)" : "(Systollic)";                       break;
                                                                                                ~~~~~~~~^~~~~~~
/home/donald/open-zwave/cpp/src/command_classes/SensorMultilevel.cpp:501:55: note: use array indexing to silence this warning
                                case SensorType_BloodPressure:                          units = "mmHg " + scale ? "(Diastollic)" : "(Systollic)";                       break;
                                                                                                        ^
                                                                                                &       [      ]
1 error generated.
gmake[1]: *** [/home/donald/open-zwave/cpp/build/support.mk:139: /home/donald/open-zwave/.lib/SensorMultilevel.o] Error 1
gmake[1]: Leaving directory '/home/donald/open-zwave/cpp/build'
gmake: *** [Makefile:23: all] Error 2
root@fileserver:/home/donald/open-zwave #

```
Best Regards,
Donald.

AlbertoLopSie, 2018-09-18:
Same problem here on MacOS 10.13 (High Sierra)(

AlbertoLopSie, 2018-09-18:
Found the fix. The offending line should be:

`units = scale ? "mmHg (Diastollic)" : "mmHg (Systollic)";                       break;`

to keep the MacOS (and probably FreeBSD) gcc compiler happy (it doesn't seems to like character string concatenation combined with the ternary op)

Fishwaldo, 2018-09-19:
Fixed. Thanks

closed 2018-09-19 by Fishwaldo